pschatzmann/arduino-audio-tools #152: player-url_icy-audiokit.ino example crashes without MetadataCallback function set.

perlix, 2022-03-31:
Running the current _player-url_icy-audiokit.ino_ example crashes my ESP32 on calling processMetaData() in line 134 of MetaDataICY.h, immediately when metadata comes in. 

```
PC: 0x00000000
EXCVADDR: 0x00000000

Decoding stack results
0x400d3de6: audio_tools::MetaDataICY::processChar(char) at F:\Arduino\libraries\arduino-audio-tools\src/AudioMetaData/MetaDataICY.h line 134
0x400d3e3a: audio_tools::ICYStreamDefault::readBytes(unsigned char*, unsigned int) at F:\Arduino\libraries\arduino-audio-tools\src/AudioHttp/ICYStream.h line 90
0x400d835e: audio_tools::BufferedTaskStream::task(void*) at F:\Arduino\libraries\arduino-audio-tools\src/AudioHttp/URLStreamESP32.h line 136
0x4008f1b6: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

After adding a printMetaData() function to the sketch and linking it with `player.setMetadataCallback(printMetaData, SELECT_ICY); `everything works fine!

pschatzmann, 2022-03-31:
When did you last update the library ? The latest correction is from 7 days ago...

pschatzmann, 2022-03-31:
If you installed the library with git clone you can easily update it with git pull

closed 2022-04-10 by perlix