pschatzmann/arduino-audio-tools #155: can´t apply effects on audiokit input

xabadabazinho, 2022-04-01:
Hi, thanks for the fantastic job.
Trying to apply effects here.
The example with a sine wave is all right (with some modification, I think there are typos). 
When I try to use the audioKit in, the is no sound. 
I tested the audiokit input with other example and it's ok - the hardware is good.
Using Platformio. 
Truly sorry if being an idiot. 
Thanks in advance.

The code:

#include "AudioTools.h"
#include <Arduino.h>
#include "AudioLibs/AudioKit.h"

SineWaveGenerator<int16_t> sine;
AudioKitStream kit;

/////////////Swap beetween this two lines///////////////////////////
//AudioEffects<SineWaveGenerator<int16_t>> effects(sine);           //works as a champ
AudioEffects<GeneratorFromStream<effect_t>> effects(kit,2, 1.0); //NOT woking, no audio
///////////////////////////////////////////////////////////////////

GeneratedSoundStream<int16_t> in(effects);
StreamCopy copier(kit, in);

Fuzz fx(6.5, 300);

void setup()
{
    Serial.begin(115200);
    AudioLogger::instance().begin(Serial, AudioLogger::Warning);
    // setup effects
    effects.addEffect(fx);
    // Setup output
    auto cfg = kit.defaultConfig(RXTX_MODE);
    kit.begin(cfg);
    // Setup sound generation based on AudioKit settings
    sine.begin(cfg, 1000);
    in.begin(cfg);
}

void loop()
{
    copier.copy();
} 
/////////////////////////////////////////////////////////////////////////////////////////

Monitor:
> Executing task: C:\Users\zeden\.platformio\penv\Scripts\platformio.exe device monitor <

--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters

Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

--- Miniterm on COM6  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
[I] I2SConfig.h : 89 - rx/tx mode: RX
[I] I2SConfig.h : 90 - port_no: 0
[I] I2SConfig.h : 91 - is_master: Master
[I] I2SConfig.h : 92 - sample rate: 44100
[I] I2SConfig.h : 93 - bits per sample: 16
[I] I2SConfig.h : 94 - number of channels: 2
[I] I2SConfig.h : 95 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 97 - auto_clear:0
[I] I2SConfig.h : 104 - buffer_count:10
[I] I2SConfig.h : 105 - buffer_size:512
[I] I2SConfig.h : 109 - pin_bck: 14
[I] I2SConfig.h : 110 - pin_ws: 15
[I] I2SConfig.h : 111 - pin_data: 0
[I] I2SConfig.h : 113 - pin_data_rx: 0
[W] AudioKit.h : 43 - sd_active = true
[I] AudioKit.h : 193 - Channels 2->2
[I] AudioOutput.h : 635 - void audio_tools::FormatConverterStream::setup()
[I] AudioOutput.h : 648 - FormatConverterStream: no conversion
[I] AudioKit.h : 540 - void audio_tools::AudioKitStream::setupActions()
[W] AudioKit.h : 552 - Mode Button ignored because of conflict: 13
[I] AudioActions.h : 42 - ActionLogic::add pin: 39 / logic: 2
[I] AudioActions.h : 61 - pin 39 -> INPUT
[I] AudioKit.h : 293 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 42 - ActionLogic::add pin: 18 / logic: 0
[I] AudioActions.h : 58 - pin 18 -> INPUT_PULLUP
[I] AudioKit.h : 293 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 42 - ActionLogic::add pin: 5 / logic: 0
[I] AudioActions.h : 58 - pin 5 -> INPUT_PULLUP
[I] SoundGenerator.h : 179 - SineWaveGenerator::begin(channels=2, sample_rate=44100, frequency=1000)
[I] SoundGenerator.h : 165 - bool audio_tools::SineWaveGenerator<T>::begin() [with T = short int]
[I] SoundGenerator.h : 38 - bits_per_sample: 16
[I] SoundGenerator.h : 39 - channels: 2
[I] SoundGenerator.h : 40 - sample_rate: 44100
[I] SoundGenerator.h : 206 - setFrequency: 1000
[I] SoundGenerator.h : 207 - active: true
[I] SoundGenerator.h : 38 - bits_per_sample: 16
[I] SoundGenerator.h : 39 - channels: 2
[I] SoundGenerator.h : 40 - sample_rate: 44100
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
...

Platformio.ini:

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
description = Audio Example
default_envs = esp32dev

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = 
	#https://github.com/pschatzmann/arduino-audiokit-hal
	https://github.com/pschatzmann/arduino-audio-tools
	#https://github.com/pschatzmann/arduino-libhelix
	#https://github.com/greiman/SdFat
	#rlogiacco/CircularBuffer@^1.3.3
	#https://github.com/pschatzmann/ESP32-A2DP
    https://github.com/pschatzmann/arduino-audiokit
lib_ldf_mode = deep+
monitor_filters = esp32_exception_decoder
build_flags = -DAUDIOKIT_BOARD=5 -Wno-unused-variable -Wno-unused-but-set-variable -Wno-unused-function -Wno-format-extra-args
monitor_speed = 115200

pschatzmann, 2022-04-13:
I was testing the GeneratorFromStream with the built in microphone:
```
#include "Arduino.h"
#include "AudioTools.h"
#include "AudioLibs/AudioKit.h"

AudioKitStream kit;
float factor = 1.0;
AudioEffects<GeneratorFromStream<effect_t>> effects(kit,2, factor); 

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);
  
  auto cfg = kit.defaultConfig(RXTX_MODE);
  cfg.input_device = AUDIO_HAL_ADC_INPUT_LINE2;
  kit.begin(cfg);
}

void loop() {
  Serial.println(effects.readSample());
}
```
I was getting some valid audio input. Maybe you need to define the input and increase the factor to amplify the signal.

[image]

xabadabazinho, 2022-04-17:
Ok, thank you very much.

closed 2022-04-17 by xabadabazinho