pschatzmann/arduino-audio-tools #161: Callback for accessing I2S stream data?

perlix, 2022-04-12:
Trying to add VU meters to the _player-url_icy-i2s.ino_ example, I wonder if this library provides a callback mechanism for eavesdropping on the I2S stream to the external DAC, i.e. something similar to the `set_stream_reader()` member of the BluetoothA2DPSink class.

pschatzmann, 2022-04-12:
No, but you can easily achieve this by creating your own subclass: e.g.
```
class MyI2S : public I2SStream {
  public:
      size_t write(uint8_t *data, size_t len) override {
         // -> add you logic here
         return I2SStream::write(data,len);
      }
}
```

perlix, 2022-04-12:
Thanks, I'll give it a try!

For adding stereo VU meter emulation to one of your bluetooth sink examples, I used the simple peak2peak algorithm below.

(_Scaling and final conversion to integers in the [0, 13] range is because the meters for that particular project are just minuscule gimmics (https://www.youtube.com/watch?v=y_OQBfdlD6Y), By choosing the right parameters and converting resulting values to, for instance, needle angles, much better looking meters can be achieved at larger displays_).

```

uint8_t n_samples = 100;
int16_t max_l, max_r, min_l, min_r;
int msecs = 100;
uint32_t tijd;
uint8_t il;
uint8_t ir;
....

void setup() {
  ....
  a2dp_sink.set_stream_reader(read_data_stream);
  tijd = millis();
}

void read_data_stream(const uint8_t *data, uint32_t length) {
  if ((millis() - tijd) > msecs && length / 2 >= n_samples) {
    int16_t *values = (int16_t*) data;
    max_l = -32768;
    max_r = -32768;
    min_l = 32768;
    min_r = 32768;
    for (int j = 0; j < n_samples; j += 2) {
      if (values[j] > max_l) max_l = values[j];
      if (values[j + 1] > max_r) max_r = values[j + 1];
      if (values[j] < min_l) min_l = values[j];
      if (values[j + 1] < min_r) min_r = values[j + 1];
    }
    // scaling and conversion to integers in the [0, 13] range, used as indexes in the VU images array
    il = (int)((il + (abs(32768 - (max_l - min_l)) / 2400)) / 2);
    ir = (int)((ir + (abs(32768 - (max_r - min_r)) / 2400)) / 2);
    tijd = millis();
  }
}
```

pschatzmann, 2022-04-12:
With the latest commit you could also try the following:
Output to MultiOutput to which you add an I2SStream and a VolumeOutput.
From the VolumeOutput you can determine the volume...

perlix, 2022-04-13:
> Output to [MultiOutput](https://pschatzmann.github.io/arduino-audio-tools/html/classaudio__tools_1_1_multi_output.html) to which you add an I2SStream and a [VolumePrint](https://pschatzmann.github.io/arduino-audio-tools/html/classaudio__tools_1_1_volume_print.html).

Just to confirm that this works. Thanks!

perlix, 2022-04-14:
> No, but you can easily achieve this by creating your own subclass in your sketch: e.g.
> ```
> class MyI2S : public I2SStream {
>   public:
>       size_t write(uint8_t *data, size_t len) override {
>          // -> add you logic here
>          return I2SStream::write(data,len);
>       }
> }
> ```
This method works great as well (just had to change `uint8_t *data` to `const uint8_t *data` ).

closed 2022-04-20 by pschatzmann