pschatzmann/arduino-audio-tools #172: Log volume to 0

podaen, 2022-04-22:
Hi phill,

I  am using:

I2SStream i2s;
VolumeStream volume(i2s);//volume controle kan niet terug naar 0
MP3DecoderHelix decoder;
AudioPlayer player(source, volume, decoder);

And I have notist that the volume can not be set to 0 anymore. Maybe it because my amp is turned in to maximum...

podaen, 2022-04-24:
Very strange... if I lower it with my keyboard in steps -0.1 than it works

if I set it directly to 0.0 it doesn't work. Volume(0.0);

void SelectPath(const char* path) {
	float volume0 = 0.0;
	Volume(float(volume0));
	//Serial.println("select station");
	//Serial.println("0 Volume");
	//delay(500);
	Serial.println(path);
	player.setPath(path);
	//unsigned long starttime = millis();
	//unsigned long endtime = starttime;
	//while ((endtime - starttime) <= 2000)
	//{
	//	endtime = millis();
	//	player.copy();
	//	Serial.print('.');
	//}
	//Serial.println("Set back Volume");
	//printName();//icy doesn't work with select path
	Serial.println("");
}
void Volume(float Volume) {
	player.setVolume(Volume);
	//volume.setVolume(Volume);
}

it would be handy having a getvolume to be sure the volume is set...

pschatzmann, 2022-04-24:
I think the issue is rather that the first setVolume() was ignored because the intialization sequence used by the Player is quite special. I just committed a correction.
Setting the volume to 0 however had always the impact of giving no sound in my tests, that were not related to the Player

podaen, 2022-04-24:
Phill something else, it would be easy to do a check if defined for all these settings in interest of updating audiotools....

podaen, 2022-04-24:
also after player.start() player.stop() the volume is reset...

podaen, 2022-04-24:
> I think the issue is rather that the first setVolume() was ignored because the intialization sequence used by the Player is quite special. I just committed a correction.

Thanks, All the issues are resolved now.

closed 2022-04-24 by podaen