pschatzmann/arduino-audio-tools #173: Why is DSP code expanded ?

sletz, 2022-04-22:
The code https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/streams-faust-i2s/Noise.dsp and https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/streams-generator-faust-i2s/volume.dsp is "expanded" ? Any reason not to keep the original "non expanded" Faust DSP code?

pschatzmann, 2022-04-22:
I didn't have the time to test what's working and what not. 
Just notice that in the current implementation the input and output buffer are the same because I did not want to wast any memory. Though I plan to have an option to change this

sletz, 2022-04-22:
But how what the expanded DSP code generated ? (by "expanded" I mean  code with `ID_325` kind of identifiers ?)

pschatzmann, 2022-04-22:
I am not aware that I did anything special. I just used FaustLive and exported from there..

sletz, 2022-04-22:
Then you could possibly just keep the original DSP source code?

pschatzmann, 2022-04-22:
Well, in Arduino there is no mechanism to automatically generate the the CPP code from the DSP code.

closed 2022-05-13 by pschatzmann