pschatzmann/arduino-audio-tools #174: player.copy () vs tasks

podaen, 2022-04-22:
I have a task with the player.copy and an I2C keypad and if I place them on the same priority without delaying anything the audio falters. it only happens with https streams that asking more resources that others

Now I m thinking of Semaphore, so the task keeps running until it finished, but when I look at the documentation I see specialized tasks for rx buffers witch one shut I use?

podaen, 2022-04-22:
This is intressting documentation...

pschatzmann, 2022-04-22:
I am not sure if I understand you correctly. Is your keyboard processing in a separate task from the audio copy.
The arduino loop() is running in it's own task and my recommendation would be to just do the audio copy there and potentially only activities which does not take up any time.
Anything which might block should go into a separate - low priority -  task.

podaen, 2022-04-22:
In a lower priorty is not an option. It simply doesn't get exicuded in time. If there is no special care task can be run aside to eachother when the are on the same priorty. Thats what I am seening.

podaen, 2022-04-22:
Do yyou delete you task player.copy once it has runned once over  it? I ussume this is when urlplayer task in the config is enabled...

podaen, 2022-04-22:
Or maybe i´ reasing the url buffersize and place the i2c task on a lower priority. I am just guessing.

pschatzmann, 2022-04-23:
If I remember right your buffers have been defined really big. I would rater try to decrease them e.g. to  1k  or 512 bytes to make the system more responsive.

podaen, 2022-04-23:
That was for the mp3 player. For the url player increasing it doesn't solve the problem. I m trying to set them to 512. I let you know how it turns out.

podaen, 2022-04-23:
I placed the copy in the loop but for now, I get random keys been pushed...

Set it to 512 does cutoff the stream. I set it back to 1024.

podaen, 2022-04-23:
I am facing still that there is no audio coming out after some time and the player.copy() is still true.

I'm investigating the get receive form the ssl_client.
```
int get_ssl_receive(sslclient_context *ssl_client, uint8_t *data, int length)
{
    //log_d( "Reading HTTP response...");   //for low level debug
    int ret = -1;

    ret = mbedtls_ssl_read(&ssl_client->ssl_ctx, data, length);

    //if (ret != 1) {
    //    Serial.println(handle_error(ret));
    //}
    //log_v( "%d bytes read", ret);   //for low level debug
    return ret;
}
```

OUTPUT:

> [ 63275][E][ssl_client.cpp:36] _handle_error(): [get_ssl_receive():380]: (418) UNKNOWN ERROR CODE (0180) : AES - Invalid data input length
> [219447][E][ssl_client.cpp:36] _handle_error(): [get_ssl_receive():380]: (417) UNKNOWN ERROR CODE (0180) : AES - Invalid input data
> error recieved: -30848
> Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.

podaen, 2022-04-23:
These are the best settings for https

> #ifndef DEFAULT_BUFFER_SIZE 
> #define DEFAULT_BUFFER_SIZE 1024//512//1024//2048//1024
> #endif
> 
> #ifndef DEFAULT_SAMPLE_RATE 
> #define DEFAULT_SAMPLE_RATE 44100
> #endif
> 
> #ifndef DEFAULT_CHANNELS 
> #define DEFAULT_CHANNELS 2
> #endif
> 
> #ifndef DEFAULT_BITS_PER_SAMPLE 
> #define DEFAULT_BITS_PER_SAMPLE 16
> #endif
> 
> #ifndef I2S_DEFAULT_PORT 
> #define I2S_DEFAULT_PORT 0
> #endif
> 
> #ifndef I2S_BUFFER_SIZE 
> #define I2S_BUFFER_SIZE 1024//1024//1024//512
> #endif
> 
> #ifndef I2S_BUFFER_COUNT 
> #define I2S_BUFFER_COUNT 20//10 // 20
> #endif
> 
> #ifndef A2DP_BUFFER_SIZE 
> #define A2DP_BUFFER_SIZE 1280
> #endif
> 
> #ifndef A2DP_BUFFER_COUNT 
> #define A2DP_BUFFER_COUNT 50
> #endif
> 
> #ifndef CODEC_DELAY_MS 
> #define CODEC_DELAY_MS 10
> #endif
> 
> #ifndef COPY_DELAY_ON_NODATA 
> #define COPY_DELAY_ON_NODATA 10
> #endif
> 
> #ifndef COPY_RETRY_LIMIT 
> #define COPY_RETRY_LIMIT 20
> #endif

You did tell me to set 

> #define  CONFIG_DISABLE_HAL_LOCKS y

I did that now and the keys are working

The stability of the player is also increased. But still the audio output breaks for sometime time (copy=true and no timout)...  and than suddenly starts back.

podaen, 2022-04-23:
> But still the audio output breaks for sometime time (copy=true and no timout)... and than suddenly starts back. And the keypad does not react anymore.

This seems a memory problem causes by CONFIG_DISABLE_HAL_LOCKS y, maybe I exicute the keypad task to much...

podaen, 2022-04-23:
This solves the reset

> void Task1Keypad(void* pvParameters) {
> 	Serial.print("xPortGetFreeHeapSize: ");
> 	Serial.println(xPortGetFreeHeapSize());//290116//Since ESP32 has 328 KiB of data RAM
> 	Serial.println("");
> 	Serial.println("keypad Task");
> 	TickType_t xLastWakeTime;
> 	// Perform an action every 10 ticks.
> 	const TickType_t xFrequency = 1000;//10//1000=evry second
> 	// Initialise the xLastWakeTime variable with the current time.
> 	xLastWakeTime = xTaskGetTickCount();
> 	for (;;) {//infinit loop
> 		//CheckKeypadCode();
> 		//Serial.println("keypad running");
> 		// Wait for the next cycle.
> 		vTaskDelayUntil(&xLastWakeTime, xFrequency);
> 		char key = keypad.getKey();
> 		Serial.print("xPortGetFreeHeapSize: ");
> 		Serial.println(xPortGetFreeHeapSize());//290116//Since ESP32 has 328 KiB of data RAM
> 		Serial.println("");
> 		//if (key) {
> 		//	Serial.println(key);
> 		//}
> 		//delay(10);//geen delay slechte audio, delay 30 haperd te veel// mischien samprifian task gebruiken voor audio
> 	}
> }

By determing the time the task is executed, prevents a lot of trouble with the player.

closed 2022-04-23 by podaen