pschatzmann/arduino-audio-tools #177: I2S microphone examples not working?

vanniaz, 2022-04-24:
I am trying the **streams-i2s-a2dp.ino** example with the same model of microphone (INMP441) but I am only getting noise. The Bluetooth connects and the information messages say that the packets are being copied, so the I2S is running, but the audio is garbled.
I tried explicitly setting the I2S configuration as in the other example **base-i2s-a2dp.ino**. If I change the format to I2S_LSB_FORMAT (as suggested in the comment) I actually hear something inside the noise, and this indicates that the microphone is working but probably the data format is wrong.
@pschatzmann since the microphone type is the same, was it working regularly in your tests? Is there something that I could be doing wrong?

According to this post [https://esp32.com/viewtopic.php?t=15185](url) the INMP441 requires a non-standard configuration, but this would need some modifications to the arduino-audio-tools library.

Thank you in advance for your help

pschatzmann, 2022-04-25:
Difficult to say. Did you double check that the issue is not related to the L/R pin and the Channel Filter. I suggest to check first with an output as CSV that you get a signal and with the filter that the signal is on both channels....

If that is not helping you can try to change the I2S setting to 32 bits - but the signal then needs to be converted down to 16bits before sending it to A2DP...

vanniaz, 2022-04-28:
I tried with **streams-i2s-serial.ino** and the values seem correct (I sent them to the Arduino serial plotter and I can actually see the waves if I whistle near the microphone).
So the problem is probably in the transfer from I2S to A2DP.

I see that in the other example **base-i2s-a2dp.ino** a conversion is performed before sending data to A2DP, shouldn't the same be done in **streams-i2s-a2dp.ino**?

So I tried to run **base-i2s-a2dp.ino** instead, but unfortunately it crashes when AD2P is starting.  However I did not spend much time on it so I will check again.

vanniaz, 2022-05-03:
Sorry for not having replied sooner, in the meantime I received my LyraT board so now I am focusing on that part. I am using your arduino-audio-kit, nice project!

closed 2022-06-27 by pschatzmann

aqildad-create, 2023-07-21:
I'm facing the same problem esp32 crashes .

Example with the same model of microphone (INMP441)