pschatzmann/arduino-audio-tools #178: A2DP Stream Issues

ofthedove, 2022-04-24:
I'd like to combine an A2DP sink stream with another stream using a mixer, then output them to an Audio Kit stream. To do this, it seems I need an A2DPStream instead of a BluetoothA2DPSink, so that I can connect the stream to the mixer.

Most examples use the BluetoothA2DPSink, but I was able to find a couple that use A2DPStream. Currently I'm looking at [sandbox/streams-a2dp-serial](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/sandbox/streams-a2dp-serial).

However, when I try to run the example, I get an error `Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.` I've been using the logging tools to try to track down this error but I'm starting to hit a wall.

The first issue I found is on [Line 160 of `src/AudioLibs/AudioA2DP.h`](https://github.com/pschatzmann/arduino-audio-tools/blob/bc2695a4eaa6e5f6cab8e338f331b64c7f235210/src/AudioLibs/AudioA2DP.h#L160). `a2dp_source` should be `a2dp_sink`, as source is a null pointer here in RX mode.

Fixing that got me a little further, but now I get the same `LoadProhibited` error on the `set_volume` function call a couple lines later. If I comment out the set volume call the board boots up successfully, but then crashes when a bluetooth device connects. It looks like, when the device connects, it tries to set the volume and that causes it to crash.

I did a little logging in `BluetoothA2DPSink.cpp` and it seems to be crashing during the lock, set, unlock sequence. That doesn't make any sense to me though, since examples using BluetoothA2DPSink work for me, and I don't see how that sequence could lead to a `LoadProhibited` error.

Here is the sketch I'm running, which is a modified version of the `streams-a2dp-serial` example. I think the only significant change is adding the Audio Logger.

[streams-a2dp-serial.ino.txt](https://github.com/pschatzmann/arduino-audio-tools/files/8550265/streams-a2dp-serial.ino.txt)

Here are the source files I modified, adding log statements and commenting out lines. (Note that I had to add `.txt` extension so GitHub would allow me to upload it.)

[AudioA2DP.h.txt](https://github.com/pschatzmann/arduino-audio-tools/files/8550250/AudioA2DP.h.txt)

[BluetoothA2DPSink.cpp.txt](https://github.com/pschatzmann/arduino-audio-tools/files/8550248/BluetoothA2DPSink.cpp.txt)

Here is a serial monitor log of the board powering up, me connecting my phone to it, and it resetting when it hits the set volume call. The core debug level is set to `debug` and the audio logger level is set to `Info`. I'm using a LyraT 4.3 board. 

[SerialMonitorLog.txt](https://github.com/pschatzmann/arduino-audio-tools/files/8550244/SerialMonitorLog.txt)

I'd love any help or advice. Thank you!

pschatzmann, 2022-04-25:
You are right. The last couple of changes have introduced some errors.
I just committed the related corrections to the ESP32-A2DP and the arduino-audio-tools projects.
The example https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-a2dp-serial is working now...

ofthedove, 2022-04-26:
Thank you so much! I've confirmed it's working on my board as well.

closed 2022-04-26 by ofthedove