pschatzmann/arduino-audio-tools #179: URL Stream Buffering

DrJaymz, 2022-04-25:
This is a great set of tools, I plugged everything together and it worked nearly first time, the only change I had to make was cfg to change pin14 to pin13 due to dodgy silkscreen on my TTGO-Display.

I set it up to play Portuguese radio station which has a 40ms ping from the UK, perhaps a little more for wireless.
The audio plays and stops twice every 5 seconds or so.  I was fairly certain this would be to do with latency or buffering.
I had a play with  local AudioConfig.h  settings to try to improve the situation. 

I tried the DEFAULT_BUFFER_SIZE,   then I found the URL stream section but I don't really know what those settings do.

At the moment the log is set to info, I see the copy 1024 flying by and every now and then the metadata print out.
When the audio stops the messages stop, for about a seconds and then carries on.

Any advise on what I might need to do to improve it?  I don't think there's any electrical issue.  I tried turning volume down to 0.1 to see if that changed anything - no different.
I tried BBC Radio which is very low latency and should be fine with our fibre internet - but its still not perfect.

Finally the player-url-i2s.ino seems to handle wifi internally, I only see SSID and password settings so I haven't messed nor fiddled with the wifi settings.

pschatzmann, 2022-04-25:
The DEFAULT_BUFFER_SIZE defines the number of bytes are copied with each copy step. I would not expect any improvements from there. By the way you better change this in your local sketch by just setting this size in the constructor of the Copier class.

My recommendation is to prefer __http__ over https. The ESP32 seems not to be very efficient using https. I noticed that any Arduino ESP32 Release >= 2.0 is worse then the original 1.0.6 version in handling https. So if you are using one of these you might want try to roll back.

You might be able to improve things a bit by defining you custom buffer_size and buffer_count i2s config parameters. With an increased buffer, if the wifi blocks, I2S still has some data left to use...

Last I always recommend changing the log level to Warning.

podaen, 2022-04-25:
I changed the board, from the old DOIT ESP32 devkit V1 to ESP32 wrover module, and already it seems longer playing than before. Not yet no audio for now. I keep you updated...

podaen, 2022-04-25:
15minutes now running on a https link and still no breaks... going to change to http station.

I don't have a firewall between it and have a super fast connection of 700Mbits. :-)

Phill, I know rollback to 1.0.6 would work, but can we look how we could solve this. The 2.0.2 is much more stable on https. They fixed some impotent bugs in the start of ssl... When I was running on 1.0.6 I have manualy updated my secure wifi library it worked. Soo I want to stay.

podaen, 2022-04-25:
The second http diden't work (no log received why), but that link worked beter with ESP32 devkit v1 selected.

Changing to next station (http), for the moment it seems stable, without breaks of what ever... I change back to the first https link and that it start to falters. eventually it restarted and now it running back stable...

podaen, 2022-04-25:
I needs some further testing to be sure it has something to do with the board. But first I have to test more with my configuration what could be wrong.

DrJaymz, 2022-04-25:
I am not using the display [yet].  I used the player-url-i2s.ino sketch more or less out of the box.

- I use platform IO and I'm using the TTGO Display board.
- I added the following URL  "http://media3.mcr.iol.pt/livefm/comercial.mp3/icecast.audio",
- The following pin  def cfg.pin_bck = 13;
- I followed the notes on using PIO.  adding the build flags etc.

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = 
    https://github.com/pschatzmann/arduino-audio-tools
    https://github.com/pschatzmann/arduino-libhelix

build_flags = -DCORE_DEBUG_LEVEL=2
    -Wno-unused-variable 
    -Wno-unused-but-set-variable 
    -Wno-unused-function 
    -Wno-format-extra-args 
monitor_speed = 115200
upload_speed = 776000
monitor_filters = esp32_exception_decoder
```

I think there is limited improvement by using http instead of https.  I also tried the BBC streams and they are doing the same so perhaps not latency at all.

I did increase the buffer count from 10 to 12, too much and it obviously won't start.  I don't think it made any difference.
```
// URLStream
#define URL_STREAM_CORE 0
#define URL_STREAM_PRIORITY 2
#define URL_STREAM_BUFFER_COUNT 12
#define STACK_SIZE 30000
#define URL_CLIENT_TIMEOUT 60000;
#define URL_HANDSHAKE_TIMEOUT 120000
```

I think tonight I will beef up the supply to the MAX8357 board, the decoupling is a bit naff and there is a bit of drop on the TTGO board's diode etc.  If you were to have a brown out on the MAX chip, would the ESP notice?  Or does it just keep outputting data?  The reason I didn't suspect it is because I turned the volume right down so it only draws a few milliamps and the info for each 1024 block stops.  Suggesting it was waiting for the next bit of the stream.

podaen, 2022-04-25:
Using an spi display together with an url player... hmmm, how much memory do you have available? I have 2mega but it won't be enough to run both on the same time for the https. I have now 12k left without display.

pschatzmann, 2022-04-25:
You are right, it makes more sense to increase the encoded buffer - instead of the i2s buffer: In this case it is the following 2 parameters:
- DEFAULT_BUFFER_SIZE, 
- URL_STREAM_BUFFER_COUNT
So it would make sense to try to max out URL_STREAM_BUFFER_COUNT

DrJaymz, 2022-04-25:
I changed it to 20.  It doesn't help and for some reason now its max volume and can't control it.
I added a 10uF low ESR cap and beefed up the wires.  That doesn't help, the PSU on the scope looks reasonably stable which is amazing because its loud as hell.

I think it is something at the networking side.  When it stops it makes a fzzt noise just as its pauses.  If I pull out the access point power then it plays for about 1 second and then I get the fzzt noise - so that is the noise it makes when its running out of receive buffer.  Sorry that last bit wasn't technical.

I will see how high I can get the buffer.

- Wise man once said - don't be a duffer always buffer.

pschatzmann, 2022-04-25:
Hmm, which sketch are you using? 
To avoid noise if there is no data you can try to set cfg.auto_clear= true;

DrJaymz, 2022-04-25:
player-url-i2s.ino except I run in platform io.

So, set it to 60 and in my loop I have added a check that runs every second.  That sets the volume and it works ok if you're not calling it constantly.

I think set to 60 and using 1024 and http its a lot better. 

Yes streaming does really need a bit more ram.  But it works just about. 

I will try using the display.  I think it will work ok provided I'm not trying to run doom on it.

pschatzmann, 2022-04-25:
If you have the issue that the initial volume setting is ignored, you will just need to trigger a reload of the latest library. (I committed a correction yesterday)..

DrJaymz, 2022-04-25:
Ahhhh. Ok.

podaen, 2022-04-25:
you don't have to listen to me, but you are still not answering my questions. Nothing will help before we clear it out.

DrJaymz, 2022-04-26:
> you don't have to listen to me, but you are still not answering my questions. Nothing will help before we clear it out.

Sorry what did you want to know the answer to?  I was asking about buffering. I think I got my answer, which is basically that the buffer is limited and perhaps not enough in some circumstances.

So it works, but its not perfect.  It can work perfectly for a hour or two and other times its just a stutter fest, and annoyingly the channel keeps changing - so I guess there's a timeout that I could increase for that issue. (also points to it being just taking too long to get the next bit of the stream).

I haven't tried the SPI display yet.  I was just about to add my libraries which I always add which gives me OTA, display, battery & wifi management.  
I got stuck again because I don't want to supply the WIFI details to ICYstream, I want a separation of concerns.  I'll have a look again later - the kids are the age where basically they suck up all my free time.

pschatzmann, 2022-04-26:
No worries, URLStream and ICYstream work w/o ssid and pasword. That's just an optional feature...

DrJaymz, 2022-04-26:
OK, I got back to this and changed the following lines

```
//ICYStream urlStream(wifi, password);
URLStream urlStream;
```
And now I can use my own wifi manager.

I happened to look at the esp32 in general.

Reply from 192.168.0.86: bytes=32 time=130ms TTL=255
Reply from 192.168.0.86: bytes=32 time=50ms TTL=255
Reply from 192.168.0.86: bytes=32 time=58ms TTL=255
Reply from 192.168.0.86: bytes=32 time=211ms TTL=255
Reply from 192.168.0.86: bytes=32 time=629ms TTL=255
Reply from 192.168.0.86: bytes=32 time=323ms TTL=255
Reply from 192.168.0.86: bytes=32 time=32ms TTL=255
Reply from 192.168.0.86: bytes=32 time=52ms TTL=255
Reply from 192.168.0.86: bytes=32 time=92ms TTL=255
Reply from 192.168.0.86: bytes=32 time=89ms TTL=255

I don't know about you guys, that seems a bit crappy to me.  I checked another board its about the same
I checked a esp8266 and its returning <10ms.  so perhaps wifi performance for some reason is the problem here.
what do you get on an esp32 on idle?

podaen, 2022-04-26:
> I haven't tried the SPI display yet. I was just about to add my libraries which I always add which gives me OTA, display, battery & wifi management.

Thats what I was talking about. You can not run it on the same time. What you could do is send an esp_now message and handle it with an other esp. But for now with the new library I don't thinnk we are there yet.

I have ota, I have spi displays, I have keyboards, I have sd cards, but with the 2.0.2 it just doesn't work quit yet.

Kids can take you. They just don't stop! :-)

> No worries: URLStream and ICYstream work w/o ssid and pasword. That's just an optional feature...

Not with select path what I implemented to the player, because the classes don't admit to that.

DrJaymz, 2022-04-26:
```
Reply from 192.168.0.86: bytes=32 time=4ms TTL=255
Reply from 192.168.0.86: bytes=32 time=4ms TTL=255
Reply from 192.168.0.86: bytes=32 time=5ms TTL=255
Reply from 192.168.0.86: bytes=32 time=3ms TTL=255
Reply from 192.168.0.86: bytes=32 time=6ms TTL=255
Reply from 192.168.0.86: bytes=32 time=3ms TTL=255
Reply from 192.168.0.86: bytes=32 time=3ms TTL=255
Reply from 192.168.0.86: bytes=32 time=5ms TTL=255
Reply from 192.168.0.86: bytes=32 time=6ms TTL=255
Reply from 192.168.0.86: bytes=32 time=3ms TTL=255
Reply from 192.168.0.86: bytes=32 time=5ms TTL=255
```

That is more like it.

You need   esp_wifi_set_ps(WIFI_PS_NONE);
the default is to have power saving.  which makes streaming performance crap.

pschatzmann, 2022-04-26:
Cool - I will try to figure out the impact...

podaen, 2022-04-26:
sound good!!!

podaen, 2022-04-26:
Can you post your code?

DrJaymz, 2022-04-26:
my display is working ok

is there a way I can share a private repo, I don't fancy all of it just now.

podaen, 2022-04-26:
just the wifi manager...

DrJaymz, 2022-04-26:
https://github.com/DrJaymz/Audio01_PT

DrJaymz, 2022-04-26:
so I have a fair bit of crap in there but you can see that I have created a commodore 64 like screen thingy and I am playing the radio and every couple of seconds I just write some stuff to the display.  That works perfectly fine it seems.

podaen, 2022-04-26:
you can delete your comment now. I will try to understand how wiffi manager works.

DrJaymz, 2022-04-26:
its fine I made it public its fine.

DrJaymz, 2022-04-26:
> 

don't understand?  Can you see it?

That WIFI manager is really good and makes everything very slick.  My function for it just appends the MAC to the hostname because I often have a few devices of the same time.

podaen, 2022-04-26:
Are you using gitub as ota server? I do. but not private.

DrJaymz, 2022-04-26:
no.  didn't know you could.

podaen, 2022-04-26:
Wait. I send you a link...

podaen, 2022-04-26:
https://www.youtube.com/watch?v=qCmdUtguwPw&t=78s

podaen, 2022-04-26:
what I can't find in your code is how you test the connection of a sertain stream url. What I get is that it is testing the general internet connection. am I right?

podaen, 2022-04-26:
I did a look up and seems like that. I m convinced it will help the issues on the input. Thanks for that.

pschatzmann, 2022-04-26:
I just committed the change with esp_wifi_set_ps(WIFI_PS_NONE);
I also took the chance to fix the broken URLStream for the ESP8266.
Can you double check if it is better now ?

DrJaymz, 2022-04-27:
what did you change with the URLStream?
The issue I had was that when I switched from ICY the callback wasn't firing.

pschatzmann, 2022-04-27:
If you mean metadata, this is only provided by ICY. I suggest that you open an issue with the ICY stream to add additinal constructors, so that you dont need to pass the userid and password.
This issue is about URLStream buffering. Do you still expect me to do something or can we close this ?

DrJaymz, 2022-04-27:
Nope I think we're done with the buffers and hopefully improved it a bit.

closed 2022-04-27 by DrJaymz