pschatzmann/arduino-audio-tools #180: Url player stops playing music after some time.

podaen, 2022-04-25:
Phill

I m investaging to work with arduino espressif 2.0.2 and look what could be the reason why the player does turn as not active. I found a something...

in AudioMozzi.h those files can not be found...

#include "mozzi_config.h"
#include "hardware_defines.h"
#include "mozzi_analog.h"
#include "Mozzi.h"

podaen, 2022-04-25:
What is mozzi... I don't know

podaen, 2022-04-25:
I suspect you have to define in the config whether it is used or not. However player.stop is referring to it soo please do.

podaen, 2022-04-25:
these three links are remarkable...

http://icecast-servers.vrtcdn.be/mnm_urb-high.mp3
https://25493.live.streamtheworld.com:443/NOSTALGIEWAFNEWWAVE.mp3
https://25493.live.streamtheworld.com:443/NOSTALGIEWAFEXTRA.mp3

the player end

```
        virtual void end() {
            Serial.println("player end");
            LOGD(LOG_METHOD);
            active = false;
            p_out_decoding->end();
            meta_out.end();
        }
```
and sometimes the connection fails

Here the complete test resluts of 78 radio stations

> 70 audio
> 71 audio
> 72 audio
> 73 audio
> 74 haperd
> 75 audio
> 76 audio
> 77 audio
> 
> 
> 0 audio
> 1 audio
> 2 https no audio
> [E] HttpRequest.h : 185 - Connect failed
> player end
> 
> 3 https no audio player not active
> 4 audio
> 5 audio
> 6 audio
> 7 audio
> 8 audio
> 9 audio
> 10 audio but stops http://icecast-servers.vrtcdn.be/ketnetradio-high.mp3
> 10. problems
> 11 audio
> 12 audio
> 13 audio
> 14 audio
> 15 audio
> 16 audio
> 17 audio
> 18 audio
> 19 audio
> 19 2x falters
> 20 audio stops http://icecast-servers.vrtcdn.be/mnm_urb-high.mp3
> 20 2x brownout
> 21 no audio https://25493.live.streamtheworld.com:443/NOSTALGIEWAFNEWWAVE.mp3
> [W] HttpHeader.h : 231 - Waiting for data...
> [W] HttpRequest.h : 145 - no CONTENT_LENGTH found in reply
> waitForData
> player end
> 
> 22 no audio
> https://25493.live.streamtheworld.com:443/NOSTALGIEWAFEXTRA.mp3
> [E] HttpRequest.h : 185 - Connect failed
> player end
> No stream-Timout
> player not active

> 23 audio
> [W] HttpHeader.h : 231 - Waiting for data...
> [W] HttpRequest.h : 145 - no CONTENT_LENGTH found in reply
> waitForData
> player end

> 24 no audio player end

> 25 audio
> 26 audio
> 27 audio
> 28 audio
> 29 audio
> 30 audio
> 31 audio
> 32 audio
> 33 audio
> 34 audio
> 35 audio
> 36 audio
> 37 no audio
> Auto Next
> write eeprom UrlSelector: 37
> Vbro
> http://stream.vbro.be:9100/vbro
> [10114398][I][WiFiClient.cpp:252] connect(): select returned due to timeout 3000 ms for fd 48
> [E] HttpRequest.h : 185 - Connect failed
> player end
> 38 audio
> 39 audio failters
> 40 audio
> 41 no audio
> [E] HttpRequest.h : 185 - Connect failed
> player end

> 42 no audio
> [E] HttpRequest.h : 185 - Connect failed
> player end
> No stream-Timout
> player not active

> 44 audio
> 45 audio
> 46 audio
> 47 audio
> 48 audio
> 49 audio
> 50 audio
> 51 audio
> 52 audio
> 53 audio
> 54 no audio
> [E] URLStream.h : 273 - Error code recieved ... stop waiting for reply
> player end

> 55 falters
> player end

> 56 audio
> 57 audio
> 58 audio
> 59 audio
> 60 audio
> 61 audio
> 62 audio
> 63 audio
> 64 audio
> 65 audio

> 66 no audio
> [E] URLStream.h : 273 - Error code recieved ... stop waiting for reply
> player end

> 67 no audio
> [E] URLStream.h : 273 - Error code recieved ... stop waiting for reply
> player end
> No stream-Timout
> player not active

> 68 audio
> 69 audio
> 70 audio
> 71 audio
> 72 haperd
> 73 haperd
> 74 haperd

All links that did work before and now they don't 

http://22183.live.streamtheworld.com/JOE.mp3,
https://23623.live.streamtheworld.com:443/TOPRETRO.mp3,
http://icecast-servers.vrtcdn.be/stubru_untz-high.mp3,
http://icecast-servers.vrtcdn.be/mnm_hits-high.mp3,
http://icecast-servers.vrtcdn.be/mnm_urb-high.mp3,
https://25493.live.streamtheworld.com:443/NOSTALGIEWAFNEWWAVE.mp3,
https://25493.live.streamtheworld.com:443/NOSTALGIEWAFEXTRA.mp3,
http://stream.vbro.be:9400/evergreen,
http://versuzradio.live-streams.nl/live,
http://23553.live.streamtheworld.com:80/RADIO_2_WESTVL_128.mp3,
https://25633.live.streamtheworld.com:443/WILLY.mp3,

podaen, 2022-04-25:
I once started with 250 links...

podaen, 2022-04-26:
For debugging it would help a lot for debuging if I could see the data that is comming in.

The header we know

> [I] HttpHeader.h : 168 - HttpHeader::readLine -> HTTP/1.0 200 OK
> [I] HttpHeader.h : 168 - HttpHeader::readLine -> Server: Icecast 2.4.3
> [I] HttpHeader.h : 168 - HttpHeader::readLine -> Date: Tue, 26 Apr 2022 00:55:15 GMT
> [I] HttpHeader.h : 168 - HttpHeader::readLine -> Content-Type: audio/mpeg
> [I] HttpHeader.h : 168 - HttpHeader::readLine -> Cache-Control: no-cache
> [I] HttpHeader.h : 168 - HttpHeader::readLine -> Expires: Mon, 26 Jul 1997 05:00:00 GMT
> [I] HttpHeader.h : 168 - HttpHeader::readLine -> Pragma: no-cache
> [I] HttpHeader.h : 168 - HttpHeader::readLine -> icy-br:128
> [I] HttpHeader.h : 168 - HttpHeader::readLine -> ice-audio-info: ice-samplerate=44100;ice-bitrate=128;ice-channels=2
> [I] HttpHeader.h : 168 - HttpHeader::readLine -> icy-br:128
> [I] HttpHeader.h : 168 - HttpHeader::readLine -> icy-name:ClubFM Oost-Vlaanderen
> [I] HttpHeader.h : 168 - HttpHeader::readLine -> icy-pub:1
> [I] HttpHeader.h : 168 - HttpHeader::readLine -> icy-url:http://www.clubfm.be
> [I] HttpHeader.h : 168 - HttpHeader::readLine -> 

But after that is all very unclear...

podaen, 2022-04-26:
Five of the links I tested where connection failed

these are the links

2/  https://23623.live.streamtheworld.com:443/TOPRETRO.mp3
22/ [https://25493.live.streamtheworld.com:443/NOSTALGIEWAFEXTRA.mp3](https://25493.live.streamtheworld.com/NOSTALGIEWAFEXTRA.mp3)
37/ http://stream.vbro.be:9100/vbro
41/ https://25633.live.streamtheworld.com:443/WILLY.mp3
42/ http://eu2.radioboss.fm:8124/stream,//english

podaen, 2022-04-26:
The reason is the same as the correction we made before with the mbed. During the cleanup of the previous ssl connection, the new http connection that has ben made runs in timout. 3000ms is to low, It has to be much more.

> http://eu2.radioboss.fm:8124/stream,
> [130297][V][ssl_client.cpp:332] stop_ssl_socket(): Cleaning SSL connection.
> Cleaning SSL connection.
> [133806][I][WiFiClient.cpp:252] connect(): select returned due to timeout 3000 ms for fd 48
> [E] HttpRequest.h : 185 - Connect failed

This is the solution for those five links. But when the links are not the same order (https and http scrambled) you will get a lot more connection fails! And eventually it will not make sens why sometimes it connects and sometimes it don't. 

I hope it cheered you all out.

podaen, 2022-04-26:
OK. I setted the timout to 10000, but it dident work for this link. It is jus a bad link that diden't work before. But it's not bad to increase it the have less failures.

I am selecting the esp doit devkit board again and see what it does

The link I am now test should work, it worked before without a problem

http://23603.live.streamtheworld.com/JOE_80S.mp3

when there is audio output

> [D] MetaDataFilter.h : 35 - size_t audio_tools::MetaDataFilter<Decoder>::write(uint8_t*, size_t) [with Decoder = libhelix::MP3DecoderHelix; size_t = unsigned int; uint8_t = unsigned char]
> [D] AudioStreams.h : 716 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
> [D] AudioStreams.h : 716 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
> [D] I2SStream.h : 124 - virtual size_t audio_tools::I2SStream::write(const uint8_t*, size_t)
> [D] I2SESP32.h : 75 - size_t audio_tools::I2SBase::writeBytes(const void*, size_t)
> [D] I2SESP32.h : 82 - i2s_write 4608 -> 4608 bytes
> [D] AudioStreams.h : 716 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
> [D] AudioStreams.h : 716 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
> [D] I2SStream.h : 124 - virtual size_t audio_tools::I2SStream::write(const uint8_t*, size_t)
> [D] I2SESP32.h : 75 - size_t audio_tools::I2SBase::writeBytes(const void*, size_t)
> [D] I2SESP32.h : 82 - i2s_write 4608 -> 4608 bytes
> [D] AudioStreams.h : 716 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
> [D] AudioStreams.h : 716 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
> [D] I2SStream.h : 124 - virtual size_t audio_tools::I2SStream::write(const uint8_t*, size_t)
> [D] I2SESP32.h : 75 - size_t audio_tools::I2SBase::writeBytes(const void*, size_t)
> [D] I2SESP32.h : 82 - i2s_write 4608 -> 4608 bytes
> [D] AudioStreams.h : 716 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
> [D] AudioStreams.h : 716 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
> [D] I2SStream.h : 124 - virtual size_t audio_tools::I2SStream::write(const uint8_t*, size_t)
> [D] I2SESP32.h : 75 - size_t audio_tools::I2SBase::writeBytes(const void*, size_t)
> [D] I2SESP32.h : 82 - i2s_write 4608 -> 4608 bytes
> [D] AudioStreams.h : 716 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
> [D] AudioStreams.h : 716 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
> [D] I2SStream.h : 124 - virtual size_t audio_tools::I2SStream::write(const uint8_t*, size_t)
> [D] I2SESP32.h : 75 - size_t audio_tools::I2SBase::writeBytes(const void*, size_t)
> [D] I2SESP32.h : 82 - i2s_write 4608 -> 4608 bytes
> [D] AudioPlayer.h : 797 - virtual void audio_tools::AudioPlayer::copy()
> [D] AudioCopy.h : 91 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
> [D] AudioOutput.h : 364 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 2048
> [D] CodecMP3Helix.h : 107 - virtual size_t audio_tools::MP3DecoderHelix::write(const void*, size_t): 2048

when it stops playing music

> [D] MetaDataFilter.h : 35 - size_t audio_tools::MetaDataFilter<Decoder>::write(uint8_t*, size_t) [with Decoder = libhelix::MP3DecoderHelix; size_t = unsigned int; uint8_t = unsigned char]
> [D] AudioStreams.h : 716 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
> [D] AudioStreams.h : 716 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
> [D] I2SStream.h : 124 - virtual size_t audio_tools::I2SStream::write(const uint8_t*, size_t)
> [D] I2SESP32.h : 75 - size_t audio_tools::I2SBase::writeBytes(const void*, size_t)
> [D] I2SESP32.h : 82 - i2s_write 4608 -> 4608 bytes
> [D] AudioStreams.h : 716 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
> [D] AudioStreams.h : 716 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
> [D] I2SStream.h : 124 - virtual size_t audio_tools::I2SStream::write(const uint8_t*, size_t)
> [D] I2SESP32.h : 75 - size_t audio_tools::I2SBase::writeBytes(const void*, size_t)
> [D] I2SESP32.h : 82 - i2s_write 4608 -> 4608 bytes
> [D] AudioStreams.h : 716 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
> [D] AudioStreams.h : 716 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
> [D] I2SStream.h : 124 - virtual size_t audio_tools::I2SStream::write(const uint8_t*, size_t)
> [D] I2SESP32.h : 75 - size_t audio_tools::I2SBase::writeBytes(const void*, size_t)
> [D] I2SESP32.h : 82 - i2s_write 4608 -> 4608 bytes
> [D] AudioStreams.h : 716 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
> [D] AudioStreams.h : 716 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
> [D] I2SStream.h : 124 - virtual size_t audio_tools::I2SStream::write(const uint8_t*, size_t)
> [D] I2SESP32.h : 75 - size_t audio_tools::I2SBase::writeBytes(const void*, size_t)
> [D] I2SESP32.h : 82 - i2s_write 4608 -> 4608 bytes
> [D] AudioStreams.h : 716 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
> [D] AudioStreams.h : 716 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
> [D] I2SStream.h : 124 - virtual size_t audio_tools::I2SStream::write(const uint8_t*, size_t)
[... cut ...]
> #define PIN_I2S_WS 25//15
> #define PIN_I2S_DATA_IN 32
> #define PIN_I2S_DATA_OUT 27//22
> #define I2S_USE_APLL false  
> #define PIN_I2S_MUTE 23
> #define SOFT_MUTE_VALUE LOW  
> #define PIN_CS SS
> #define PIN_ADC1 34 
> #define PIN_ADC2 14
> 
> #define I2S_AUTO_CLEAR true
> 
> // URLStream
> #define URL_STREAM_CORE 0
> #define URL_STREAM_PRIORITY 1//2
> #define URL_STREAM_BUFFER_COUNT 20//10
> #define STACK_SIZE 30000
> #define URL_CLIENT_TIMEOUT 60000;
> #define URL_HANDSHAKE_TIMEOUT 120000

podaen, 2022-04-26:
The second link stops after a minute with the same working settings form the previous comment I have made here.

http://22183.live.streamtheworld.com/JOE.mp3

With //#define USE_URLSTREAM_TASK commented the link works for an halve an hour.

it has some difficulty some times. It falters... and than it stops

podaen, 2022-04-26:
what I am thinking is that it can't handle the in input.

podaen, 2022-04-26:
Doing the test now with one channel and double speed...

#define DEFAULT_SAMPLE_RATE 88200//44100
#define DEFAULT_CHANNELS 1//2

And I have still two channels... Now running for 10 minutes and still active. :-)

podaen, 2022-04-26:
it still breaks :-/

podaen, 2022-04-26:
it is very strang that it starts playing again after some time. It is like he has find its begin back...

podaen, 2022-04-27:
Soo I set the settings back to the originals, and it's outputting sometimes emply peaces

> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> 85 72 B8 18 38 70 F8 E7  - 6A 14 E8 41 7B B5 9D 50 
> 86 31 58 B4 A3 23 E9 65  - 7B 35 A7 FF A3 91 5D 19 
> 52 CF 6A 31 D2 97 DD 3B  - 7E E8 88 B4 4B CC 9E 7D 
> 94 E8 41 A2 E7 A6 40 D5  - 90 2D 60 3 82 F A4 F 
> 48 81 E8 A7 5E AC 92 24  - 0 3 40 0 22 4D BF 44 
> B4 D2 38 F4 72 23 D6 47  - 97 C2 42 89 15 4D 0 E2 
> F4 A4 18 51 1D CD 26 31  - 99 5A B6 DE CA E9 63 AA 
> 21 FC 7F C6 33 F8 9B FF  - FE DA 5D E8 FF F5 4C 6E 
> 0 4 E8 0 6 88 14 AB  - DF B5 4A FF A1 2A E 6C 
> F0 5C 9D CA C1 EB 97 35  - AA 1B B B9 97 2E F3 E1 
> C5 BE 59 23 85 44 16 44  - F3 10 94 72 A 98 3 C1 
> 53 CC BB 58 BB 74 C4 9A  - D9 4E B4 8D 2A 45 6B 78 
> A0 22 C A2 0 87 7F A  - 44 2 2D 88 A2 6E 65 BD 
> 91 15 56 B4 33 F5 EA C6  - 28 36 2 A 24 D6 37 AB 
> 3C B5 45 BD 9B 5E CB FF  - AF FE 9D 11 4C 5F 75 21 
> 50 C5 47 A E6 36 FA 12  - 10 1C 26 C1 1 40 FB 9C 
> 4D 8C A5 B6 34 78 C4 2D  - 25 EE 1C 39 61 AB D1 51 
> 33 F6 F3 FC 1C A8 6 29  - 97 97 39 9C 94 77 EB 21 
> A4 CD 67 F1 5F 15 28 73  - CB 97 F A4 7 29 58 3E 
> FC B9 60 1A 62 A 6A 29  - 99 71 89 81 80 0 0 0 
> 0 0 0 0 0 0 0 0  - 0 0 0 0 0 0 FF FB 
> 92 64 0 3 3 DA 2B 4B  - C3 6 15 A2 34 20 8 D9 
> 4 22 6E F AD 3F 20 AC  - 24 73 18 F6 0 61 30 0 
> 88 0 0 C 0 0 5F 0  - F9 88 E6 BD 24 4C F1 E2 
> 71 A9 A5 AE 2C B9 BB E7  - E CA 99 D3 5B 59 61 41 
> 83 A2 65 33 C1 1B B7 9C  - 66 1 B0 D4 DD D9 B B5 
> F D5 8A 4B A 14 75 40  - 35 20 1 45 1D B2 23 7 
> 0 A3 30 AE 4A 97 8D 28  - EB 23 2 2E 5B D5 58 4E 
> 34 CB 36 96 32 6E 6C 52  - F6 FB 5F FD FC EB D9 A3 
> 15 68 ED 4D DF 58 E9 D7  - E3 70 37 CE 43 A7 96 7F 
> A8 FF FE CC 61 2D 93 AB  - F3 F6 74 AD 23 2 81 58 
> 12 50 2 EA 7B CC A A4  - DD 67 A1 7A DC BA 15 D4 
> 49 A 5B C8 99 31 7E D5  - 13 15 86 5D B 25 43 41 
> C6 A9 BF FF F2 BD 67 BF  - FF FE 25 C D0 D9 B1 99 
> F3 21 40 2E 24 4A 2 24  - D8 90 11 50 55 C5 8C B9 
> 2C 9 9D 35 E7 1A 5D 1A  - 95 51 BB 33 F2 D8 5 21 
> 5D 47 1A 66 32 8 85 84  - C2 11 2A C4 4D 92 8D 19 
> BC 5A 5B E9 16 7F E3 8B  - 6C 64 B2 24 48 91 22 69 
> 64 2C C5 8 95 55 91 99  - BE 6D D5 F6 6E 2A A9 55 
> 55 E3 33 31 F1 B5 5F 55  - 55 6F FD BF 52 63 E7 B 
> 6D 7F 5E 33 31 B3 31 6C  - 61 40 45 1B BC 9E 36 FF 
> F9 75 C7 64 97 C2 82 8E  - E9 A1 58 8A FC A6 C4 8 
> 40 0 0 0 84 82 57 3D  - C4 6B 5 65 44 47 84 A0 
> D5 47 84 2E A8 1A 5 4E  - C4 47 A0 AB 95 9D E5 89 
> 30 B0 34 78 A9 D5 84 87  - B8 3B 12 FF FF DF C7 FF 
> FF FC A9 D0 68 F0 34 F0  - E8 4B A 84 93 10 53 51 
> 4C CB 8C 4C C 0 0 0  - 0 0 0 0 0 0 0 0 
> FF FB 92 64 0 F F2 6F  - 64 B0 10 60 2E 32 2F 2 
> A4 A2 0 26 0 0 0 1  - A4 0 0 0 20 0 0 34 
> 80 0 0 4 0 B B5 96  - 72 CF FC D5 A8 20 56 59 
> 65 92 CB 2C B2 5F FF FC  - C5 32 8C 1A 22 30 6A 28 
> C1 A3 2 C2 23 4 C4 14  - 60 D0 A0 98 D1 3 90 E4 
> 52 3B EF FF D5 15 51 51  - 51 51 55 35 45 45 54 54 
> 54 54 77 FD 4C 24 22 31  - 51 46 D 18 24 34 60 65 
> 55 52 20 0 69 CD 3B 49  - A7 16 CE 51 65 16 59 47 
> C5 B3 B5 DC 6E 6C D4 9C  - 70 19 96 86 99 FF FF FF 
> FF FF FF FF FF E2 C2 AC  - 10 8A 8A B 45 85 85 45 
> 52 98 82 9A 8A 66 5C 62  - 60 60 0 0 0 0 0 0 
> 0 0 0 0 0 0 0 0  - 0 0 0 0 0 0 0 0 
> 0 0 0 0 0 0 0 0  - 0 0 0 0 0 0 0 0 
> 0 0 0 0 0 0 0 0  - 0 0 0 0 0 0 0 0 
> 0 0 0 0 0 0 0 0  - 0 0 0 0 0 0 0 0 
> 0 0 0 0 0 0 0 0  - 0 0 0 0 0 0 0 0 
> 0 0 0 0 0 0 0 0  - 0 0 0 0 0 0 0 0 
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> 0 0 0 0 0 0 0 0  - 0 0 0 0 0 0 0 0 
> 0 0 0 0 0 0 0 0  - 0 0 0 0 0 0 0 0 
[... 72 lines cut ...]
CD 59 BE 97 3B A9 B9 56  - F1 62 24 3 F 75 E3 59 
A3 73 CC BE 5C FB 56 D7  - B5 89 6B C3 27 5 6 6 
1A 5B D6 B 2F 73 4C CE  - B2 4B 12 4E 97 43 18 B0 
48 2C 46 8A F1 77 74 8C  - 42 D5 88 71 E0 9C 51 CF 
9 F4 76 FC AA E7 6D 4A  - 17 CE 30 0 32 32 9E 4B 
1C [I] AudioCopy.h : 121 - StreamCopy::copy 161 -> 161 -> 161 bytes - in 1 hops

This is not a random stop!

6B FE 7D 7F F4 D6 58 D  - 25 73 37 8A 83 AA 31 15 
28 E5 FA 91 47 50 82 D4  - 22 48 48 18 1A 48 74 A1 
5 99 81 81 1C D4 ED B0  - 9D A5 CD 25 1D 20 2E B1 
54 9 B1 2E E2 A4 2 6C  - BB 60 EC 25 E9 1B 2C 17 
64 68 82 B2 B5 EA 70 9A  - 9B 70 84 5A D8 A9 86 80 
81 ED 6C B2 F4 97 F1 33  - 2 D3 45 21 95 42 78 87 
EA 32 19 5B FE A6 7F FF  - 11 FF B BF FF FF F9 56 
56 C7 58 F2 49 A8 9 7C  - 35 89 DA 47 97 4A C1 A2 
F4 96 20 96 8B 8E 17 DE  - A7 30 71 DF A8 75 A7 51 
9 4 6B A1 91 4B 47 A5  - EE 74 E4 97 3E 64 67 97 
73 [I] AudioCopy.h : 121 - StreamCopy::copy 161 -> 161 -> 161 bytes - in 1 hops

The second test with same settings, the input stream cutoff at almost the same time it stops at exactly 161 bytes
The data it self is not the same. So just a cutoff after one minute or an amount of packages. I think that the server is just trowing me out for a reason. It sound like bad internet connection, firewall or just not fast enough... And I think I would be the last because it was working on 1.0.6. PS: it is a http link with a strong budget server (http://22183.live.streamtheworld.com/JOE.mp3)

And I did set the  esp_wifi_set_ps(WIFI_PS_NONE); power mode to non sleep.

podaen, 2022-04-27:
I found this in my header:

> connection: close

But the server uses:

> connection: keep-alive

I tried, but it keeps cutting off. 

these I found In my browser

[image]

But some setting I can not find

Accept-Encoding
Accept-Language
Upgrade-Insecure-Requests:1

podaen, 2022-04-27:
> The second test with same settings, the input stream cutoff at almost the same time

I did the test over and set the library power saving to true, but still it cuttoff.

Now I tried to set decrease the audio output in the serial monitor by comment this in the AudioOutput.h

```
        virtual size_t write(const uint8_t* data, size_t len) {   
            if (!active) return 0;
             LOGD(LOG_METHOD);
            //for (size_t j=0;j<len;j++){
            //    out_ptr->print(data[j], HEX);
            //    out_ptr->print(" ");
            //    pos++;
            //    if (pos == 8){
            //        Serial.print(" - ");
            //    }
            //    if (pos == 16){
            //        Serial.println();
            //        pos = 0;
            //    }
            //}
            return len;
        }
```
And now it still running after 10 minutes, I see some that the buffer isn't always full...

> [I] AudioCopy.h : 121 - StreamCopy::copy 164 -> 164 -> 164 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 164 -> 164 -> 164 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 956 -> 956 -> 956 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 681 -> 681 -> 681 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 560 -> 560 -> 560 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 328 -> 328 -> 328 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 52 -> 52 -> 52 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 690 -> 690 -> 690 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 164 -> 164 -> 164 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 164 -> 164 -> 164 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 717 -> 717 -> 717 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 164 -> 164 -> 164 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[... cut ...]
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 164 -> 164 -> 164 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> [I] AudioCopy.h : 121 - StreamCopy::copy 164 -> 164 -> 164 bytes - in 1 hops

podaen, 2022-04-27:
Did you know there is a maximum on the header length

https://esp32.com/viewtopic.php?t=24750

pschatzmann, 2022-04-27:
My current impression is this is an issue of your specific URL. I think after some time they start to send silence. 
I changed the URLStream to ICYStream which is much more strict because it expects a metadata entry every n bytes (defined by metaint). I continue to get valid data - but just with no sound.

We could double check this by sending the data to a MultiOutput which consists of I2SStream and a VolumeStream. In the loop we can print the actual volume and I expect this to be 0 when we receive silence

podaen, 2022-04-27:
> I continue to get valid data - but just with no sound.

> I] AudioCopy.h : 121 - StreamCopy::copy 512 -> 512 -> 512 bytes - in 1 hops

In the beginning yeah, but after a few seconds, those are in between

> [I] AudioCopy.h : 121 - StreamCopy::copy 84 -> 84 -> 84 bytes - in 1 hops

I am going to see what a working link does but I think it is always the size of the default buffer

pschatzmann, 2022-04-27:
OK - I have the issue with other URLs as well...

podaen, 2022-04-27:
yes

pschatzmann, 2022-04-27:
Did you already check that the issue is not a broken I2S functionality: Is the player from files working stable ?

podaen, 2022-04-27:
I think my link is a DAB+ radio. and with a leak of performance during the update it slowed down. Even with improved setting we won't get there. Is there a we have to give priority to the input to get default buffer full.

pschatzmann, 2022-04-27:
I deactivated the task and added the MeasuringStream on the receiving side with the following result:
With audible audio: 15:54:15.025 -> [I] AudioStreams.h : 959 - Bytes per second: 15000
When audio is breaking up 15:54:19.814 -> [I] AudioStreams.h : 959 - Bytes per second: 88000
So the issue is not that the data is coming too slow...

I have more the impression now that the I2S functionality is broken because it should stall the processing.

I will play some music from the SD card to double check.

podaen, 2022-04-27:
How do you you know that 15kbs-88kbs is enough? I suspect that the input is between 128 and 192kb/s...

But yeah, I think also that the I2S breaks. Because I still see there is audio, but won't come out.

pschatzmann, 2022-04-27:
I think we have the issue also playing files: after some (long) time we have the same effect. Just with the difference that we reach the end of the file quickly and when we switch to the next file the error is corrected by the file change logic.

podaen, 2022-04-27:
diden' t see anything that was higher. I let you do now and I would happy to test it.

pschatzmann, 2022-04-27:
Maybe it's worth to activate the logging in the mp3 decoder so that we have a chance to figure out if the issue is coming from there...

podaen, 2022-04-27:
How we could do? It will overload the serial monitor. The characters need te be low.

podaen, 2022-04-27:
Are you already testing, I don't know how to enable it...

podaen, 2022-04-27:
How can I enabled it in the Url-player?

pschatzmann, 2022-04-27:
I updated the settings in the libhelix project. 
MP3_MAX_FRAME_SIZE  is defined in MP3DecoderHelix.h
HELIX_LOG_LEVEL  in helix_log.h

podaen, 2022-04-27:
Then I increased to much for the sd-player the previous time. I set the MP3_MAX_OUTPUT_SIZE 1024*10. Is that ok? Im having now indeed a continues output of my default buffer size. It causes a lot of afford finding it. Why not setting it as default to 2889??

pschatzmann, 2022-04-27:
I was was talking about MP3_MAX_FRAME_SIZE which is the buffer for the unencoded frame. MP3_MAX_OUTPUT_SIZE is for the result of the decoding.

podaen, 2022-04-27:
Yeah but why is there a maximum on the MP3_MAX_FRAME_SIZE? For my decoder board I can't set above I2S_BUFFER_SIZE not higher than 1024. But I don't have issues by setting the the frame size to 3000.

The speed is fluctuating. My log starts to slow down, but I still have music

> [I] AudioCopy.h : 121 - StreamCopy::copy 512 -> 512 -> 512 bytes - in 1 hops
> libhelix - appendToBuffer: 512 (at 0x3ffaef88)
> libhelix - -> Written 512 of 512 - Counter 35164
> [I] AudioCopy.h : 121 - StreamCopy::copy 512 -> 512 -> 512 bytes - in 1 hops
> libhelix - appendToBuffer: 512 (at 0x3ffaef88)
> libhelix - -> Written 512 of 512 - Counter 35165
> [I] AudioCopy.h : 121 - StreamCopy::copy 512 -> 512 -> 512 bytes - in 1 hops
> libhelix - appendToBuffer: 418 (at 0x3ffaef88)
> libhelix - -> Written 418 of 512 - Counter 35166
> libhelix - appendToBuffer: 94 (at 0x3ffaef88)
> libhelix - -> Written 512 of 512 - Counter 35167
> [I] AudioCopy.h : 121 - StreamCopy::copy 512 -> 512 -> 512 bytes - in 1 hops
> libhelix - appendToBuffer: 512 (at 0x3ffaef88)
> libhelix - -> Written 512 of 512 - Counter 35168
> [I] AudioCopy.h : 121 - StreamCopy::copy 512 -> 512 -> 512 bytes - in 1 hops
> libhelix - appendToBuffer: 512 (at 0x3ffaef88)
> libhelix - -> Written 512 of 512 - Counter 35169
> [I] AudioCopy.h : 121 - StreamCopy::copy 512 -> 512 -> 512 bytes - in 1 hops
> libhelix - appendToBuffer: 512 (at 0x3ffaef88)
> libhelix - -> Written 512 of 512 - Counter 35170
> [I] AudioCopy.h : 121 - StreamCopy::copy 512 -> 512 -> 512 bytes - in 1 hops
> libhelix - appendToBuffer: 460 (at 0x3ffaef88)
> libhelix - -> Written 460 of 512 - Counter 35171

> I just can not figure out yet what is happending...

Don't worry about it for now. It is running stable. If it breaks I let you know and I keep the logging on.

podaen, 2022-04-27:
Oeps, it just stops putting audio out. I restarted and now I am sure it is running much faster now. I turn of the logging and see if it keeps running...

podaen, 2022-04-27:
I am inspecting the code and I found this...

> Error (active)	E0020	identifier "__builtin_expect" is undefined	player-url-i2s	E:\Documents\Visual Studio 2019\Projects\Arduino\libraries\arduino-libhelix\src\MP3DecoderHelix.h	156

pschatzmann, 2022-04-27:
But, this is not in my code...

podaen, 2022-04-27:
it just stoped with me... Same procedure as yours

```
libhelix - -> Written 512 of 512 - Counter 58596
[I] AudioCopy.h : 121 - StreamCopy::copy 512 -> 512 -> 512 bytes - in 1 hops
```

pschatzmann, 2022-04-27:
This is very strange: Copy is providing the info that the write to EncodedStream has been processed. In the write we call the encoder in order to determine this result. But the encoder is not called. This is somehow not possible!

podaen, 2022-04-27:
Why we need an encoder? Ah for the mp3. This is the log for your url-player

> [I] AudioCopy.h : 121 - StreamCopy::copy 512 -> 512 -> 512 bytes - in 1 hops
> [D] AudioPlayer.h : 794 - virtual void audio_tools::AudioPlayer::copy()
> [D] AudioCopy.h : 91 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
> [D] AudioOutput.h : 380 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 512
> [D] CodecMP3Helix.h : 107 - virtual size_t audio_tools::MP3DecoderHelix::write(const void*, size_t): 512
> [D] MetaDataFilter.h : 35 - size_t audio_tools::MetaDataFilter<Decoder>::write(uint8_t*, size_t) [with Decoder = libhelix::MP3DecoderHelix; size_t = unsigned int; uint8_t = unsigned char]
> libhelix - write 512
> libhelix - writeFrame 512
> libhelix - appendToBuffer: 512 (at 0x3ffaef88)
> libhelix - appendToBuffer 2100 + 512  -> 2612
> libhelix - synchronizeFrame
> libhelix - -> frameRange -> 0 - 417
> libhelix - -> we are at beginning of synch word
> libhelix - decode 417
> libhelix - -> bytesLeft 2612 -> 2195  = 417 
> libhelix - -> End of frame (417) vs end of decoding (417)
> libhelix - => provideResult: 2304
> [D] AudioStreams.h : 724 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
> [D] AudioStreams.h : 724 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
> [D] I2SStream.h : 125 - virtual size_t audio_tools::I2SStream::write(const uint8_t*, size_t)
> [D] I2SESP32.h : 75 - size_t audio_tools::I2SBase::writeBytes(const void*, size_t)
> [D] I2SESP32.h : 82 - i2s_write 4608 -> 4608 bytes
> libhelix -  -> decoded 417 bytes - remaining buffer_size: 2195
> libhelix - -> Written 512 of 512 - Counter 448
> [I] AudioCopy.h : 121 - StreamCopy::copy 512 -> 512 -> 512 bytes - in 1 hops
> [D] AudioPlayer.h : 794 - virtual void audio_tools::AudioPlayer::copy()
> [D] AudioCopy.h : 91 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
> [D] AudioOutput.h : 380 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 161
> [D] CodecMP3Helix.h : 107 - virtual size_t audio_tools::MP3DecoderHelix::write(const void*, size_t): 161
> [D] MetaDataFilter.h : 35 - size_t audio_tools::MetaDataFilter<Decoder>::write(uint8_t*, size_t) [with Decoder = libhelix::MP3DecoderHelix; size_t = unsigned int; uint8_t = unsigned char]
> libhelix - write 161
> libhelix - writeFrame 161
> libhelix - appendToBuffer: 161 (at 0x3ffaef88)
> libhelix - appendToBuffer 2195 + 161  -> 2356
> libhelix - synchronizeFrame
> libhelix - -> frameRange -> 0 - 418
> libhelix - -> we are at beginning of synch word
> libhelix - decode 418
> libhelix - -> bytesLeft 2356 -> 1938  = 418 
> libhelix - -> End of frame (418) vs end of decoding (418)
> libhelix - => provideResult: 2304
> [D] AudioStreams.h : 724 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
> [D] AudioStreams.h : 724 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
> [D] I2SStream.h : 125 - virtual size_t audio_tools::I2SStream::write(const uint8_t*, size_t)
> [D] I2SESP32.h : 75 - size_t audio_tools::I2SBase::writeBytes(const void*, size_t)
> [D] I2SESP32.h : 82 - i2s_write 4608 -> 4608 bytes
> libhelix -  -> decoded 418 bytes - remaining buffer_size: 1938
> libhelix - -> Written 161 of 161 - Counter 449
> [I] AudioCopy.h : 121 - StreamCopy::copy 161 -> 161 -> 161 bytes - in 1 hops
> [D] AudioPlayer.h : 794 - virtual void audio_tools::AudioPlayer::copy()

pschatzmann, 2022-04-27:
Potential memory optimization: 
- You have MP3_MAX_OUTPUT_SIZE 1024*10
-  The log provides the info
libhelix - => provideResult: 2304

So this could be decreased quite a bit...

podaen, 2022-04-27:
I am going to feed my decoder board directly from the power supply. The 32ma it takes does not seems a problem for the esp32. I see now that the provideResult: 2304 stays the same. 

> So this could be decreased quite a bit...

So my decoder is fast enough. Good to know.

it is now running now for an hour without stopping on two esp's with the same code. One sometimes brownsout, the other not. I ' m going to make my own board with a beter chip. But it will take some time.

I feel there is nothing that can be done about the settings exept this...

#define COPY_RETRY_LIMIT 20 

this is new...

podaen, 2022-04-28:
Now I placed the copu_retry_limit on 10
I connected the grounding all together
Replaced the chip

Still I have a brownout, I read is an issue with esp32 chips without the E. In the new once they solved that problem.

About the power saving I found some adititional settings

First I tested with WiFi.settxpower, but it damaged the chip. I couldn't connect to the internet anymore, soo I replaced it.

These are the additional settings that work for the url player. But I have a program that reboots an switches form url player to others soo I want to keep this in my shetch. And phill could be setted in the config file. Because I don't want when I am using A2DP.

```
    //esp_wifi_set_ps(WIFI_PS_NONE);
    WiFi.setSleep(false);
    //WiFi.enableLongRange(true);// This damaged the chip
    //WiFi.setTxPower(WIFI_POWER_13dBm);/// This damaged the chip

    esp_bluedroid_disable();
    esp_bluedroid_deinit();
    esp_bt_controller_disable();
    esp_bt_controller_deinit();
```

The setSleep controls also the modem sleep when only in STA mode.

But I notest somthing strang. When I closed the door where the esp is it stops outputting oudio, is it the inputs or output stream I don't know. What I know is for that far esp I have an rssi -67. The close one has an rssi of -53. 

Now that been know... I have a larger antenna +-20cm that is corresponding with the maximum db of datasheet for the esp I have  Time to test it... The RSSI is now: -72 (closer to 0 the beter) So we diden't improve according to the rssi :-/
I m still going to test it if it runs more like it suppose to be.

podaen, 2022-04-28:
If I close my door it keeps outputting audio. But sometime it that also before. 

I am going doing stupid now... I m going to try now without antenna.

podaen, 2022-04-28:
Without antenna it doesn't connect. Maybe it was because the antenna could set strait. Anyway I replaced the short antenna and now I have -62.

podaen, 2022-04-28:
When I am connecting to server directly I get the message "Your browser is not the best, try another one". So my question is now, can we use oter user agents than the default?

pschatzmann, 2022-04-28:
I don't think that you want to do that and you risk to receive some protocol that we can't understand....

podaen, 2022-04-28:
> This is very strange: Copy is providing the info that the write to EncodedStream has been processed. In the write we call the encoder in order to determine this result. But after the issue the encoder is not called any more. This is somehow not possible!

Maybe we are bombarded by photons of the solar eruption. It is know that the sun is more active now. This could explane this strange phenomena... it could explane a few cases but not all. With the 1.0.6 I had some break at unexpected moments too.

podaen, 2022-04-29:
I did the test. Even changing to N, set other channel with a higher bandwide, disabling the ble... the ping did get shorter with  power saving to false.

> //Pingen naar 192.168.1.178 met 512 bytes aan gegevens :
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 39 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 10 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 11 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 7 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 8 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 10 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 4 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 5 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 6 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 9 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 8 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 6 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 6 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 6 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 8 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 11 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 6 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 6 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 8 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 6 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 13 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 13 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 10 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 6 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 7 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 10 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 12 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 15 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 8 ms TTL = 255
> //Antwoord van 192.168.1.178 : bytes = 512 tijd = 14 ms TTL = 255

> Buff Size: 512	Time: 7142	Bytes: 1048576	Throughput :	1168	kbps1168
> Buff Size: 512	Time: 8244	Bytes: 1048576	Throughput :	1016	kbps1016
> Buff Size: 512	Time: 6870	Bytes: 1048576	Throughput :	1216	kbps1216
> Buff Size: 512	Time: 6739	Bytes: 1048576	Throughput :	1240	kbps1240
> Buff Size: 512	Time: 6360	Bytes: 1048576	Throughput :	1312	kbps1312
> Buff Size: 512	Time: 6800	Bytes: 1048576	Throughput :	1232	kbps1232
> Buff Size: 512	Time: 6645	Bytes: 1048576	Throughput :	1256	kbps1256
> Buff Size: 512	Time: 7924	Bytes: 1048576	Throughput :	1056	kbps1056
> Buff Size: 512	Time: 6241	Bytes: 1048576	Throughput :	1344	kbps1344
> Buff Size: 512	Time: 6324	Bytes: 1048576	Throughput :	1320	kbps1320
> Buff Size: 512	Time: 7214	Bytes: 1048576	Throughput :	1160	kbps1160
> Buff Size: 512	Time: 7879	Bytes: 1048576	Throughput :	1064	kbps1064
> Buff Size: 512	Time: 6289	Bytes: 1048576	Throughput :	1328	kbps1328
> Buff Size: 512	Time: 6145	Bytes: 1048576	Throughput :	1360	kbps1360
> Buff Size: 512	Time: 6253	Bytes: 1048576	Throughput :	1336	kbps1336
> Buff Size: 512	Time: 6421	Bytes: 1048576	Throughput :	1304	kbps1304
> Buff Size: 512	Time: 6192	Bytes: 1048576	Throughput :	1352	kbps1352
> Buff Size: 512	Time: 6119	Bytes: 1048576	Throughput :	1368	kbps1368
> Buff Size: 512	Time: 6440	Bytes: 1048576	Throughput :	1296	kbps1296

DrJaymz, 2022-04-29:
My audio stops sometimes permanently and yet the metadata carries on.  If I reset the esp32 using its reset button it connects and metadata but no audio.

In my case I think that the I2S receiver borks on a packet.  That is why when using the reset pin on the esp32 which obviously doesn't reset the I2S device it doesn't recover. Its perhaps layout issue since the chip I'm using is a class D amp it probably radiates like hell.

DrJaymz, 2022-04-29:
I have stuck the scope on it and I'll see what is happening when it stops.  I think the clock carries on but I'll see what the data does.

DrJaymz, 2022-04-29:
It is probably a memory issue.  Its unlikely to be an espressif issue per-se.

I would say its either a memory leak or heap fragmentation.
I'm not that experienced on the esp32 arduino framework to dig into the internals, but I id run a few esp* commands to get the largest free memory block and whilst it doesn't seem to run out it does fluctuate all the time and I wondered if all the audio buffers and such were static or whether there is memory that keeps being created.  Arduino struggles with that only because its memory is so limited.

pschatzmann, 2022-04-29:
My gut feeling is also that this might be related to some memory corruption. 
Unfortunately these things are hard to find.

DrJaymz, 2022-04-29:
yes, you might run an arduino project for years without an issue but start streaming and soon megabytes are passing through it and rare issues become a bit more pronounced.  Is there any convenient monitoring functions for or stats for the player?  Like frames dropped or something like that?  You could probably have a panic and restart streaming so quick that the listening doesn't really notice.

DrJaymz, 2022-04-29:
Spoke to soon... Its just borked.

[image]

DrJaymz, 2022-04-29:
data has stopped but still got output clock.

pschatzmann, 2022-04-29:
The weather forecast for this week-end is bad. So I plan to have a closer look...

DrJaymz, 2022-04-29:
Luckily for me its not too bad here - so go flying!

podaen, 2022-04-29:
The are not yet flying than the did before the corona.

[40 more replies]

closed 2022-05-13 by pschatzmann