pschatzmann/arduino-audio-tools #183: Trying the measuring stream example

podaen, 2022-04-26:
[I canet f](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-communication/communication-ip-receive_measure)

I want to try it, but I can't find MeasuringStream out;

Thanks

pschatzmann, 2022-04-26:
Just wrap your final input or output in this class and use it instead. Example: https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-communication/communication-ip-send/communication-ip-send.ino
ps. The log level needs to be on info to see the result in the log.

podaen, 2022-04-26:
it is not for measuring an http link?

pschatzmann, 2022-04-26:
Obviously you did not process any data yet, since the log states that it's waiting for data.
If you want to look at the data you can use a HexDumpStream as output.

podaen, 2022-04-26:
yep, see now some data

podaen, 2022-04-26:
Now, how can I found the beginning of the frame?

podaen, 2022-04-26:
Soo, I was right,

for this link

url.begin("http://22183.live.streamtheworld.com/JOE.mp3", "audio/mp3");

the input stops

podaen, 2022-04-26:
Now I am going to config it and see if data still coming in. Seems a good strategie.

podaen, 2022-04-26:
what could I do to change some settings of the input audio?

pschatzmann, 2022-04-26:
https://github.com/pschatzmann/arduino-audio-tools/wiki/Performance

podaen, 2022-04-27:
This maybe also important...

>   long rssi = WiFi.RSSI();
>   Serial.print("Signal strength (RSSI):");
>   Serial.print(rssi);
>   Serial.println(" dBm");

pschatzmann, 2022-04-27:
I think the measuring stream question has been answered. 
In order to avoid any confusion I am closing this issue...

closed 2022-04-27 by pschatzmann