pschatzmann/arduino-audio-tools #184: ICYStream requires wifi username and password

DrJaymz, 2022-04-27:
There appears to be no way to use ICY streaming without passing Wi-Fi credentials in.

Since its always preferable to not hard code things, one prefers to use a Wi-Fi manager or some other configuration tool which means we'd need to separate Wi-Fi from audio tools.

podaen, 2022-04-27:
It seems related to wifimanager. I have ICY and I setup my own internet connection, without using...

`AudioSourceIcyUrl Urlsource(urlStream, urls, "audio/mp3");`

DrJaymz, 2022-04-27:
what is AudioSourceIcyUrl  ?

podaen, 2022-04-27:
Forgot this, you have to create a local file AudioSourceIcyUrl.h and include in to your code.

```
#pragma once
#include "AudioTools.h"

namespace audio_tools {

/**
 * @brief URLStream which provides the ICY Http Parameters
 * 
 */
class AudioSourceIcyUrl : public AudioSourceURL {
    public:
        template<typename T, size_t N>
        AudioSourceIcyUrl(URLStream& urlStream, T(&urlArray)[N], const char* mime, int start=0)
        : AudioSourceURL(urlStream, urlArray, mime,start) {
        }

        const char *icyValue(const char* name) {
            return actual_stream->httpRequest().reply().get(name);
        }
        const char *icyName() {
            return icyValue("icy-name");
        }
        const char *icyDescription() {
            return icyValue("icy-description");
        }
        const char *icyGenre() {
            return icyValue("icy-genre");
        }

        /// Returns the last section of a url: https://22323.live.streamtheworld.com/TOPRETRO.mp3 gives TOPRETRO.mp3
        const char *urlName(){
            const char* result = "";
            Str tmpStr(toStr());
            int pos = tmpStr.lastIndexOf("/");
            if (pos>0){
                result = toStr()+pos+1;
            } 
            return result;
        }

        /// Returns the icy name if available otherwise we use our custom logic
        const char* name() {
            Str result(icyName());
            if (result.isEmpty()){
                result.set(urlName());
            }
            return result.c_str();
        }
};

}
```

DrJaymz, 2022-04-27:
I think this is basically what I was asking for at the start.
ultimately, what I was looking for was the metadata from the stream.

In its simplest form, I just wanted to create an internet radio receiver and display it on the screen. I know that it works with the SPI screen because I tested that.
The URLstream works without Wi-Fi credentials but doesn't have the metadata (and it stops playing after a bit completely, memory leak?) 
The ICYstream is linked to Wi-Fi.

Both of them are marginal, in terms of, they don't quite have enough streaming buffer.  Thats a hardware limitation.  Hopefully we improved that by reducing the Wi-Fi latency by disabling power saving.

podaen, 2022-04-27:
Yeah, but you don't get far whit AudioSourceIcyUrl, if you want to select a station.

I do it like this

url[select]
stationname[select]

just make a list a list of every name according to your url.

> Both of them are marginal, in terms of, they don't quite have enough streaming buffer. Thats a hardware limitation. Hopefully we improved that by reducing the Wi-Fi latency by disabling power saving.

Diden't try it yet. I see also some breaks in the input stream, but for now I am looking in to get not dropt out of the server. Maybe updating the request header (if possible with esp) could also improve the connection speed...

podaen, 2022-04-27:
In urlstream.h you can find 

bool is_power_save = true;//false

I did try it, but it diden't work for me.

pschatzmann, 2022-04-27:
I aligned the constructors of ICYstream to correspond to the ones available by URLStream.

pschatzmann, 2022-04-27:
> In urlstream.h you can find
> bool is_power_save = true;//false

I don't understand. Why would you want to make it slow again ?

podaen, 2022-04-27:
Sorry about that, my mistake.

DrJaymz, 2022-04-28:
> I aligned the constructors of ICYstream to correspond to the ones available by URLStream.

Did you commit the changes?

pschatzmann, 2022-04-28:
Oh, I just noticed that I forgot 1 file...It should be committed now!

DrJaymz, 2022-04-29:
I have this working just fine.

closed 2022-04-29 by DrJaymz