pschatzmann/arduino-audio-tools #192: URLStream truncating the first few seconds of the mp3 file

AnonymousChatter, 2022-05-03:
Hello Phil,

Thank you for the wonderful library, it works brilliantly on the ESP32-C3.

I am currently trying out one of the example project: "streams-url_mp3-i2s". 
The only changes done to the sketch are:
1. Pin-out for the i2S, its sample rate & number of channels (16000 & 1 corresponding to the file).
2. WIFI credentials
2. URL which points to a static mp3 file on my web server.

The  mp3 playback is missing the first few seconds when it starts playing (about 5~6 seconds). 
Do you have any suggestion on code workaround in the sketch? Currently I am just thinking of appending few seconds of silent at the beginning of the file.

Thanks again...

pschatzmann, 2022-05-04:
Hmm, that's strange. This is the first time I hear that something gets list in the beginning.
I suggest to try with a WAV file, because this file format is so simple that there is a low risk that something can go wrong on the software side.

For the mp3 - did you try with different files ? 
You can also check if removing the metadata makes a difference

TomatoCube18, 2022-05-04:
Thanks for the answer, and totally agree with your suggestion of switching over to Wav file to reduce the "complexity" :)

But since this is just a fun side project for me (with no time concern), I think it would be a shame to quit when it is so close to perfection. 

I go with your suggestion and re-encode the mp3 with different parameters and even trying out different software encoders. In the end, it is still missing a chunk of sound data. Of course when it is encoded at a lower bit rate the duration (that is missing) is far more obvious. 
Then it occurs to me that maybe the decoder is not ready when the internet data is starting to come in?  (Not sure if I am making any sense) I think I did read about the use of RTOS in this library in one of the thread...

So I did a quick experiment. Instead of 
1) decoder.begin() 2) URLStream.begin() 3) Enter Arduino Loop() 4) Copier.copy() 
I change the ordering to 
1) URLStream.begin() 2) Copier.copy() 3) decoder.begin() 4) Entering Arduino Loop() 5) Copier.copy()

Seems to have fixed my problem... 

Here is a bit of the code snippet in case my explanation is hard to follow (Sorry).

// - The Original Location -
//  dec.setNotifyAudioChange(i2s);
//  dec.begin();

// mp3 radio
  urlStream.begin(url,"audio/mp3");
  copier.copy();

// - The New Location -
  dec.setNotifyAudioChange(i2s);
  dec.begin();
}

void loop(){
    copier.copy();
}

pschatzmann, 2022-05-04:
Thanks for your input. Maybe another thought: did you check the initial sampling rate and channels of I2S. If this is changed by the mp3 decoder I could think that something like this happens...

AnonymousChatter, 2022-05-04:
Oh .... not too sure if it is a concern. I am running the code on the ESP32-C3. instead of the Original ESP32 or ESP8266.
RiscV architecture....

pschatzmann, 2022-05-04:
It would be really interesting to know if this only an issue with MP3DecoderHelix.  
Therefore the suggestion to double check with a WAV file.
If this also happening with a WAV file, I tend to think that it might be related to the C3. I used short MP3 files in my [arduino-simple-tts](https://github.com/pschatzmann/arduino-simple-tts) and never had any issues with the beginning ...

AnonymousChatter, 2022-05-04:
This is what I have done...
1. I extracted both the knghtsng_wav & alice_wav into a WAV file and store it on the web server.
2. Start the sketch from "streams-url_wav-i2S.ino" example
3. Change the i2S pins & change the sample rate 11025 (knghtsng_wav) & 8000 (alice_wav)
4. Change bits_per_sample to 16

C3 suffers really badly for knghtsng_wav (choppy playback). But seem to do okay for alice_wav.

I can't really tell if there is any missing audio at the beginning of the playback. (not scientific, just purely human hearing)
But then again missing (truncating) one chunk of WAV data at the beginning might not be as noticeable as missing a chunk of mp3 data due to its raw Data size? 

========

On another note....

I extracted the mp3 from your header file  in the "streams-memory_mp3_short-i2s-2" example and place the zero.mp3  in the web server.
It plays back perfectly!! I am really at a lost now...
I think we could conclude that something is wrong with my encoded mp3s (no matter what parameters that I put, I even try to re-encode my mp3 with the same parameters as zero.mp3)?

AnonymousChatter, 2022-05-04:
Hello Phil,

I think I finally got it!! Thank you so much for your help.
I try to re-encode the mp3 this round using Audacity + FFMpeg. It is finally working.

Maybe this might be helpful to other readers encountering the same problem in the future.
Adobe Audition, NCH WavePad, Permute 3 encoded mp3s lead to a few missing seconds of audio at the start.

closed 2022-05-05 by AnonymousChatter