pschatzmann/arduino-audio-tools #201: Installation on PlatformIO

podaen, 2022-05-09:
Can't find The library from PIO Home -> Libraries

[image]

pschatzmann, 2022-05-09:
Which library. I have the impression that you did not read the Wiki: https://github.com/pschatzmann/arduino-audio-tools/wiki/Working-with-PlatformIO

podaen, 2022-05-09:
I did get the msg that the library was not registered yet in the platformIO. That why I am telling you. It's very easy to install like that. I missed it in the readme, but now I see its there and I was looking for IDE title with in it Arduino, PlatformIO... Maybe it is because I'm really bizzy to get all the setting right for PlatformIO.

closed 2022-05-09 by podaen