pschatzmann/arduino-audio-tools #207: Complilation error

Chirawat, 2022-05-15:
I tested communication-generator-rtsp.ino and got an error. I already checked platglue.h seems nothing need to fix.
I want to ask you about Arduino version are you using? My version is 1.8.19. 

Here is error message.

`In file included from C:\Program Files (x86)\Arduino\libraries\Micro-RTSP-Audio\src/AudioStreamer.h:14:0,
                 from C:\Program Files (x86)\Arduino\libraries\arduino-audio-tools\src/AudioLibs/MicroRTSPAudio.h:3,
                 from C:\Program Files (x86)\Arduino\libraries\arduino-audio-tools\examples\examples-communication\communication-generator-rtsp\communication-generator-rtsp.ino:14:
C:\Program Files (x86)\Arduino\libraries\Micro-RTSP-Audio\src/platglue.h:1:1: error: expected unqualified-id before '.' token
 ../include/platglue.h
 ^
exit status 1
Error compiling for board ESP32 Dev Module.
`

pschatzmann, 2022-05-15:
platglue.h in src is a symbolic link. On OS/X and Linux this is working fine.
Maybe Windows makes something strange here: You can try to replace the symbolic links with their corresponding files from the include directory...

Chirawat, 2022-05-16:
> I assume that you have installed the [Micro-RTSP-Audio](https://github.com/pschatzmann/Micro-RTSP-Audio) project. platglue.h in src is a symbolic link. On OS/X and Linux this is working fine. Maybe Windows makes something strange here: You can try to replace the symbolic links with their corresponding files from the include directory...

I worked, becuase of Windows does not support this. I changed file name (platglue.h to _platglue.h) now done compiling...

Thanks

closed 2022-05-16 by Chirawat