pschatzmann/arduino-audio-tools #208: URL Streams Performence

podaen, 2022-05-15:
When a stream starts to failter, you would like to know what the performence is of a given stream. So a library has been created with an example how to staf. The result can be valuated. I have mad this library to measure the time between two buffers been filled and to be empty. This will give you an idea how quick host respons according to the demand.

[Stream Analyser](https://github.com/podaen/Arduino-UrlStream_Analyser)

pschatzmann, 2022-05-15:
I am not sure if this makes a lot of sense:  I would expect that if you connect to a streaming service you might get some buffered data quickly and then if the buffer is used up, you get the streaming speed which is necessary to play the audio, or the service might even block you completely because you were downloading too much the data too quickly and not with the expected streaming speed. 

You could use your tests with a file based url however.

podaen, 2022-05-15:
I have seem different. It depends from where you are connecting. You could get a very log data gettering in once. This test case make me clear how to handle the reads between and I think it has has been learning full

pschatzmann, 2022-05-15:
Do you have any example URLs which are rendered too slow to play and the result is broken up audio ?

podaen, 2022-05-15:
Analyzing the output has never been the upset. Understanding how it works and get and orrientated view what could be espected and fine tune the stream. Out of my tests I m positive and not thinking anymore that some streams just are  ad and can't be handled.

podaen, 2022-05-15:
Bordering, by check the availibility to much slows the  receives down.

> I was getting 32000 to 37000 bytes per second:

Intresting, this means that arround an average of 34bytes are received at high speed. To get a full frame of 512bytes it will take 15ms in idial circumstances... No need to check it faster.

podaen, 2022-05-15:
I did the test, the maximum goos down but the average stays more or less the same.

podaen, 2022-05-16:
You want to take some advice?

pschatzmann, 2022-05-16:
Sure

podaen, 2022-05-16:
Handle your own bufferfill.

pschatzmann, 2022-05-16:
I think that's what I am doing. If you see the copy log it is using the range that you specified in the copier - and if nothing is specified it uses DEFAULT_BUFFER_SIZE bytes

podaen, 2022-05-16:
What does one package contain?

pschatzmann, 2022-05-17:
https://docs.fileformat.com/audio/mp3/

podaen, 2022-05-17:
I think I all said it. Just fill in the buffer.

closed 2022-05-20 by podaen