pschatzmann/arduino-audio-tools #213: Low volume on ESP32-Audio-Kit headphone output

jwktje, 2022-05-21:
First of all, much respect for this library. Really impressed by everything it can do.

I'm running a sketch to play a WAV from an SD card and it works well. But the volume is very soft no matter what I change in the configuration. What am I doing wrong?

I have the ESP32-Audio-Kit V2.2 with the chip version ESP32-A1S 2974 and have set this in AudioKitSettings.h:
`#define AUDIOKIT_BOARD 5`

Here is my sketch:

```
#include "WiFi.h"

#define USE_SDFAT
#include "AudioTools.h"
#include "AudioLibs/AudioKit.h"

const char *startFilePath="/";
const char* ext="wav";
int speedMz = 10;
AudioSourceSdFat source(startFilePath, ext, PIN_AUDIO_KIT_SD_CARD_CS, speedMz);
AudioKitStream kit;
WAVDecoder decoder;
AudioPlayer player(source, kit, decoder);

void setup() {
  Serial.begin(115200);
  Serial.println("Boot");

  AudioLogger::instance().begin(Serial, AudioLogger::Warning);

  // setup output
  auto cfg = kit.defaultConfig(TX_MODE);
  cfg.sample_rate = 48000;
  kit.setVolume(100);
  kit.begin(cfg);
  // setup player
  player.setVolume(1);
  player.begin();
}

void loop()
{
  player.copy();
  kit.processActions();
}
```

pschatzmann, 2022-05-21:
As far as I can see you use 
 - kit.setVolume(100); // full volume in the kit
 - player.setVolume(1);  // almost no volume 

I would increase the volume the player is providing to the kit...

jwktje, 2022-05-21:
Hey @pschatzmann 
Thanks for the quick response. Setting the AI_THINKER_ES8388_VOLUME_HACK absolutely increased the volume a bunch. I'm not getting any distortion either at max volume so that seems to be fine.

Just wondering. On my headphones (instead of my earbuds) the volume could still be slightly higher ideally. How would I go about this? Is an extra IC as a Headphone AMP the only option?

pschatzmann, 2022-05-25:
I am not sure. The AI_THINKER_ES8388_VOLUME_HACK logic is not from me, so I suggest that you double check the implementation and compare it with the [ES8388 DataSheet](https://datasheet.lcsc.com/lcsc/1912111437_Everest-semi-Everest-Semiconductor-ES8388_C365736.pdf)

jwktje, 2022-05-26:
@pschatzmann I'm troubleshooting it now. I'm mostly confused as to why the volume between 0 and 100 gets divided by 3 before writing the registers. Do you know why that is?

pschatzmann, 2022-05-26:
No, but I noticed that this is already done line this in the original implementation from Espressif

jwktje, 2022-05-26:
@pschatzmann I just found that too. And I can't find a logical reason in the memory map of the ES8388 so I probably lack the expertise to figure it out.

But whatever the reason, if Espressif does it this way in their driver then ait would be safe to assume that (for some reason) the range of volume goes from 0 to 33, and that this is the absolute max.

So without using additional amps, this is probably the most volume I'm gonna get. If anyone has any tips on how to crack more volume out of it, I'd love to hear. Otherwise I should close this issue.

pschatzmann, 2022-06-26:
Please note that both the player and the AudioKit have their separate independent volume control.
To get max volume you would set
- player.setVolume(1.0);
- kit.setVolume(1.0);

closed 2022-06-26 by pschatzmann

funkyfourier, 2023-04-18:
Could this also have something to do with `cfg.channels`? With my mono wav files, the volume from the headphone jack was boosted considerably when setting `cfg.channels = 1` as opposed to the default setting. Also `kit.setVolume(1.0);` had to be set to get the extra boost.