pschatzmann/arduino-audio-tools #226: Audiokit SD card sound

oliver1427, 2022-06-06:
Hello! I just received a hackerbox with the esp32 audio dev kit. I managed to overcome a lot of hurdles and eventually got the mp3 stream working. After that, I copied around like 100 - 200 animal crossing: new leaf mp3s to a micro SD card to test with the SD card player feature. I got it to compile and play, but it sounds like someone’s playing you the song to you through a telephone.

Best regards,
                      Oliver
https://user-images.githubusercontent.com/91022217/172086889-30f61061-5be8-4ebf-919b-278f684f4427.MOV

[image]
[image]
[image]

oliver1427, 2022-06-06:
Check the video link for sound.

pschatzmann, 2022-06-06:
I never heard this kind of problem before. 
I wild guess: the board might not get enough power and is is therefore breaking up. 

- Did you test your set up with a sine tone ? (this might use less power)
- Did you try to power via USB cable ?

oliver1427, 2022-06-06:
I tested it with the sine tone, and it sounds fine, same with the mp3 streams. Sounds the same too on usb power, and on earphones. Four ohm three watt speakers. I think I tried setting it to warning and info with no effect.

pschatzmann, 2022-06-06:
If you set the log level to info you should be able to identify if there are some delays when the breakups occur.
It seems that your issue is related to the SD drive somehow. I would suggest that you concentrate your tests on the reading of files.

oliver1427, 2022-06-06:
Update, got it working, but only with one coconut mall mp3.

oliver1427, 2022-06-06:
I tried a mp3 stream of a animal crossing mp3, but got the same result

oliver1427, 2022-06-06:
I think my audio kit hates animal crossing

oliver1427, 2022-06-06:
Could it be because it’s copywrited?

oliver1427, 2022-06-06:
Thanks. Animal crossing is like a crafting kind of game which I tend just to like the music from. I will try the other decoder.

oliver1427, 2022-06-06:
I think I have concluded that I was just using crappy mp3s. Thank you for all the help!