pschatzmann/arduino-audio-tools #229: Question on A2DPStream.begin() Blocking Behaviour

chrissbarr, 2022-06-11:
I'm currently using your ESP32_A2DP library for a project that does A2DP -> I2S on an ESP32. It's working well for me, but I'd like to bring another source into the picture and also do some processing on the signal, so I have been looking to use this audio tools library.

I'm looking to move to use the A2DPStream class, as I'd like to use A2DP in conjunction with the InputMixer and MultiOutput classes. My reading is I need to use the A2DPStream (rather than the BluetoothA2DPSink) to do this.

However, trying to use the A2DPStream as a replacement for the BluetoothA2DPSink, I notice the A2DPStream.begin() function blocks until a device is connected, as can be seen here:

https://github.com/pschatzmann/arduino-audio-tools/blob/e9f3666f7d12417de7eef3dc4266f06ddb1b1506/src/AudioLibs/AudioA2DP.h#L166

Is there any way of running the A2DPStream without this blocking? Can A2DPStream be configured to output an 'empty' stream if a device is not connected?

Not directly relevant to my question, but the setup I am hoping to work towards would look something like:
```
                        A2DPStream =>|                          |=> I2S
                                     | InputMixer=>MultiOutput=>|
MemoryStream => DecodedAudioStream =>|                          | AudioRealFFT
```
Does this seem feasible, or am I missing something that makes this unworkable?

pschatzmann, 2022-06-11:
I would suggest to use https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-basic-api/basic-a2dp-i2s/basic-a2dp-i2s.ino as a starting point.

I think Instead of an InputMixer you need to use an OutputMixer: In the callback you write to the OutputMixer and you copy from the DecodedAudioStream to the OutputMixer.

If you want to copy from the DecodedAudioStream to the OutputMixer even when no callbacks are called (yet) you can manage this with a separate state variable and do this copy in the loop - but you stop doing this as soon as the callback logic is working...

Denis-2021, 2022-06-15:
> I'm currently using your ESP32_A2DP library for a project that does A2DP -> I2S on an ESP32. It's working well for me, but I'd like to bring another source into the picture and also do some [...]

Have you been able to solve the problem that A2DPStream.begin() function blocks until a device is connected?

I am trying to switch between 2 outputs once I want to output to a bluetooth speaker via A2DP and than switch the sound to an I2S device.
My problem is that when the stream switches from A2DP to I2S it works for couple of seconds and than the A2DP trying to reconnect and blocks everything.

pschatzmann, 2022-06-15:
Same answer here: A2DPStream is designed to block so don't use it if this is disturbing and use the basic API instread!

chrissbarr, 2022-06-26:
Thanks for the replies @pschatzmann. The approach you suggest seems reasonable enough. I haven't given this a go yet, might be a few weeks until I can get the chance. Feel free to close this issue if you want.

closed 2022-06-26 by pschatzmann