pschatzmann/arduino-audio-tools #230: Image too big - Global variables

arikfe, 2022-06-12:
Trying to update the basic SD card reader to A2DP sink example described here [https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-player/player-sd-a2dp](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-player/player-sd-a2dp) to use a UrlStream as a source  instead of SD card
I have tried to compile and run the sin wave example from the ESP32-A2DP [https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-generator-a2dp/streams-generator-a2dp.ino](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-generator-a2dp/streams-generator-a2dp.ino) and it worked great ( or horribly wrong for my ears ).
Also I have been seccusfull in compiling the basic player-sd-a2dp example by I have only a SD card reader from an ili9341 board and it did not work yet.
I would like to stream from network to bluetooth speaker using my ESP32.
I'm using ESP32 mh-et live mini devkit with 4MB flash size.
```
#include "AudioTools.h"
#include "AudioLibs/AudioA2DP.h"
#include "AudioCodecs/CodecMP3Helix.h"

A2DPStream out = A2DPStream::instance();  // A2DP input - A2DPStream is a singleton!
MP3DecoderHelix decoder;

const char *urls[] = {
  "http://192.168.1.xxx:pppp/sound.mp3"
};
const char *wifi = "my-wifi ssid";
const char *password = "my-wifi-password";
URLStream urlStream(wifi, password);
AudioSourceURL source(urlStream, urls, "audio/mp3");
AudioPlayer player(source, out, decoder);
void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Warning);

  // setup output - We send the test signal via A2DP - so we conect to the "LEXON MINO L" Bluetooth Speaker
  auto cfg = out.defaultConfig(TX_MODE);
  cfg.name = "BlueTooth speaer";
  //cfg.auto_reconnect = true;  // if this is use we just quickly connect to the last device ignoring cfg.name
  out.begin(cfg);

  // setup player
  player.setVolume(0.1);
  player.begin();

}

```

I'm getting this error 
```
Sketch uses 1745482 bytes (133%) of program storage space. Maximum is 1310720 bytes.
Global variables use 60072 bytes (18%) of dynamic memory, leaving 267608 bytes for local variables. Maximum is 327680 bytes.
text section exceeds available space in board
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
Error compiling for board MH ET LIVE ESP32MiniKit.
```

pschatzmann, 2022-06-12:
Just use -> Tools -> Partition Size -> Huge App and the error should go away...

arikfe, 2022-06-12:
Thank you for the quick response.
From the issue it is written that it is mainly a memory problem.
I'll check it out later.

closed 2022-06-22 by pschatzmann