pschatzmann/arduino-audio-tools #231: [I2S] No output produced in Slave mode

GuillaumeLeclerc, 2022-06-12:
Hello,

I'm trying out the I2S sine wave example and I can't get the slave mode to work. I'm debugging the output with a logic analyzer and this is what I get:

### Master mode
[image]

### Slave Mode (BCLK and LRCLK are external)

[image]

The only thing I changed between the two is going from master -> slave, and connecting BCLK and LRCLK.

Any idea what could be going on ?

pschatzmann, 2022-06-12:
I don't really have any experience with this, but I suggest to compare the frequencies of the master clock between the 2 devices and if they do not match set it with cfg.fixed_mclk = freq to make them match

pschatzmann, 2023-04-26:
Closed due to inactivity

closed 2023-04-26 by pschatzmann