pschatzmann/arduino-audio-tools #232: one esp32 master and multiple esp32 slave audio receiver

aqildad-create, 2022-06-13:
i have successfully test this     ES32 1 RAW AUDIO DATA SEND TO A2DP  --> ESP32 2 RECEAVE DATA AND SEND TO I2S SPEAKER.
but now
I WANT TO HAVE MULTI ESP32 RECEAVER AND EACH WILL SEND DATA TO I2S SPEAKER ,

pls guide

pschatzmann, 2022-06-13:
I am not really sure what you mean: but in a nutshell it is only possible at any point of time to connect __one source to one sink__. If you want to use multiple devices you need to close the connection before you can reconnect with another source or the same source with another sink.

aqildad-create, 2022-06-14:
i want to connect **3 esp32 as BLUE TOOTH SINK** with **one SOURCE-esp32** . the ESP32 source will transmit audio and all three esp32 SINK will receive and play on their i2s speakers at same period of time . how is this possible ?

pschatzmann, 2022-06-14:
As I explained to you this is not possible! The ESP32 A2DP API does not provide this functionality and I even think the Bluetooth A2DP standard does not support this!