pschatzmann/arduino-audio-tools #233: audio is clear but keeps muting and unmuting again every second

RitikDutta, 2022-06-14:
hey, first of all thanks for the wonderful library, it was impossible for me to work on audio kit board via ESP-ADF tools, ur library saved my money on esp audio kit..

every thing working fine my board is esp32 a1s 2974. 
i have tried ur example sd mp3 player. but the voice was too low, i did audio hack to 1 even then audio is low.
finally i got idea and wanted to use the headphone jack from esp-audio kit to my external sound amplifier. it solved my problem the sound was clear and loud but, another problem arise,
the audio keeps muting and un muting every second. i tried to use LR pins from audio kit it solves the muting problem but the audio becomes very glitchy. 

(setting audio hack back to 1 gives no output from headphone jack)

realy appreaciate the library and your help.

pschatzmann, 2022-06-14:
I suggest to try to comment out the kit.processActions(); in the loop()
I assume that is some pin conflict. Unfortunately I don't have any detailed information about 2974 boards

pschatzmann, 2022-06-14:
Good point: the approach for buffer underruns would be to increase the i2s buffer_size * buffer_count. Or to decrease the I2S sampling_rate slightly...

RitikDutta, 2022-06-15:
@pschatzmann commenting out kit.processActions(); in the loop() solved the problem. thanks for the quick solution
now the voice is loud and clear.

@joba-1 thanks for looking into

closed 2022-06-15 by RitikDutta