pschatzmann/arduino-audio-tools #234: ESP32-A1S SD card on custom board doesn't work

jwktje, 2022-06-14:
Hey Phil,

I'm really loving the library so far. I made a custom PCB with the ESP32-A1S after working with the AudioKit succesfully. 
I wired up the SD like mentioned [here](https://github.com/espressif/arduino-esp32/tree/master/libraries/SD) which worked fine in other projects. However I can't get it to work with the arduino-audio-tools library.

Where can I define the pins that are used for the SD card so that my current wiring works?

pschatzmann, 2022-06-14:
I would suggest to stick with the same pins that the AudioKit and Lyrat are using: This gives you the advantage that you can also use sd_mmc easily.

The logic which is defining the pins is defined in https://github.com/pschatzmann/arduino-audiokit/blob/main/src/AudioKitHAL.h: just search for p_spi->begin(PIN_AUDIO_KIT_SD_CARD_CLK, PIN_AUDIO_KIT_SD_CARD_MISO, PIN_AUDIO_KIT_SD_CARD_MOSI, PIN_AUDIO_KIT_SD_CARD_CS);   with your pins after th

pschatzmann, 2022-06-14:
You would add your logic after   kit.begin(cfg); 
and it would be   SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI, SD_CS);

closed 2022-06-14 by jwktje

jwktje, 2022-07-22:
@pschatzmann I'm also still working on this issue. I can only get 128kbps MP3's to run without glitching. 
Even when using an `AudioConfigLocal.h` with this content; 

```
#pragma once
#define DEFAULT_BUFFER_SIZE 1024 // 2048
#define DEFAULT_SAMPLE_RATE 44100
#define I2S_BUFFER_SIZE 512
#define I2S_BUFFER_COUNT 5 // 20
```

What can I do to improve this? Ideally I'd want to play 256kbps or higher. Is that possible? Or should I switch to WAV? What would get me the best performance?

pschatzmann, 2022-07-23:
The AudioPlayer plays all files in all subdirectories. You can select a specific directory. To select one or more individual files you can call setFileFilter() on the AudioSource.

I am confused regarding your MP3. I am usually using mp3 with a sample rate of 44.1kHz which gives a bitrate of 1.4Mbps w/o any issues.
Did you set the log level to Warning ? 
Have your tried an alternative MP3 Decoder e.g. LibMAD ?

jwktje, 2022-07-24:
I haven't tried LibMAD but I did increase both the buffer count and size to no effect. On every of my 3 projects that use your library the mp3 performance is glitchy on anything above 128kbps. They all use the ESP32-A1S and the MP3DecoderHelix. Could the Helix decoder be the bottleneck? WAV works without issue. 

In general should WAV give better performance than MP3 or shouldn't that make a difference?

pschatzmann, 2022-07-25:
WAV is stored in PCM format, so the processor doesn't need to do any decoding but can just copy the audio data from the source to the destination. If you use any compressed format the processor needs to do some decoding which requires a lot of calculations. 

I rarely created any MP3 files myself and I never had any issues with the mp3 internet streams. 
The few MP3 files I created in Audicity are by default with Medium Quality (145-185kbps) and this is working just fine.
 
Can you maybe share your example ?

jwktje, 2022-08-20:
Hey Phil. I was on holiday without computer hence the late response.

I also created my MP3's with Audacity but on a higher bitrate (256kbps and up). 
The lower ones (below 192kbps) all worked fine. 
But it's good to know that WAV is less CPU intensive. I can easily spare the space on the SD card so I will go with WAV files. 

Thanks for confirming my suspicion that WAV is simpler to use for the MCU.