pschatzmann/arduino-audio-tools #237: Thoughts on FilteredStream/ConverterNChannel and FIR for SDRs

thaaraak, 2022-06-17:
I am building a little SDR - the FIR filters are working out fantastic but one of the needs in an SDR is to:

1) Phase shift left and right channel by 90 degrees (this works fine)
2) Then either add or subtract left from/to right (this basically removes the unwanted sideband)

Point 2) needs to be in the API as typically you want to be able to programmatically switch between the two.

Now I have hacked ConverterNChannels to do the above

```
  size_t convert(uint8_t *src, size_t size) {
    int count = size / channels / sizeof(T);
    T *sample = (T *)src;

    T left,right;

    for (size_t j = 0; j < count; j++) {

    	left = filters[0]->process(sample[j*2]);
    	right = filters[1]->process(sample[j*2+1]);

        sample[j*2] = left + right;
        sample[j*2+1] = left - right;

    }
    return size;
  }
```

but obviously thats not the right way. Some thoughts I had:
1) Add a method to FilterStream to allow the caller to pass in an arbitrary Converter
2) Modify ConverterNChannels to support the above

I would prefer not to add the processing overhead of another chained FilterStream if possible

Thoughts?

ian

pschatzmann, 2022-06-17:
I am currently providing the following functionality  that provides conversion:
- In the copier.copy() you can add a BaseConverter. You can also chain multiple converters with a MultiConverter
- Alternatively you can use a ConvertedStream instead  (to which you pass a converter (including a MultiConverter) )

thaaraak, 2022-06-17:
Heh - I saw you looking at liquid-dsp ;)  I didn't get a chance to play around with gnu-radio.

Let me try your suggestions

thaaraak, 2022-06-19:
I decided to go with adding a base converter to copier.copy()  worked great

closed 2022-06-22 by pschatzmann