pschatzmann/arduino-audio-tools #240: FFT - some observations

perlix, 2022-06-21:
Using the MultiOutput class, I managed to visualize played audio in a 32-bins frequency spectrum on a TFT dislay.
Perhaps my observations regarding the AudioRealFFT class and the two suggested alternative libraries could be helpful to others.

1. When using _AudioRealFFT_ together with WiFi, I had to put  `#include "AudioLibs/AudioRealFFT.h"` after the line that establishes the WiFi connection (in my case: `ICYStream urlStream(wifi, password);` ), otherwise WiFi wouldn't connect.
2. Using _AudioKissFFT_ instead of _AudioRealFFT_ produced much lower magnitudes overall, so it needed different scaling on the display. Also, this lib appears to consume more cpu time.
3. Using _AudioESP32FFT_ worked similarly to _AudioRealFFT_. I just had to rename the object, as this lib already uses 'fft' internally.

That said, the 'player-url_icy-i2s' example with added display works perfectly now and I'm impressed by how this library's modular design made it all so simple!

----

Tested with latest commit of all used libraries.
Arduino IDE version: 1.8.19.
Arduino-ESP32 package version: 1.0.6.
Hardware: TTGO-TM-Music (ESP32 with built-in PCM5102 DAC and ST7789 TFT display).
Selected in IDE: ESP32 Dev Module, PSRAM Disabled; Partition: Huge App.

pschatzmann, 2022-06-22:
I don't think there is any issue that is asking for a change of code.

I will add a link to your comments to the FFT Wiki

closed 2022-06-22 by pschatzmann

RWBRWB, 2022-06-22:
> Using the MultiOutput class, I managed to visualize played audio in a 32-bins frequency spectrum on a TFT dislay. Perhaps my observations regarding the AudioRealFFT class and the two suggested alternative libraries could be helpful to others.
> 

>     1. When using _AudioRealFFT_ together with WiFi, I had to put  `#include "AudioLibs/AudioRealFFT.h"` after the line that establishes the WiFi connection (in my case: `ICYStream urlStream(wifi, password);` ), otherwise WiFi wouldn't connect.
> 
>     2. Using _AudioKissFFT_ instead of _AudioRealFFT_ produced much lower magnitudes overall, so it needed different scaling on the display. Also, this lib appears to consume more cpu time.
> 
>     3. Using _AudioESP32FFT_ worked similarly to _AudioRealFFT_. I just had to rename the object, as this lib already uses 'fft' internally.
> 
> 
> That said, the 'player-url_icy-i2s' example with added code for a display works perfectly now and I'm impressed by how this library's modular design made it all so simple!
> 
> Tested with latest commit of all used libraries. Arduino IDE version: 1.8.19. Arduino-ESP32 package version: 1.0.6. Hardware: TTGO-TM-Music (ESP32 with built-in PCM5102 DAC and ST7789 TFT display). Selected in IDE: ESP32 Dev Module, PSRAM Disabled; Partition: Huge App.

@Perlix I'm also trying to display a 30-40 band spectrum for the audio playing on a LCD display while using Phil's web radio streaming library. 

I already have the web streaming working perfectly so I just need to add this AudioRealFFT library & feed it the audio data that the web radio stream is providing. 

Are you able to share your code so I can see how you accomplished it and then I can convert to work with my LCD?

I'm sure it's easy but I learn quickly by just looking at examples so if you can share what's working for you that would be awesome. 

Thanks!

StuartIanNaylor, 2022-10-24:
@pschatzmann there are two other FFT libs that crop up quite often for optimised FFT and that is https://gitlab.mpcdf.mpg.de/mtr/pocketfft and for more complex started seeing https://github.com/Themaister/muFFT crop up

Also with FFT and DSP with the vector instructions of the esp32-s3 how well did Esspressif manage to optimise there libs.
Would love to see how a s3 optimised pocketfft compares against https://github.com/espressif/esp-dsp

pschatzmann, 2022-10-24:
Feel free to make a comparision or extend my library with your preferred implementation. 
I am already providing 4 alternatives as you can find in https://github.com/pschatzmann/arduino-audio-tools/wiki/FFT

StuartIanNaylor, 2022-10-24:
Ooof no as still recovering from my 1st C hack :) PocketFFT seems to of gained choice by a lot of the ML libs but prob would use the esspressif FFT as its optimised for the S3. I might try benching a S3 with what you have provided.

pschatzmann, 2022-10-25:
Since the Espressif DSP Library is part of Arduino now, I have also added an implementation which uses this functionality:
[ttps://github.com/pschatzmann/arduino-audio-tools/wiki/FFT](https://github.com/pschatzmann/arduino-audio-tools/wiki/FFT)