pschatzmann/arduino-audio-tools #241: SineFromTable setAmplitude only goes up

Denis-2021, 2022-06-22:
Hi.
I am still playing with your amazing library :)

This time I am trying to change the volume of a sine wave that I output over I2S.
I have setup SineFromTable<int16_t> sineWave(0); se the amplitude will start from 0.

Next I call a function that should change the volume, but for some reason it only works up and not down
sineWave.setAmplitude(mappedVolume); //0-32767

I have searched the library and found the following code:

        void updateAmplitudeInSteps() {
            float diff = amplitude_to_be - amplitude; 
            if (abs(diff) > max_amplitude_step){
                diff = (diff<0) ? -max_amplitude_step : max_amplitude_step;
            } 
            if (diff>=1.0){
                amplitude += diff;
            }
        } 

It looks like it only increase the amplitude and never decreases it.
I have tried adding the following code:

if (diff<=-1.0){
amplitude -= diff;
}

unfortunately it didn't work, when I try to decrease the volume I get a lot of noise on my speaker instead of a sine wave.

pschatzmann, 2022-06-22:
I suggest that you have a look at the Wiki. There is a chapter about volume control...

closed 2022-06-26 by pschatzmann

Denis-2021, 2022-07-14:
This line fixed it! 
Thank you very much. :)