pschatzmann/arduino-audio-tools #243: A2DP->I2S produces static noise

garageeks, 2022-06-25:
Dear all,
I just begin experimenting with this nifty library.

I wanted to start with a A2DP source to I2S output, however I only get static noise.
I'm running 2.0.3 framework and a MAX98375 amplifier.
It works properly when streaming a web radio.

However it only emits static noise when using this code. What am I missing?

Many thanks
Nick

```
#include "AudioTools.h"
#include "AudioLibs/AudioA2DP.h"

I2SStream i2sStream;                            // Access I2S as stream
A2DPStream a2dpStream = A2DPStream::instance(); // access A2DP as stream
VolumeStream volume(a2dpStream);
StreamCopy copier(i2sStream, volume);
ConverterFillLeftAndRight<int16_t> filler(LeftIsEmpty); // fill both channels

// Arduino Setup
void setup(void) {
    Serial.begin(115200);
    AudioLogger::instance().begin(Serial, AudioLogger::Warning);

    // set intial volume
    volume.setVolume(0.3);
    
    // start bluetooth
    Serial.println("starting A2DP...");
    auto cfgA2DP = a2dpStream.defaultConfig(RX_MODE);
    cfgA2DP.name = "STOPCHARGE";
    a2dpStream.begin(cfgA2DP);

    // start i2s input with default configuration
    Serial.println("starting I2S...");
    a2dpStream.setNotifyAudioChange(i2sStream); // i2s is using the info from a2dp
    auto config = i2sStream.defaultConfig(TX_MODE);
    config.i2s_format = I2S_STD_FORMAT;
    config.pin_bck = 27;
    config.pin_ws = 26;
    config.pin_data = 25;
    i2sStream.begin(config);

}

// Arduino loop - copy data
void loop() {
   // copier.copy(filler);
    copier.copy();
}
```

pschatzmann, 2022-06-25:
The Stream API for A2DP is very inefficient and I recommend to use the [basic API](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-basic-api/basic-a2dp-i2s/basic-a2dp-i2s.ino) instead. If you just want to output to I2S it might be even better just to use the [A2DP library directly](https://github.com/pschatzmann/ESP32-A2DP/blob/main/examples/bt_music_receiver_simple/bt_music_receiver_simple.ino).

I think your issue is somehow related to the VolumeStream. It seems to work if you do the volume on the output side:
``` 
VolumeStream volume(i2sStream);
StreamCopy copier(volume, a2dpStream);
```
 
There is also potentially a small issue with the setNotifyAudioChange() logic: I would suggest to move the start bluetooth section to the end, so that when a2dp starts it updates the i2s settings. Like this A2DP is already started when you call setNotifyAudioChange and this has no impact. However this is not really important since the default values of the I2S are the right ones for this case.

garageeks, 2022-06-25:
Thank you for the quick reply!
I tried to apply VolumeStream class to the i2s stream as suggested but still static noise.
By getting rid of it altogether the thing started to sound as expected! Moreover, volume control from the smartphone still work.
It has many dropouts due to the inefficiency you mentioned.
With the basic API mentioned works perfectly, however I wonder if I can still apply filters to the audio stream.
In this way I would try to software-correct the quality of the crappy speaker and box I put together.

pschatzmann, 2022-06-25:
With the Basic API you can still use Filters: You would just write to the FilteredStream instead of directly I2s

garageeks, 2022-06-25:
Thank you for the tip and for this amazing work! Lots of different projects are much easier with it.

closed 2022-06-25 by garageeks