pschatzmann/arduino-audio-tools #245: Can't seem to run some examples on Feather ESP32-S3

luukio, 2022-06-26:
With all [drivers](https://learn.adafruit.com/adafruit-esp32-s3-feather/arduino-ide-setup) and latest version of these audio-tools, I am getting this error while trying to run different examples on a Feather ESP32-S3. 

Arduino: 1.8.19 (Mac OS X), Board: "Adafruit Feather ESP32-S3 2MB PSRAM, QIO 80MHz, Core 1, Core 1, USB-OTG (TinyUSB), Enabled, Disabled, Disabled, USB-OTG CDC (TinyUSB), Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), 921600, None"

> In file included from /Users/lrombouts@ideo.com/Documents/Arduino/libraries/arduino-audio-tools/src/AudioAnalog/AnalogAudio.h:3,
>                  from /Users/lrombouts@ideo.com/Documents/Arduino/libraries/arduino-audio-tools/src/AudioTools.h:20,
>                  from /Users/lrombouts@ideo.com/Documents/Arduino/libraries/arduino-audio-tools/examples/examples-stream/streams-generator-i2s/streams-generator-i2s.ino:8:
> /Users/lrombouts@ideo.com/Documents/Arduino/libraries/arduino-audio-tools/src/AudioAnalog/AnalogAudioESP32.h:7:10: fatal error: soc/dac_channel.h: No such file or directory
>  #include "soc/dac_channel.h"
>           ^~~~~~~~~~~~~~~~~~~
> compilation terminated.
> exit status 1
> Error compiling for board Adafruit Feather ESP32-S3 2MB PSRAM.

Not sure if this is a bug or I am missing something here?

pschatzmann, 2022-06-26:
I have just committed a correction, that should work if you use the generic ESP32S3 Dev Module board setting.
If you want to use another board you will need to add an additional rule in the AudioConfig.h
 ```
#if defined(ESP32)  && (defined(ARDUINO_ESP32S3_DEV) || defined(YOUR_BOARD_DEV))
#define ESP32S3
#define ESP32X
#endif
 ```

closed 2022-06-26 by pschatzmann

Rasmushauschild, 2022-10-31:
Sorry for reopening an old thread, but I am having the exact same issue with the same board (Adafruit Feather ESP32-S3), and I can't for the life of me figure out where the problem is. This is the error I get:

`In file included from lib/arduino-audio-tools-main/src/AudioAnalog/AnalogAudio.h:3,
                 from lib/arduino-audio-tools-main/src/AudioTools.h:20,
                 from src/main.cpp:7:
lib/arduino-audio-tools-main/src/AudioAnalog/AnalogAudioESP32.h:7:10: fatal error: soc/dac_channel.h: No such file or directory
 #include "soc/dac_channel.h"
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.`

Any help is greatly appreciated!

pschatzmann, 2022-10-31:
Only the ESP32 supports the DAC and ADC via I2S!
See the overview in https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/i2s.html

Rasmushauschild, 2022-11-01:
Ah, thanks for pointing that out!
I won't be needing DAC or ADC via I2S, as I plan on using PDM for audio input.

Can you point me in the right direction regarding what I need to modify to get the AudioTools library to compile without DAC/ADC support?

pschatzmann, 2022-11-01:
Can you double check if you use an actual version ? If you double check in AudioConfig.h the USE_I2S_ANALOG is not available for ESP32C3, ESP32S3, ESP32S2 - so this should not get this error...

Rasmushauschild, 2022-11-01:
I see `#define AUDIOTOOLS_VERSION "0.8.0"` in AudioConfig.h

I use Platform.io for this project, so I'm unable to select ESP32 S3 Dev Module as the board.
I tested it with Arduino IDE, selecting ESP32 S3 Dev Module in there, and then it compiles without any issues, so the board is definitely what has been causing the problem.

Could I extend support for the adafruit_feather_esp32_s3 board, or is there a better solution?

Thank you! I appreciate your help!

pschatzmann, 2022-11-01:
In PlatformIO this is easy: just add -DARDUINO_ESP32S3_DEV to the build_flags

Rasmushauschild, 2022-11-01:
Works like a charm!
Thank you very much!

Johan-kas, 2024-08-01:
Hello, I use this code for my project

#include "AudioTools.h"
#include "BluetoothA2DPSink.h"

I2SStream i2s;
BluetoothA2DPSink a2dp_sink(i2s);

void setup() {
    auto cfg = i2s.defaultConfig();
    cfg.pin_bck = 14;
    cfg.pin_ws = 15;
    cfg.pin_data = 22;
    i2s.begin(cfg);

    a2dp_sink.start("MyMusic");
}

void loop() {
}

but I always get this error

`Sketch uses 1874517 bytes (143%) of program storage space. Maximum is 1310720 bytes.
Global variables use 61804 bytes (18%) of dynamic memory, leaving 265876 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
text section exceeds available space in board

Compilation error: text section exceeds available space in board`
Can you help me please!

Johan-kas, 2024-08-01:
> Hello, I use this code for my project
>
> #include "AudioTools.h" #include "BluetoothA2DPSink.h"
>
> I2SStream i2s; BluetoothA2DPSink a2dp_sink(i2s);
>
> void setup() { auto cfg = i2s.defaultConfig(); [...]

pschatzmann, 2024-08-01:
I don't understand: I have answered this question already a 1000 times
What is so difficult about reading the A2DP FAQ ?