pschatzmann/arduino-audio-tools #246: ESP32C3 Dev Module - doesn't want to compile

MichaelPNolan, 2022-06-26:
Simply trying to get started. I've build synth projects already from other libraries and my own forks.

So when I use the board definition ESP32C3 Dev Module I get compiler issues and this is not a problem so much with a normal ESP32.

I can't even get past this simple "Note: candidate" int and int24_t  type issue

Any advice or just give up on this board for i2s with your library?

=-------------------------
Compiler warnings etc below from trying to compile your example streams-generator-i2s as seen on 
https://www.pschatzmann.ch/home/2022/04/15/using-a-esp32-c3-with-my-arduino-audio-tools-library/
=========================

streams-generator-i2s:19:1: error: 'I2SStream' does not name a type; did you mean 'Stream'?
 I2SStream out;
 ^~~~~~~~~
 Stream
streams-generator-i2s:20:19: error: 'out' was not declared in this scope
 StreamCopy copier(out, sound);                             // copies sound into i2s
                   ^~~
/home/owner/Arduino/Esp32-c3/streams-generator-i2s/streams-generator-i2s.ino: In function 'void setup()':
streams-generator-i2s:31:17: error: 'out' was not declared in this scope
   auto config = out.defaultConfig(TX_MODE);
                 ^~~
In file included from /home/owner/Arduino/libraries/arduino-audio-tools/src/AudioTools.h:14,
                 from /home/owner/Arduino/Esp32-c3/streams-generator-i2s/streams-generator-i2s.ino:11:
/home/owner/Arduino/libraries/arduino-audio-tools/src/AudioTools/Converter.h: In instantiation of 'size_t audio_tools::ChannelReducer<T>::convert(uint8_t*, uint8_t*, size_t) [with T = audio_tools::int24_t; size_t = unsigned int; uint8_t = unsigned char]':
/home/owner/Arduino/libraries/arduino-audio-tools/src/AudioTools/Converter.h:498:57:   required from 'size_t audio_tools::ChannelConverter<T>::convert(uint8_t*, uint8_t*, size_t) [with T = audio_tools::int24_t; size_t = unsigned int; uint8_t = unsigned char]'
/home/owner/Arduino/libraries/arduino-audio-tools/src/AudioTools/AudioStreamsConverter.h:84:16:   required from 'size_t audio_tools::ChannelFormatConverterStreamT<T>::convert(const uint8_t*, size_t) [with T = audio_tools::int24_t; size_t = unsigned int; uint8_t = unsigned char]'
/home/owner/Arduino/libraries/arduino-audio-tools/src/AudioTools/AudioStreamsConverter.h:37:33:   required from 'size_t audio_tools::ChannelFormatConverterStreamT<T>::write(const uint8_t*, size_t) [with T = audio_tools::int24_t; size_t = unsigned int; uint8_t = unsigned char]'
/home/owner/Arduino/libraries/arduino-audio-tools/src/AudioTools/AudioStreamsConverter.h:144:104:   required from here
/home/owner/Arduino/libraries/arduino-audio-tools/src/AudioTools/Converter.h:389:19: error: conversion from 'int' to 'audio_tools::int24_t' is ambiguous
                 T total = 0;
                   ^~~~~
In file included from /home/owner/Arduino/libraries/arduino-audio-tools/src/AudioTools/AudioTypes.h:4,
                 from /home/owner/Arduino/libraries/arduino-audio-tools/src/AudioTimer/AudioTimerDef.h:2,
                 from /home/owner/Arduino/libraries/arduino-audio-tools/src/AudioTimer/AudioTimerESP32.h:4,
                 from /home/owner/Arduino/libraries/arduino-audio-tools/src/AudioTimer/AudioTimer.h:3,
                 from /home/owner/Arduino/libraries/arduino-audio-tools/src/AudioTools.h:11,
                 from /home/owner/Arduino/Esp32-c3/streams-generator-i2s/streams-generator-i2s.ino:11:
/home/owner/Arduino/libraries/arduino-audio-tools/src/AudioBasic/Int24.h:33:3: note: candidate: 'audio_tools::int24_t::int24_t(const int32_t&)'
   int24_t(const int32_t &in) {
   ^~~~~~~
/home/owner/Arduino/libraries/arduino-audio-tools/src/AudioBasic/Int24.h:27:3: note: candidate: 'audio_tools::int24_t::int24_t(const int16_t&)'
   int24_t(const int16_t &in) {
   ^~~~~~~
/home/owner/Arduino/libraries/arduino-audio-tools/src/AudioBasic/Int24.h:23:3: note: candidate: 'audio_tools::int24_t::int24_t(void*)'
   int24_t(void *ptr) {
   ^~~~~~~
In file included from /home/owner/Arduino/libraries/arduino-audio-tools/src/AudioTools.h:14,
                 from /home/owner/Arduino/Esp32-c3/streams-generator-i2s/streams-generator-i2s.ino:11:
/home/owner/Arduino/libraries/arduino-audio-tools/src/AudioTools/Converter.h: In instantiation of 'size_t audio_tools::ChannelEnhancer<T>::convert(uint8_t*, uint8_t*, size_t) [with T = audio_tools::int24_t; size_t = unsigned int; uint8_t = unsigned char]':
/home/owner/Arduino/libraries/arduino-audio-tools/src/AudioTools/Converter.h:500:58:   required from 'size_t audio_tools::ChannelConverter<T>::convert(uint8_t*, uint8_t*, size_t) [with T = audio_tools::int24_t; size_t = unsigned int; uint8_t = unsigned char]'
/home/owner/Arduino/libraries/arduino-audio-tools/src/AudioTools/AudioStreamsConverter.h:84:16:   required from 'size_t audio_tools::ChannelFormatConverterStreamT<T>::convert(const uint8_t*, size_t) [with T = audio_tools::int24_t; size_t = unsigned int; uint8_t = unsigned char]'
/home/owner/Arduino/libraries/arduino-audio-tools/src/AudioTools/AudioStreamsConverter.h:37:33:   required from 'size_t audio_tools::ChannelFormatConverterStreamT<T>::write(const uint8_t*, size_t) [with T = audio_tools::int24_t; size_t = unsigned int; uint8_t = unsigned char]'
/home/owner/Arduino/libraries/arduino-audio-tools/src/AudioTools/AudioStreamsConverter.h:144:104:   required from here
/home/owner/Arduino/libraries/arduino-audio-tools/src/AudioTools/Converter.h:432:15: error: conversion from 'int' to 'audio_tools::int24_t' is ambiguous
             T value = 0;
               ^~~~~
In file included from /home/owner/Arduino/libraries/arduino-audio-tools/src/AudioTools/AudioTypes.h:4,
                 from /home/owner/Arduino/libraries/arduino-audio-tools/src/AudioTimer/AudioTimerDef.h:2,
                 from /home/owner/Arduino/libraries/arduino-audio-tools/src/AudioTimer/AudioTimerESP32.h:4,
                 from /home/owner/Arduino/libraries/arduino-audio-tools/src/AudioTimer/AudioTimer.h:3,
                 from /home/owner/Arduino/libraries/arduino-audio-tools/src/AudioTools.h:11,
                 from /home/owner/Arduino/Esp32-c3/streams-generator-i2s/streams-generator-i2s.ino:11:
/home/owner/Arduino/libraries/arduino-audio-tools/src/AudioBasic/Int24.h:33:3: note: candidate: 'audio_tools::int24_t::int24_t(const int32_t&)'
   int24_t(const int32_t &in) {
   ^~~~~~~
/home/owner/Arduino/libraries/arduino-audio-tools/src/AudioBasic/Int24.h:27:3: note: candidate: 'audio_tools::int24_t::int24_t(const int16_t&)'
   int24_t(const int16_t &in) {
   ^~~~~~~
/home/owner/Arduino/libraries/arduino-audio-tools/src/AudioBasic/Int24.h:23:3: note: candidate: 'audio_tools::int24_t::int24_t(void*)'
   int24_t(void *ptr) {
   ^~~~~~~
exit status 1
'I2SStream' does not name a type; did you mean 'Stream'?

pschatzmann, 2022-06-26:
I have just committed a correction, that should work if you use the generic ESP32C3 board setting. 
If you want to use another board you will need to add an additional rule in the AudioConfig.h

closed 2022-06-26 by pschatzmann

MichaelPNolan, 2022-06-26:
Thanks - i'm going to try a few boards ... cheers