pschatzmann/arduino-audio-tools #250: MemoryStream constructor does not initialise data

timr7777, 2022-07-02:
MemoryStream initialised from byte array does not set the vector content, producing noise in the example I tried.
Test code below (test ran on ESP32).
Quick fix attached, but I do not have much experience of c++, and I did not spend long on this.
```
#include "AudioTools.h"

uint8_t raw[] = {
  0x12, 0x34, 0x56, 0x78, 0x9a, 0xbc, 0xde, 0xf0
};

MemoryStream my_stream( raw, sizeof( raw ) / sizeof( raw[0] ));

void setup() {
  Serial.begin(115200);

  uint8_t bytes[4];
  my_stream.readBytes( bytes, sizeof( bytes ) );

  if( memcmp( raw, bytes, sizeof( bytes ) ) == 0 )
    Serial.print( "pass\n" );
  else  
    Serial.print( "fail\n" );
}

void loop() {
  // put your main code here, to run repeatedly:
}
``
[01_memory_stream_init.txt](https://github.com/pschatzmann/arduino-audio-tools/files/9033481/01_memory_stream_init.txt)
`

timr7777, 2022-07-02:
[01_memory_stream_init.txt](https://github.com/pschatzmann/arduino-audio-tools/files/9033501/01_memory_stream_init.txt)

pschatzmann, 2022-07-02:
I agree, one of my last commits was not good. I am currently working to correct this bug and add PSRAM support....

pschatzmann, 2022-07-04:
I committed the correction: the MemoryStream should work again as expected.