pschatzmann/arduino-audio-tools #251: Noice on output

IsiTapier, 2022-07-04:
I'm getting a lot of noice on the aux out. Any way to prevent this from happening?

IsiTapier, 2022-07-04:
oh it seems to be on the input not on the output

pschatzmann, 2022-07-09:
You did not really provide any hardware information, so I assume that you refer to the known hardware bug of the AudioKit:
https://github.com/Ai-Thinker-Open/ESP32-A1S-AudioKit/issues/31

perlix, 2022-07-10:
In case OP does refer to the newer Audiokit: using the [thaaraak/es8388](https://github.com/thaaraak/es8388) library solved the input noise issue on my 'type 5' Audiokits. It may not entirely mute the mics, but their sensitivity is reduced to almost zero, making a LINE IN to LINE OUT loopback sound crystal clear.

closed 2022-07-28 by pschatzmann

pschatzmann, 2023-05-21:
https://github.com/pschatzmann/arduino-audio-tools/issues/251