pschatzmann/arduino-audio-tools #253: MP3 Does not finish playing

barrettj, 2022-07-05:
I adapted this code from streams-sd_mp3-i2s.ino and https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-memory_mp3_short-i2s/streams-memory_mp3_short-i2s.ino#L29 and for the life of me I cannot get it to play the audio file in full - it always just says "a is for" and then cuts out (if I just use the code from the first sample it ends with static, if I use the second it just ends early).  Do you have any idea what I'm doing wrong?

Audio File: https://www.dropbox.com/s/4yaeommryputqjs/a.mp3?dl=0

```
#include <SPI.h>
#include <SD.h>
#include "AudioTools.h"
#include "AudioCodecs/CodecMP3Helix.h"

const int chipSelect=10;
I2SStream i2s; // final output of decoded stream
VolumeStream volume(i2s);
MP3DecoderHelix helix;
EncodedAudioStream decoder(&volume, &helix); // Decoding stream
StreamCopy copier; 
File audioFile;

void setup(){
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);  

  // setup file
  if(!SD.begin()){
     Serial.println("Card Mount Failed");
     return;
  }
  audioFile = SD.open("/a.mp3");

  // setup i2s
  auto config = i2s.defaultConfig(TX_MODE);

  config.i2s_format = I2S_STD_FORMAT;
  config.pin_bck = 15;
  config.pin_ws = 14;
  config.pin_data = 22;
  
  i2s.begin(config);

  // setup I2S based on sampling rate provided by decoder
  decoder.setNotifyAudioChange(i2s);
  decoder.begin();

 // set initial volume
  volume.begin(config); // we need to provide the bits_per_sample and channels
  volume.setVolume(0.7);

  // begin copy
  copier.begin(decoder, audioFile);
}

void loop(){
  if (copier.available()) {
    copier.copy();
  } else {

    auto info = decoder.decoder().audioInfo();
    LOGI("The audio rate from the mp3 file is %d", info.sample_rate);
    LOGI("The channels from the mp3 file is %d", info.channels);
        helix.end(); // flush output
    i2s.end();
    stop();
  }
}
```

It also outputs: ```
[I] C:\Users\xbox\OneDrive\esp32audio\streams-sd_mp3-i2s\streams-sd_mp3-i2s.ino : 65 - The audio rate from the mp3 file is 0
[I] C:\Users\xbox\OneDrive\esp32audio\streams-sd_mp3-i2s\streams-sd_mp3-i2s.ino : 66 - The channels from the mp3 file is 0

```

barrettj, 2022-07-14:
Anyone have any ideas. I’ve wasted so much time on this project because it looked like it would do exactly what I want but it’s now been blocking me for weeks.

I’m willing to pay a $100 bounty via PayPal or Venmo or something to anyone who can genuinely get me going with this library or another library (and I’m starting to lean towards preferring another library as this one seems like it’s so concerned with doing other things it’s not actually testing simple cases like just playing a file from SD)

pschatzmann, 2022-07-14:
I think this is a buffering problem that is in the mp3 decoder.

- If you use e.g. wav files instead with a WAVDecoder the problem will go away
- you can try with some other mp3 decoders
- Some users were using some additional silence at the end of the mp3 file others are playing short mp3 file with just some silence at the end of the output.

The mp3 decoder is designed for constant streaming so in this case this does not really matter. It is my understanding that this problem is only with the last file that you played

barrettj, 2022-07-15:
> I am not sure if I can reproduce your issue. I added a test case
> https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/tests/short-mp3/short-mp3.ino
>
> Can you share one of your mp3 files ?

There should be one in the original post, let me know if the link isn’t working (I had to Dropbox it as GitHub doesn’t seem to natively allow mp3s to be uploaded): https://www.dropbox.com/s/4yaeommryputqjs/a.mp3?dl=0

pschatzmann, 2022-07-15:
I replaced my test file with yours and I can confirm that the test sketch from above is working perfectly and no audio is lost at the end.

Just apply the changes to your sketch or use the sketch by replacing AudioKitStream with I2SStrem...

barrettj, 2022-07-16:
I can confirm that by changing to the MAD decoder and either adding a delay in the else of loop or using copyAll that it plays 99% percent now - but there's a slight hiccup right at the end that's not part of the file.  Any ideas on how to get rid of that?

barrettj, 2022-07-16:
https://share.icloud.com/photos/067qzd-neH7d2debF6PC61C_Q

It’s the click right at the end of “apple”

barrettj, 2022-07-16:
I've tried using your sample, it displays a lot to the serial monitor but never plays any audio that I can hear:

```
#include <SPI.h>
#include <SD.h>
#include "AudioTools.h"
//#include "AudioCodecs/CodecMP3Helix.h"
#include "AudioCodecs/CodecMP3MAD.h"

URLStream url(redacted);  // or replace with ICYStream to get metadata
I2SStream i2s; // final output of decoded stream
MP3DecoderMAD decoder;
EncodedAudioStream dec_stream(i2s, decoder); // Decoding stream
StreamCopy copier(dec_stream, url); // copy url to decoder

void setup(){
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);  

  // setup i2s
  auto config = i2s.defaultConfig(TX_MODE);

  config.i2s_format = I2S_STD_FORMAT;
  config.pin_bck = 15;
  config.pin_ws = 14;
  config.pin_data = 22;
  config.channels = 1;
  config.sample_rate = 24000;
  
  i2s.begin(config);

  dec_stream.begin();

  url.begin("https://pschatzmann.github.io/arduino-audio-tools/resources/short-mp3.mp3","audio/mp3");
}

void loop(){
  if (copier.available()) {
    copier.copy();
  } else {
    delay(1000);
    decoder.end(); // flush output
    i2s.end();
    stop();
  }
}
```

```
....
[I] URLStream.h : 225 - WiFiClientSecure
[I] URLStream.h : 225 - WiFiClientSecure
[I] HttpRequest.h : 182 - process connecting to host pschatzmann.github.io port 443
[I] HttpRequest.h : 170 - connect pschatzmann.github.io
[I] HttpRequest.h : 193 - Free heap: 161424
[I] HttpHeader.h : 357 - HttpRequestHeader::setValues - path: /arduino-audio-tools/resources/short-mp3.mp3
[I] HttpHeader.h : 254 - HttpHeader::write
[I] HttpHeader.h : 366 - HttpRequestHeader::write1stLine
[I] HttpHeader.h : 381 - -> GET /arduino-audio-tools/resources/short-mp3.mp3 HTTP/1.1
[I] HttpHeader.h : 191 -  -> Host: pschatzmann.github.io 
[I] HttpHeader.h : 191 -  -> Connection: close 
[I] HttpHeader.h : 191 -  -> Accept: audio/mp3 
[I] HttpHeader.h : 297 -  -> <CR LF> 
[I] HttpRequest.h : 217 - Request written ... waiting for reply
[I] HttpHeader.h : 224 - HttpHeader::read
[W] HttpHeader.h : 231 - Waiting for data...
[I] HttpHeader.h : 161 - HttpHeader::readLine -> HTTP/1.1 200 OK
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Connection: close
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Content-Length: 5785
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Server: GitHub.com
[... 60 lines cut ...]
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Content-Length: 5785
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Server: GitHub.com
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Content-Type: audio/mp3
[I] HttpHeader.h : 161 - HttpHeader::readLine -> permissions-policy: interest-cohort=()
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Last-Modified: Fri, 15 Jul 2022 13:36:36 GMT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Access-Control-Allow-Origin: *
[I] HttpHeader.h : 161 - HttpHeader::readLine -> ETag: "62d16d64-1699"
[I] HttpHeader.h : 161 - HttpHeader::readLine -> expires: Sat, 16 Jul 2022 00:27:23 GMT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Cache-Control: max-age=600
[I] HttpHeader.h : 161 - HttpHeader::readLine -> x-proxy-cache: MISS
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-GitHub-Request-Id: 305E:0F59:417774:710086:62D20393
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Accept-Ranges: bytes
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Date: Sat, 16 Jul 2022 00:20:06 GMT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Via: 1.1 varnish
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Age: 162
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Served-By: cache-chi-klot8100178-CHI
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Cache: HIT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Cache-Hits: 1
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Timer: S1657930806.256620,VS0,VE1
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Vary: Accept-Encoding
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Fastly-Request-ID: 2210d6c320cde54ad985baff024832fbec3db4ea
[I] HttpHeader.h : 161 - HttpHeader::readLine -> 
[I] URLStream.h : 84 - size: 5785

```

pschatzmann, 2022-07-16:
Strange I consistently get 
```
...
10:07:50.493 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Fastly-Request-ID: 69d462334ce46c0b2b72fbaccd8fc44c12311e0a
10:07:50.528 -> [I] HttpHeader.h : 161 - HttpHeader::readLine -> 
10:07:50.528 -> [I] URLStream.h : 93 - size: 5785
10:07:50.528 -> [I] I2SStream.h : 107 - virtual void audio_tools::I2SStream::setAudioInfo(audio_tools::AudioBaseInfo)
10:07:50.528 -> [I] AudioTypes.h : 48 - sample_rate: 24000
10:07:50.528 -> [I] AudioTypes.h : 49 - channels: 1
10:07:50.528 -> [I] AudioTypes.h : 50 - bits_per_sample: 16
10:07:50.528 -> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:07:50.562 -> [I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
10:07:50.733 -> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:07:50.809 -> [I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
10:07:51.050 -> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:07:51.156 -> [I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
10:07:51.403 -> [I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:07:51.471 -> [I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
10:07:51.576 -> [I] AudioCopy.h : 121 - StreamCopy::copy 301 -> 301 -> 301 bytes - in 1 hops
```

- Are you using Arduino or PlatformIO
- What is the ESP32 Version from the Platform Manager

barrettj, 2022-07-16:
> Are you using Arduino or PlatformIO
Arduino

> What is the ESP32 Version from the Platform Manager
I'm not 100% certain what is meant by this, hopefully these screen shots answers the question - if not please let me know, googling "arduino platform manager" didn't give me much to go on.  The board itself I'm using is an Esp32-wroom-32D.

[image]
[image]

> Have you tried any other sketches with URLStream or ICYStream ?

No, my full use case reading and writing mp3 (it's not a hard requirement they are MP3s but compression would be nice, and I assumed easy to test) files to flash storage (either onboard or SD card), playing from the speaker, recording from a microphone, and possibly having files come over and be replaced via wifi or bluetooth via cellphone app (which is actually the part of this project that I'm way more familiar with, I'm a long time iOS/Apple dev; our SUPER long time hardware guy had a family emergency and is basically MIA for a few months and no idea if he'll return so I'm sort of trying to pick up that role).  So I haven't investigated any remote playing/urlstream as my use case is all "local" in regards to the files.

pschatzmann, 2022-07-16:
ESP32 1.0.6 is pretty old. The latest version is 2.0.4 now

barrettj, 2022-07-16:
Is there something I'm doing wrong?  I don't see anything newer than 1.0.6
[image]
[image]

pschatzmann, 2022-07-17:
https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

barrettj, 2022-07-18:
Thank you, I must have gotten that value from an outdated tutorial.

I can confirm that updating to 2.0.4 did not have an effect on any of my issues.

barrettj, 2022-07-19:
> If you use e.g. wav files instead with a WAVDecoder the problem will go away
 
I'm attempting to do this (using the same code that works from above but replacing it with WAVDecoder and the path to files) and I'm getting no output and the following to serial when trying to play:

```
[I] I2SConfig.h : 89 - rx/tx mode: TX
[I] I2SConfig.h : 90 - port_no: 0
[I] I2SConfig.h : 91 - is_master: Master
[I] I2SConfig.h : 92 - sample rate: 24000
[I] I2SConfig.h : 93 - bits per sample: 16
[I] I2SConfig.h : 94 - number of channels: 1
[I] I2SConfig.h : 95 - i2s_format: I2S_LSB_FORMAT
[I] I2SConfig.h : 97 - auto_clear:1
[I] I2SConfig.h : 104 - buffer_count:10
[I] I2SConfig.h : 105 - buffer_size:512
[I] I2SConfig.h : 109 - pin_bck: 15
[I] I2SConfig.h : 110 - pin_ws: 14
[I] I2SConfig.h : 111 - pin_data: 22
[I] AudioStreams.h : 751 - setVolume: 1.000000
[I] AudioStreams.h : 751 - setVolume: 0.700000
[I] AudioCopy.h : 66 - buffer_size=1024
[I] CodecWAV.h : 49 - WAVHeader len: 1024
[I] CodecWAV.h : 215 - WAVHeader sound_pos: 58
[I] CodecWAV.h : 216 - WAVHeader channels: 1 
[I] CodecWAV.h : 217 - WAVHeader bits_per_sample: 8
[I] CodecWAV.h : 218 - WAVHeader sample_rate: 8000 
[I] CodecWAV.h : 219 - WAVHeader format: 7
[I] CodecWAV.h : 314 - WAV sample_rate: 8000
[I] CodecWAV.h : 315 - WAV data_length: 6420
[I] CodecWAV.h : 316 - WAV is_streamed: 0
[I] CodecWAV.h : 317 - WAV is_valid: true
[E] CodecWAV.h : 323 - WAV format not supported: 7
[I] AudioCopy.h : 257 - try write - 2 
[I] AudioCopy.h : 257 - try write - 3 
[I] AudioCopy.h : 257 - try write - 4 
[I] AudioCopy.h : 257 - try write - 5 
[I] AudioCopy.h : 257 - try write - 6 
[I] AudioCopy.h : 257 - try write - 7 
[I] AudioCopy.h : 257 - try write - 8 
[I] AudioCopy.h : 257 - try write - 9 
[I] AudioCopy.h : 257 - try write - 10 
[I] AudioCopy.h : 257 - try write - 11 
[I] AudioCopy.h : 257 - try write - 12 
[I] AudioCopy.h : 257 - try write - 13 
[I] AudioCopy.h : 257 - try write - 14 
[I] AudioCopy.h : 257 - try write - 15 
[I] AudioCopy.h : 257 - try write - 16 
[I] AudioCopy.h : 257 - try write - 17 
[I] AudioCopy.h : 257 - try write - 18 
[I] AudioCopy.h : 257 - try write - 19 
[I] AudioCopy.h : 257 - try write - 20 
[I] AudioCopy.h : 257 - try write - 21 
[E] AudioCopy.h : 251 - write to target has failed!
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 0 bytes - in 22 hops
[I] AudioCopy.h : 257 - try write - 2 
[I] AudioCopy.h : 257 - try write - 3 
[I] AudioCopy.h : 257 - try write - 4 
[I] AudioCopy.h : 257 - try write - 5 
[I] AudioCopy.h : 257 - try write - 6 
[I] AudioCopy.h : 257 - try write - 7 
[I] AudioCopy.h : 257 - try write - 8 
[I] AudioCopy.h : 257 - try write - 9 
[I] AudioCopy.h : 257 - try write - 10 
[I] AudioCopy.h : 257 - try write - 11 
[I] AudioCopy.h : 257 - try write - 12 
[I] AudioCopy.h : 257 - try write - 13 
[I] AudioCopy.h : 257 - try write - 14 
[I] AudioCopy.h : 257 - try write - 15 
[I] AudioCopy.h : 257 - try write - 16 
[I] AudioCopy.h : 257 - try write - 17 
[... 94 lines cut ...]
[I] AudioCopy.h : 257 - try write - 2 
[I] AudioCopy.h : 257 - try write - 3 
[I] AudioCopy.h : 257 - try write - 4 
[I] AudioCopy.h : 257 - try write - 5 
[I] AudioCopy.h : 257 - try write - 6 
[I] AudioCopy.h : 257 - try write - 7 
[I] AudioCopy.h : 257 - try write - 8 
[I] AudioCopy.h : 257 - try write - 9 
[I] AudioCopy.h : 257 - try write - 10 
[I] AudioCopy.h : 257 - try write - 11 
[I] AudioCopy.h : 257 - try write - 12 
[I] AudioCopy.h : 257 - try write - 13 
[I] AudioCopy.h : 257 - try write - 14 
[I] AudioCopy.h : 257 - try write - 15 
[I] AudioCopy.h : 257 - try write - 16 
[I] AudioCopy.h : 257 - try write - 17 
[I] AudioCopy.h : 257 - try write - 18 
[I] AudioCopy.h : 257 - try write - 19 
[I] AudioCopy.h : 257 - try write - 20 
[I] AudioCopy.h : 257 - try write - 21 
[E] AudioCopy.h : 251 - write to target has failed!
[I] AudioCopy.h : 121 - StreamCopy::copy 334 -> 334 -> 0 bytes - in 22 hops
```

Any ideas?

pschatzmann, 2022-07-19:
Like the error message says [E] CodecWAV.h : 323 - WAV format not supported: 7
Only PCM is supported...

barrettj, 2022-07-19:
My apologies, I didn’t notice the E next to that line (I did see it, but I expected something like wave format not supported: true if it was an error instead of the 7)

barrettj, 2022-07-19:
Any ideas what might be causing it?

https://user-images.githubusercontent.com/361635/179761691-e58eba2d-3b5d-4239-9d3c-f030ff1e55bd.MOV

pschatzmann, 2022-07-19:
I suggest that you set the log level to Warning. ps. If you want to keep the file size low you can use a smaller sampling rate and 1 channel.
How did you create your file ?

pschatzmann, 2022-07-19:
Do you have this issue when you restart by rebooting or with your logic triggered in the loop at the end of the file ?

pschatzmann, 2022-07-28:
I did some tests with your files and after changing the log level to Warning the quality is perfect.
I assume that you might have added some lag in your loop processing, so that the I2S does not get enough data!
You can try to increase the copy size (in the constructor) or process your events only every nth loop.

closed 2022-07-28 by pschatzmann