pschatzmann/arduino-audio-tools #259: [ESP32-C3, INMP441] 'streams-i2s-webserver_wav' example doesn't play any audio in the browser

Technerder, 2022-07-10:
Like the title says I connected an INMP441 sensor to an [ESP32-C3 Dev board breakout](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/hw-reference/esp32c3/user-guide-devkitc-02.html), and upon changing some values to use the pins I connected the microphone to, and opening up Chrome, no audio appears to be streamed to the browser.

Code:
```c
/**
 * @file streams-i2s-webserver_wav.ino
 *
 *  This sketch reads sound data from I2S. The result is provided as WAV stream which can be listened to in a Web Browser
 *
 * @author Phil Schatzmann
 * @copyright GPLv3
 */

#include "AudioTools.h"

//AudioEncodedServer server(new WAVEncoder(),"ssid","password");  
AudioWAVServer server("REMOVED", "REMOVED"); // the same a above

I2SStream i2sStream;    // Access I2S as stream
ConverterFillLeftAndRight<int16_t> filler(LeftIsEmpty); // fill both channels - or change to RightIsEmpty

void setup(){
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // start i2s input with default configuration
  Serial.println("starting I2S...");
  auto config = i2sStream.defaultConfig(RX_MODE);
  config.i2s_format = I2S_STD_FORMAT; // if quality is bad change to I2S_LSB_FORMAT https://github.com/pschatzmann/arduino-audio-tools/issues/23
  config.sample_rate = 8000;
  config.channels = 1;
  config.bits_per_sample = 16;
  config.pin_bck = 5;
  config.pin_ws = 4;
  config.pin_data_rx = 6;
  i2sStream.begin(config);
  Serial.println("I2S started");

  // start data sink
  server.begin(i2sStream, config, &filler);
}

// Arduino loop  
void loop() {
  // Handle new connections
  server.doLoop();  
}
```

Pinouts (INMP441 -> ESP32):
```output
GND: GND
VDD: 3.3
SD: 6
L/R: GND
WS: 4
SCK: 5
```

What am I doing wrong?

pschatzmann, 2022-07-10:
I suggest that you start with [streams-i2s-serial](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-stream/streams-i2s-serial) as described in [External-ADC](https://github.com/pschatzmann/arduino-audio-tools/wiki/External-ADC) to figure out if the input side is working as expected and you need to adjust your sketch.

Most likely you receive the data on the right channel and the filter is clearning it out. If you receive noise with bits_per_sample 16 you should try it with 32 bits.

Technerder, 2022-07-10:
Flashing `streams-generator-webserver_wav` seems to indicate that the output is working as I hear audio through Chrome now. `streams-i2s-serial` on the other hand just prints out `0, 0` regardless of whether the INMP441's `L/R` pin is pulled up/down. Is there anything wrong with how I'm assigning the specific pins in the `i2sStream` config?

This is the `streams-i2s-serial` code I modified to use my custom pin config values
```c
/**
   @file streams-i2s-serial.ino
   @author Phil Schatzmann
   @brief see https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-i2s-serial/README.md

   @author Phil Schatzmann
   @copyright GPLv3
*/

#include "AudioTools.h"

I2SStream i2sStream; // Access I2S as stream
CsvStream<int32_t> csvStream(Serial);
StreamCopy copier(csvStream, i2sStream); // copy i2sStream to csvStream

// Arduino Setup
void setup(void) {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  auto cfg = i2sStream.defaultConfig(RX_MODE);
  cfg.i2s_format = I2S_STD_FORMAT; // or try with I2S_LSB_FORMAT
  cfg.bits_per_sample = 32;
  cfg.channels = 2;
  cfg.sample_rate = 44100;
  cfg.is_master = true;
  // this module nees a master clock if the ESP32 is master
  cfg.use_apll = false;  // try with yes
  cfg.pin_bck = 5; // <------------ I (Technerder) added this
  cfg.pin_ws = 4; // <------------ I (Technerder) added this
  cfg.pin_data_rx = 6; // <------------ I (Technerder) added this
  i2sStream.begin(cfg);

  // make sure that we have the correct channels set up
  csvStream.begin(cfg);

}

// Arduino loop - copy data
void loop() {
  copier.copy();
}
```

Changing `cfg.sample_rate` to 8000 also didn't seem to change anything.

pschatzmann, 2022-07-11:
Try with 
  cfg.pin_data = 6; // <------------ I (Technerder) added this

Technerder, 2022-07-12:
That seems to have fixed it. For future reference, what is the difference between `pin_data_rx` and `pin_data`?

pschatzmann, 2022-07-12:
Initially I only supported RX_MODE and TX_MODE for I2S where only one data pin (pin_data_rx) is needed.

Later I added the RXTX_MODE which can do data reading and writing at the same time and needs 2 data pins. For this case the pin_data does the output and pin_data_rx does the input.

Technerder, 2022-07-12:
Ah, that makes sense. Thanks for all the help!

closed 2022-07-12 by Technerder