pschatzmann/arduino-audio-tools #260: No audio out .

aqildad-create, 2022-07-14:
i am using [base-i2s-a2dp] AND [basic-a2dp-i2s] but no success .   

what are pin out use in this for i2s in this [basic-a2dp-i2s] ?? 

For trouble shooting Digital Micro phone serial data is working . but with [base-i2s-a2dp] AND [basic-a2dp-i2s]  no success .

so my confusion is what are the pin out use in this [basic-a2dp-i2s]   and how to change the pins if needed ?

aqildad-create, 2022-07-14:
Thank you , one more thing sir How to add echo base trouble repeat to this
microphone ? Is this possible ?  How ?

pschatzmann, 2022-07-15:
I suggest that you try to increase the i2s buffer

aqildad-create, 2022-07-15:
From where the i2s Buffer can be increase ?  and also what is the volume range of ( setVolume() )  minimum and maximum range ?

pschatzmann, 2022-07-15:
This information should be available in the class documentation:
https://pschatzmann.github.io/arduino-audio-tools/html/namespaceaudio__tools.html

Just search for the corresponding class and method...
The I2S configuration is in I2SConfig

aqildad-create, 2022-07-15:
//  i2sStream.begin(i2sStream.defaultConfig(RX_MODE));
   auto cfg = i2sStream.defaultConfig(RX_MODE);
   cfg.buffer_size =1024;
    i2sStream.begin(cfg);

in above when i  change the buffer_size more then 1024 then esp32 RESET CONTINOUSLY AFTER FEW SECONDS . what i am doing wrong ??

pschatzmann, 2022-07-15:
If you set the cfg.buffer_size you should also define the cfg. buffer_count. 

I assume that you just don't have enough RAM available. You can analyze the stack trace to figure out the root cause of your issue.

atomic14 has some nice expanation about these settings: https://www.atomic14.com/2021/04/20/esp32-i2s-dma-buf-len-buf-count.html

closed 2022-07-21 by pschatzmann