pschatzmann/arduino-audio-tools #263: Unable to play mp3 from SD-Card (no output?)

klein0r, 2022-07-15:
Thanks for the great libraries and examples! I bought a new ESP32-A1S-AudioKit (V2.2 A149) and tried to use one of your examples (with changed ``AUDIOKIT_BOARD`` flag). Everything looks fine, **but I have no sound output**. Tested with speakers and earphone jack.

Any suggestions what is wrong?

## File

I've created a Doorbell.mp3 and copied it to a 32GB exFAT SD-Card.

```
$ file ~/Doorbell.mp3 
Doorbell.mp3: Audio file with ID3 version 2.2.0, contains:MPEG ADTS, layer III, v1, 64 kbps, 44.1 kHz, Stereo MPEG ADTS, layer III, v1, 64 kbps, 44.1 kHz, Stereo
```

**Are there any requirements to the mp3? Like Stereo/Mono, kHz, ... ?**

## Example

```
[platformio]
description = Audio Example
default_envs = esp32dev

[env:esp32dev]
platform = espressif32@4.4.0
board = esp32dev
framework = arduino
build_flags = -DCORE_DEBUG_LEVEL=5 -Wno-unused-variable -Wno-unused-but-set-variable -Wno-unused-function -Wno-format-extra-args
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
lib_deps =
  Wire

  https://github.com/pschatzmann/arduino-audio-tools.git#v0.8.7
  https://github.com/pschatzmann/arduino-audiokit.git#v0.6.0
  https://github.com/pschatzmann/arduino-libhelix.git#v0.7
  https://github.com/greiman/SdFat.git#2.2.0

port = /dev/cu.usbserial-0001
```

```c
#define AUDIOKIT_BOARD 5
#define SD_FAT_TYPE 2 // 1 for FAT16/FAT32, 2 for exFAT, 3 for FAT16/FAT32 and exFAT.

#define USE_SDFAT
#define USE_HELIX // or USE_MAD
#include "AudioTools.h"
#include "AudioLibs/AudioKit.h"

const char *startFilePath = "/";
const char* ext = "mp3";
int speedMz = 10;
AudioSourceSdFat source(startFilePath, ext, PIN_AUDIO_KIT_SD_CARD_CS, speedMz);
AudioKitStream kit;
MP3DecoderHelix decoder; // or change to MP3DecoderMAD
AudioPlayer player(source, kit, decoder);

void next(bool, int, void*) {
   player.next();
}

void previous(bool, int, void*) {
   player.previous();
}

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // setup output
[... 233 lines cut ...]
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
libhelix -  -> invalid frame size: -800 / max: 1600
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
libhelix -  -> invalid frame size: -132 / max: 1600
[I] AudioCopy.h : 121 - StreamCopy::copy 884 -> 884 -> 884 bytes - in 1 hops
[W] AudioPlayer.h : 804 - -> timeout - moving to next stream
[W] AudioPlayer.h : 227 - -> nextStream: 1
[I] AudioPlayer.h : 332 - directory: '/'
[I] AudioPlayer.h : 339 - -> getFile: '': 1 - closed
[W] AudioPlayer.h : 236 - -> selectStream: 1 ''
[I] MetaDataID3.h : 551 - virtual void audio_tools::MetaDataID3::end()
```

pschatzmann, 2022-07-15:
Just read the readme or the FAQ of the [AudioKit](https://github.com/pschatzmann/arduino-audiokit) project. 
For platformio you find the necessary information also in the Wiki

klein0r, 2022-07-15:
Thanks a lot! I had to add ``build_flags = -DAUDIOKIT_BOARD=5`` to my platform.ini. It works now! 👍 

```
build_flags = -DAUDIOKIT_BOARD=5 -DCORE_DEBUG_LEVEL=5 -Wno-unused-variable -Wno-unused-but-set-variable -Wno-unused-function -Wno-format-extra-args
```

closed 2022-07-15 by klein0r