pschatzmann/arduino-audio-tools #269: esp32 reset while playing MP3 file and event handler occur esp reset

aqildad-create, 2022-07-21:
when ever event handler occur while audio is playing from SD card esp32 reset ... how to manage this situation ?

pschatzmann, 2022-07-21:
I am afraid that I don't understand what you talk about. 
There are no events in the audio framework.

aqildad-create, 2022-07-21:
yes i agree their is not event in audio frame work but in my application their is event handler like when button press events , so when even even occur in audio is playing then ESPreset

aqildad-create, 2022-07-21:
server.on("/update", HTTP_GET, [] (AsyncWebServerRequest *request)  I HAVE SOME EVENTS LIKE THIS IN MY APPLICATION . , ,

pschatzmann, 2022-07-21:
What do you mean with ESPreset ?

aqildad-create, 2022-07-21:
ESP32 RESET , SAME LIKE WHEN WE PRESS THE RESET BUTTON , PROGRAM START
AGAIN .

aqildad-create, 2022-07-21:
WHEN AUDIO IS PLAYING ALL EVENTS ARE WORKING PROPERLY , ALSO AUDIO IS PLAYING PROPERLY WHEN NO EVENT OCCUR . WHEN EVER EVENT OCCUR AND AUDIO IS PLAYING , THEN ESP32 RESET

pschatzmann, 2022-07-21:
You would need to analyze the stacktrace to figure out what's going on.
Most likely you just run out of memory...

aqildad-create, 2022-07-21:
How to check the stack trace ?  As per my observation it's not the memory
issue .

Basically whenever we jump from  main loop of arduino to handle the event
and if at that time audio is playing then  ESP32 crash .

First I need to stop the audio then handle the event . but i don't know how
to stop the audio before attending the event .

pschatzmann, 2022-07-21:
You could try to handle this with a global bool flag; In the event just set the flag to false and do not call any audio copy when the flag is false.

closed 2022-07-28 by pschatzmann