pschatzmann/arduino-audio-tools #270: OutputMixer Arduino environment flush() override compile error

n0bel, 2022-07-24:
```
C:\SyncedWorkSpace\Workspaces\ArduinoWorkspace\libraries\arduino-audio-tools-main\src/AudioTools/AudioOutput.h:474:10: error: 'void audio_tools::OutputMixer<T>::flush() [with T = short int]' marked 'override', but does not override
     void flush() override {
          ^
```
Solution:
Line 474 should be 
```
void flush() FLUSH_OVERRIDE {
```

pschatzmann, 2022-07-25:
This does not make any sense to me. 
Where would FLUSH_OVERRIDE be defined ?

n0bel, 2022-07-26:
Arduino ESP32 using Audio Kit V2.2 A210
I pulled the code from the 'main' branch 2 days ago.

FLUSH_OVERRIDE  is already defined in AudioConfig.h in the folowing way:

```
#ifdef IS_DESKTOP
#define USE_URL_ARDUINO
#define FLUSH_OVERRIDE override
#endif

#ifndef FLUSH_OVERRIDE
#define FLUSH_OVERRIDE 
#endif
```

Some other classes already use FLUSH_OVERRIDE

example  in AudioOutput.h AudioPrint, line 31
```
        void flush() FLUSH_OVERRIDE {
            write((const uint8_t*)tmp, tmpPos-1);
            tmpPos=0;
        }
```
example in AudioStreams.h AudioStreamX line 95
```
  virtual void flush() FLUSH_OVERRIDE {}
```

It appears to be used to include 'override' when compiled in desktop mode, and not otherwise.
I'm guessing the Stream class (Print) in desktop mode requires override.

Without using FLUSH_OVERRIDE or removing 'override' code with the Mixer included will not compile under Arduino.

pschatzmann, 2022-07-26:
Hmm, I can't reproduce your Arduino issue: I tested with RP2040 and ESP32 2.03

n0bel, 2022-07-26:
I'm using pre-2.0 arduino and esp32 1.0.6.   I probably need to point to the new repository..  And graduate to 2.0, or go down the platform io path. 

However the issue should not be platform dependent.

The problem only occurs if you use OutputMixer.  So the mixer example should show the problem.

If you would try using OutputMixer, and it works under version 2+, I'll upgrade my environment.  Up until now I've had no need to upgrade my environment to the 2+ versions.

As a side note I absolutely love this set of tools, its making my project much easier to implement.  No arduino based code comes close to supporting all you do.

pschatzmann, 2022-07-26:
I agree: I would like ESP32 1.0.6 to work as well.
I renamed flush() to flushMixer() to avoid any conflicts across different versions and Arduino implementations.

closed 2022-07-28 by pschatzmann