pschatzmann/arduino-audio-tools #277: Play certain mp3 in microsd card

luke2023, 2022-07-31:
Hello,I now have three different music:
Rickroll 
Sunnyday
Baby shark 
Is it possible to play a specific mp3 ?
I am making a music player
Using example:player-sd-a2dp
Thank you
Lin Che(KSAC)

pschatzmann, 2022-08-03:
If you check the documentation you should be able to find the relevant methods. 
See https://pschatzmann.github.io/arduino-audio-tools/html/classaudio__tools_1_1_audio_player.html

closed 2022-08-03 by pschatzmann

luke2023, 2022-08-11:
@pschatzmann Good afternoon!I noticed this website is offline?[image]

pschatzmann, 2022-08-11:
Thanks, I just restarted the server...

luke2023, 2022-08-11:
@pschatzmann Great!