pschatzmann/arduino-audio-tools #285: SD card recording

paralogicdr, 2022-08-02:
Hi everyone,
I searched the codes and examples but could not find how to record audio to SD card, would you help me

pschatzmann, 2022-08-02:
Just create a file and use it as copy destination

paralogicdr, 2022-08-02:
Can a StreamCopy class be used for this? Can I use filename as destination?

pschatzmann, 2022-08-02:
StreamCopy should work: The destination should be a File object or an EncodedAudioStream where you pass the file object.

paralogicdr, 2022-08-02:
Thank you Phil

closed 2022-08-03 by pschatzmann

pschatzmann, 2022-09-01:
I finally added some [file based recording example]()https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/streams-audiokit-sd-audiokit/streams-audiokit-sd-audiokit.ino

paralogicdr, 2022-09-02:
Thank you Phil, I admire your passion for your work

Android için Outlook<https://aka.ms/AAb9ysg> edinin

paralogicdr, 2022-09-12:
Hi Phil, I am working on my project, I am using ac101 based audiokit. I want to collect electrocardiogram (with ad8232 heart rate module) data to my boards io13 and ppg data(with  pulse sensor https://pulsesensor.com/) to io 4. My new problem is like this:

Android için Outlook<https://aka.ms/AAb9ysg> edinin

paralogicdr, 2022-09-12:
When I try to use webserver to send audio data, it seems ADC ports cannot read analog values. When I delete the server.Begin(kit,config) method on the sketch then sensors are able to collect readings. What can be the reason

Android için Outlook<https://aka.ms/AAb9ysg> edinin

pschatzmann, 2022-09-12:
I have never tried, but somebody has the same issue:
https://github.com/pschatzmann/arduino-audio-tools/issues/346

paralogicdr, 2022-09-12:
Thank you Phil, in my case I encounter this problem only server runs. When I try to Stream audiokit to audiokit there is no problem, do you have any suggestuons?

Android için Outlook<https://aka.ms/AAb9ysg> edinin

#346<https://github.com/pschatzmann/arduino-audio-tools/issues/346>

pschatzmann, 2022-09-12:
If you read https://github.com/espressif/arduino-esp32/issues/4782. They pretend that it is resolved with release 2.0.3.
Upgrade and maybe try other ADC pins...

paralogicdr, 2022-10-04:
> Thank you Phil, I tahinli this problem is about adc2 group pins. I will use io34

Hi Phil
I hope you are doing well
I have a question
As you know I want to make an electronic sthetoscope like device which can collect simultaneous heart sounds, ecg and pulse oxymetry.
Ecg and pulse oxymetry sensors are analog, and I can collect signals with ADC s without problem.
I can Stream sounds Via webserver.
İt would be so useful for me to be able to stream ecg and oxymetry signals as third and fourth channels of i2s
Is it possible? if it is possible, how can I do this
Thank you

Android için Outlook<https://aka.ms/AAb9ysg> edinin

If you read espressif/arduino-esp32#4782<https://github.com/espressif/arduino-esp32/issues/4782>. They pretend that it is resolved with release 2.0.3.

pschatzmann, 2022-10-05:
To keep things consistent, I would suggest that you try to create a second webserver (on a differerent port) and publish the Ecg and pulse oxymetry data (e.g. as csv or html values)

Here is a html example with my TinyHttp library:
https://github.com/pschatzmann/TinyHttp/blob/main/examples/serverStream/serverStream.ino

paralogicdr, 2022-10-05:
Thank you Phil
I try to upload your TinyHttp example sketches and I had library issues,” 'Log' was not declared in this scope “ like messages. Should I include more files?

Windows için Posta<https://go.microsoft.com/fwlink/?LinkId=550986> ile gönderildi

Kimden: Phil ***@***.***>
Gönderilme: 5 Ekim 2022 Çarşamba 10:24
Kime: ***@***.***>
Bilgi: ***@***.***>; ***@***.***>
Konu: Re: [pschatzmann/arduino-audio-tools] SD card recording (Issue #285)

To keep things consistent, I would suggest that you try to create a second webserver (on a differerent port) and publish the Ecg and pulse oxymetry data (e.g. as csv or html values)

Here is a html example with my TinyHttp library:
https://github.com/pschatzmann/TinyHttp/blob/main/examples/serverStream/serverStream.ino

pschatzmann, 2022-10-05:
Did you install the TinyHttp library ?

paralogicdr, 2022-10-05:
Yes, I did.

Windows için Posta<https://go.microsoft.com/fwlink/?LinkId=550986> ile gönderildi

Kimden: Phil ***@***.***>
Gönderilme: 5 Ekim 2022 Çarşamba 16:33
Kime: ***@***.***>
Bilgi: ***@***.***>; ***@***.***>
Konu: Re: [pschatzmann/arduino-audio-tools] SD card recording (Issue #285)

Did you install the TinyHttp library ?

pschatzmann, 2022-10-05:
Oh, I see: Log has been renamed to HttpLogger

paralogicdr, 2022-10-05:
Thank you Phil,
There can be a similar issue with “stream”

Windows için Posta<https://go.microsoft.com/fwlink/?LinkId=550986> ile gönderildi

Kimden: Phil ***@***.***>
Gönderilme: 5 Ekim 2022 Çarşamba 16:44
Kime: ***@***.***>
Bilgi: ***@***.***>; ***@***.***>
Konu: Re: [pschatzmann/arduino-audio-tools] SD card recording (Issue #285)

Oh, I see: Log has been renamed to HttpLogger

paralogicdr, 2022-10-05:
I think you should correct the class definition for httpLogger

'class tinyhttp::HttpLoggerClass' has no member named 'setLogger'; did you mean 'setLevel'?
Windows için Posta<https://go.microsoft.com/fwlink/?LinkId=550986> ile gönderildi

Kimden: Phil ***@***.***>
Gönderilme: 5 Ekim 2022 Çarşamba 16:44
Kime: ***@***.***>
Bilgi: ***@***.***>; ***@***.***>
Konu: Re: [pschatzmann/arduino-audio-tools] SD card recording (Issue #285)

Oh, I see: Log has been renamed to HttpLogger

pschatzmann, 2022-10-05:
Your right: I have updated all example and tested the log server example agin...

pschatzmann, 2022-10-05:
I think you need to upgrade the AudioTools as well...

pschatzmann, 2022-10-05:
Oh, I see. GET is defined twice, so you need to specify which GET you mean: In this case if would be tinyhttp::GET

paralogicdr, 2022-10-11:
Thank you Phil, I tahinli this problem is about adc2 group pins. I will use io34

Android için Outlook<https://aka.ms/AAb9ysg> edinin