pschatzmann/arduino-audio-tools #287: need apposite of streams-i2s-webserver_wav

aqildad-create, 2022-08-05:
streams-i2s-webserver_wav This example is working . but i need opposite to this means from webserver(mobile microphone ) to esp32-i2s-speaker . 

means i want to play audio from mobile microphone via webserver to i2s speaker attached with ESP32 . do you have any related example of this ? or how this will achieved ?

pschatzmann, 2022-08-05:
Wouldn't this be covered by the regular streaming examples: e.g. https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-stream/streams-url_wav-i2s

aqildad-create, 2022-08-06:
[I] I2SConfig.h : 95 - rx/tx mode: TX
[I] I2SConfig.h : 96 - port_no: 0
[I] I2SConfig.h : 97 - is_master: Master
[I] I2SConfig.h : 98 - sample rate: 16000
[I] I2SConfig.h : 99 - bits per sample: 32
[I] I2SConfig.h : 100 - number of channels: 1
[I] I2SConfig.h : 101 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 103 - auto_clear:1
[I] I2SConfig.h : 110 - buffer_count:10
[I] I2SConfig.h : 111 - buffer_size:512
[I] I2SConfig.h : 115 - pin_bck: 27
[I] I2SConfig.h : 116 - pin_ws: 26
[I] I2SConfig.h : 117 - pin_data: 25
[I] URLStream.h : 78 - virtual bool audio_tools::URLStreamDefault::begin(const char*, const char*, MethodID, const char*, const char*): http://192.168.1.37:12101/api/text-to-speech?play=false
[I] Url.h : 73 - Url::parse
[I] Url.h : 109 - url->http://192.168.1.37:12101/api/text-to-speech?play=false
[I] Url.h : 110 - host->192.168.1.37
[I] Url.h : 111 - protocol->http
[I] Url.h : 112 - path->/api/text-to-speech?play=false
[I] Url.h : 113 - port->12101
.
[I] URLStream.h : 253 - WiFiClient
[I] HttpRequest.h : 193 - process connecting to host 192.168.1.37 port 12101
[I] HttpRequest.h : 181 - connected 0 timeout 60000
[E] HttpRequest.h : 196 - Connect failed

why its connect failed ?? i tried multiple time and internet is working

pschatzmann, 2022-08-06:
Just use a URL to which you have access. 
192.168.1.37 is a local address any you will need to install [rhasspy](https://rhasspy.readthedocs.io/en/latest/) an one of your machines and change the url.

There are plenty of other examples e.g. https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-stream which shows decodes a public mp3 url

aqildad-create, 2022-08-06:
@file url_raw-I2S_external_dac.ino i tried this one . it connect but no audio on i2s speaker ..  pin out of i2s dac is correct because audio from sd card is working.

starting I2S...
[I] I2SConfig.h : 95 - rx/tx mode: TX
[I] I2SConfig.h : 96 - port_no: 0
[I] I2SConfig.h : 97 - is_master: Master
[I] I2SConfig.h : 98 - sample rate: 8000
[I] I2SConfig.h : 99 - bits per sample: 16
[I] I2SConfig.h : 100 - number of channels: 2
[I] I2SConfig.h : 101 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 103 - auto_clear:1
[I] I2SConfig.h : 110 - buffer_count:10
[I] I2SConfig.h : 111 - buffer_size:512
[I] I2SConfig.h : 115 - pin_bck: 27
[I] I2SConfig.h : 116 - pin_ws: 26
[I] I2SConfig.h : 117 - pin_data: 25
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
.
Copy ended

aqildad-create, 2022-08-06:
> Just use a URL to which you have access. 192.168.1.37 is a local address any you will need to install [rhasspy](https://rhasspy.readthedocs.io/en/latest/) an one of your machines and change the url.
> 
> There are plenty of other examples e.g. https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-stream/streams-url_mp3_helix-i2s which shows decodes a public mp3 url using the helix decoder
> 
> I suggest that you test the URLs first with your web browser...

This work , but practically its not applicable because internet slow issue , we cannot listen the voices very clearly . do you have any solution for this ?

aqildad-create, 2022-08-06:
> Just use a URL to which you have access. 192.168.1.37 is a local address any you will need to install [rhasspy](https://rhasspy.readthedocs.io/en/latest/) an one of your machines and change the [...]

This work , but practically its not applicable because internet slow issue , we cannot listen the voices very clearly . do you have any solution for this ?

aqildad-create, 2022-08-06:
> 

This work , but practically its not applicable because internet slow issue , we cannot listen the voices very clearly . do you have any solution for this ?

aqildad-create, 2022-08-06:
> @file url_raw-I2S_external_dac.ino i tried this one . it connect but no audio on i2s speaker .. pin out of i2s dac is correct because audio from sd card is working.
>
> starting I2S... [I] I2SConfig.h [...]

esp32 Reset after this . why is this i change the FLASH SIZE TO huge ap , may be issue with memory , but it reset with this example , other example like play mp3 from url is working but why this application is not working

aqildad-create, 2022-08-06:
i change 1024 to 512 , i though may the buffer reset the ESP in this StreamCopy copier(i2s, music, 512); // copy music to i2s 
but still ESP32 reset . Below is the behavior .
 
further i run raw-i2s and url-mp3-i2s is successfully working .

 i am confuse why only url-raw-i2s example is not working . pls guide why is this because if their is any issue in ESP or memory issue why only this example is not working all other example i try is working . 

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
.[I] URLStream.h : 78 - virtual bool audio_tools::URLStreamDefault::begin(const char*, const char*, MethodID, const char*, const char*): https://pschatzmann.github.io/arduino-audio-tools/resources/audio-8000.raw
[I] Url.h : 73 - Url::parse
[I] Url.h : 109 - url->https://pschatzmann.github.io/arduino-audio-tools/resources/audio-8000.raw
[I] Url.h : 110 - host->pschatzmann.github.io
[I] Url.h : 111 - protocol->https
[I] Url.h : 112 - path->/arduino-audio-tools/resources/audio-8000.raw
[I] Url.h : 113 - port->443
[I] URLStream.h : 248 - WiFiClientSecure
[I] URLStream.h : 248 - WiFiClientSecure
[I] HttpRequest.h : 193 - process connecting to host pschatzmann.github.io port 443
[I] HttpRequest.h : 181 - connected 1 timeout 60000
[I] HttpRequest.h : 204 - Free heap: 207792
[I] HttpHeader.h : 357 - HttpRequestHeader::setValues - path: /arduino-audio-tools/resources/audio-8000.raw
[I] HttpHeader.h : 254 - HttpHeader::write
[I] HttpHeader.h : 366 - HttpRequestHeader::write1stLine
[I] HttpHeader.h : 381 - -> GET /arduino-audio-tools/resources/audio-8000.raw HTTP/1.1
[I] HttpHeader.h : 191 -  -> Host: pschatzmann.github.io 
[I] HttpHeader.h : 191 -  -> Connection: close 
[I] HttpHeader.h : 191 -  -> Accept: */* 
[I] HttpHeader.h : 297 -  -> <CR LF> 
[I] HttpRequest.h : 227 - Request written ... waiting for reply
[I] HttpHeader.h : 224 - HttpHeader::read
[W] HttpHeader.h : 231 - Waiting for data...
[I] HttpHeader.h : 161 - HttpHeader::readLine -> HTTP/1.1 200 OK
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Connection: close
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Content-Length: 6212544
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Server: GitHub.com
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Content-Type: application/octet-stream
[I] HttpHeader.h : 161 - HttpHeader::readLine -> permissions-policy: interest-cohort=()
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Last-Modified: Thu, 04 Aug 2022 21:10:20 GMT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Access-Control-Allow-Origin: *
[I] HttpHeader.h : 161 - HttpHeader::readLine -> ETag: "62ec35bc-5ecbc0"
[I] HttpHeader.h : 161 - HttpHeader::readLine -> expires: Sat, 06 Aug 2022 14:56:43 GMT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Cache-Control: max-age=600
[I] HttpHeader.h : 161 - HttpHeader::readLine -> x-proxy-cache: MISS
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-GitHub-Request-Id: B962:7C28:499787:9E3787:62EE7ED3
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Accept-Ranges: bytes
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Age: 329
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Date: Sat, 06 Aug 2022 14:52:12 GMT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Via: 1.1 varnish
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Served-By: cache-qpg1253-QPG
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Cache: HIT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Cache-Hits: 0
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Timer: S1659797533.769650,VS0,VE2
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Vary: Accept-Encoding
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Fastly-Request-ID: 3fef3d7bdcc57418b326cb559fb99db203ab5a83
[I] HttpHeader.h : 161 - HttpHeader::readLine -> 
[I] URLStream.h : 93 - size: 6212544

starting I2S...
[I] I2SConfig.h : 95 - rx/tx mode: TX
[I] I2SConfig.h : 96 - port_no: 0
[I] I2SConfig.h : 97 - is_master: Master
[... 37 lines cut ...]
Backtrace:0x400dc3c5:0x3ffb27100x400d409d:0x3ffb2750 0x400d4145:0x3ffb2780 0x400d595c:0x3ffb27a0 0x400d5a77:0x3ffb27c0 0x400d5b06:0x3ffb2800 0x400d9c3d:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
.[I] URLStream.h : 78 - virtual bool audio_tools::URLStreamDefault::begin(const char*, const char*, MethodID, const char*, const char*): https://pschatzmann.github.io/arduino-audio-tools/resources/audio-8000.raw
[I] Url.h : 73 - Url::parse
[I] Url.h : 109 - url->https://pschatzmann.github.io/arduino-audio-tools/resources/audio-8000.raw
[I] Url.h : 110 - host->pschatzmann.github.io
[I] Url.h : 111 - protocol->https
[I] Url.h : 112 - path->/arduino-audio-tools/resources/audio-8000.raw
[I] Url.h : 113 - port->443
[I] URLStream.h : 248 - WiFiClientSecure
[I] URLStream.h : 248 - WiFiClientSecure
[I] HttpRequest.h : 193 - process connecting to host pschatzmann.github.io port 443

pschatzmann, 2022-08-06:
You need to analyse the stacktrace: https://github.com/me-no-dev/EspExceptionDecoder

aqildad-create, 2022-08-06:
Ok I edit bool variable in main loop to stop copying once complete . With this  ESP reset stop Restart but still no audio out put on I2S speaker . WHY IS THIS ?

Below is edited code and Uart out put for reference . 

bool stopp= false;
// Arduino loop - repeated processing: copy input stream to output stream
void loop() {
if(stopp==false)
{
  int len = copier.copy();
  if (len){
      Serial.print(".");
  } else {
      i2s.end();
      Serial.println("\nCopy ended");
      delay(10000);
      stopp=true;
  }
}

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
.[I] URLStream.h : 78 - virtual bool audio_tools::URLStreamDefault::begin(const char*, const char*, MethodID, const char*, const char*): https://pschatzmann.github.io/arduino-audio-tools/resources/audio-8000.raw
[I] Url.h : 73 - Url::parse
[I] Url.h : 109 - url->https://pschatzmann.github.io/arduino-audio-tools/resources/audio-8000.raw
[I] Url.h : 110 - host->pschatzmann.github.io
[I] Url.h : 111 - protocol->https
[I] Url.h : 112 - path->/arduino-audio-tools/resources/audio-8000.raw
[I] Url.h : 113 - port->443
[I] URLStream.h : 248 - WiFiClientSecure
[I] URLStream.h : 248 - WiFiClientSecure
[I] HttpRequest.h : 193 - process connecting to host pschatzmann.github.io port 443
[I] HttpRequest.h : 181 - connected 1 timeout 60000
[I] HttpRequest.h : 204 - Free heap: 206592
[I] HttpHeader.h : 357 - HttpRequestHeader::setValues - path: /arduino-audio-tools/resources/audio-8000.raw
[I] HttpHeader.h : 254 - HttpHeader::write
[I] HttpHeader.h : 366 - HttpRequestHeader::write1stLine
[I] HttpHeader.h : 381 - -> GET /arduino-audio-tools/resources/audio-8000.raw HTTP/1.1
[I] HttpHeader.h : 191 -  -> Host: pschatzmann.github.io 
[I] HttpHeader.h : 191 -  -> Connection: close 
[I] HttpHeader.h : 191 -  -> Accept: */* 
[I] HttpHeader.h : 297 -  -> <CR LF> 
[I] HttpRequest.h : 227 - Request written ... waiting for reply
[I] HttpHeader.h : 224 - HttpHeader::read
[W] HttpHeader.h : 231 - Waiting for data...
[I] HttpHeader.h : 161 - HttpHeader::readLine -> HTTP/1.1 200 OK
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Connection: close
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Content-Length: 6212544
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Server: GitHub.com
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Content-Type: application/octet-stream
[I] HttpHeader.h : 161 - HttpHeader::readLine -> permissions-policy: interest-cohort=()
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Last-Modified: Thu, 04 Aug 2022 21:10:20 GMT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Access-Control-Allow-Origin: *
[I] HttpHeader.h : 161 - HttpHeader::readLine -> ETag: "62ec35bc-5ecbc0"
[I] HttpHeader.h : 161 - HttpHeader::readLine -> expires: Sat, 06 Aug 2022 15:31:34 GMT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Cache-Control: max-age=600
[I] HttpHeader.h : 161 - HttpHeader::readLine -> x-proxy-cache: MISS
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-GitHub-Request-Id: F88C:562C:41C5:157524:62EE86FE
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Accept-Ranges: bytes
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Age: 104
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Date: Sat, 06 Aug 2022 15:23:18 GMT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Via: 1.1 varnish
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Served-By: cache-qpg1223-QPG
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Cache: HIT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Cache-Hits: 0
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Timer: S1659799399.568862,VS0,VE2
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Vary: Accept-Encoding
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Fastly-Request-ID: 56b987dbe1fa648fa161ca4848260ae4038681a3
[I] HttpHeader.h : 161 - HttpHeader::readLine -> 
[I] URLStream.h : 93 - size: 6212544

starting I2S...
[I] I2SConfig.h : 95 - rx/tx mode: TX
[I] I2SConfig.h : 96 - port_no: 0
[I] I2SConfig.h : 97 - is_master: Master
[I] I2SConfig.h : 98 - sample rate: 8000
[I] I2SConfig.h : 99 - bits per sample: 16
[I] I2SConfig.h : 100 - number of channels: 2
[I] I2SConfig.h : 101 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 103 - auto_clear:1
[I] I2SConfig.h : 110 - buffer_count:10
[I] I2SConfig.h : 111 - buffer_size:512
[I] I2SConfig.h : 115 - pin_bck: 27
[I] I2SConfig.h : 116 - pin_ws: 26
[I] I2SConfig.h : 117 - pin_data: 25
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
.
Copy ended

pschatzmann, 2022-08-07:
The [example](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-url_raw-i2s/streams-url_raw-i2s.ino) has been corrected

aqildad-create, 2022-08-11:
Sir i test the above corrected example , it run but no audio out . 

My observations for this example is  

1: copy ended in 2or 3 second . i check the audio-8000 raw audio file in audacity its approximate 1minute length file ,  then how in this example copy end in 1-2 seconds . .  

2: why no audio out here , i2s channel is ok because file from sd card is playing ? and internet mp3 player is working . then why no audio out here ?

 ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
.[I] URLStream.h : 78 - virtual bool audio_tools::URLStreamDefault::begin(const char*, const char*, MethodID, const char*, const char*): https://pschatzmann.github.io/arduino-audio-tools/resources/audio-8000.raw
[I] Url.h : 73 - Url::parse
[I] Url.h : 109 - url->https://pschatzmann.github.io/arduino-audio-tools/resources/audio-8000.raw
[I] Url.h : 110 - host->pschatzmann.github.io
[I] Url.h : 111 - protocol->https
[I] Url.h : 112 - path->/arduino-audio-tools/resources/audio-8000.raw
[I] Url.h : 113 - port->443
[I] URLStream.h : 248 - WiFiClientSecure
[I] URLStream.h : 248 - WiFiClientSecure
[I] HttpRequest.h : 193 - process connecting to host pschatzmann.github.io port 443
[I] HttpRequest.h : 181 - connected 1 timeout 60000
[I] HttpRequest.h : 204 - Free heap: 207348
[I] HttpHeader.h : 357 - HttpRequestHeader::setValues - path: /arduino-audio-tools/resources/audio-8000.raw
[I] HttpHeader.h : 254 - HttpHeader::write
[I] HttpHeader.h : 366 - HttpRequestHeader::write1stLine
[I] HttpHeader.h : 381 - -> GET /arduino-audio-tools/resources/audio-8000.raw HTTP/1.1
[I] HttpHeader.h : 191 -  -> Host: pschatzmann.github.io 
[I] HttpHeader.h : 191 -  -> Connection: close 
[I] HttpHeader.h : 191 -  -> Accept: */* 
[I] HttpHeader.h : 297 -  -> <CR LF> 
[I] HttpRequest.h : 227 - Request written ... waiting for reply
[I] HttpHeader.h : 224 - HttpHeader::read
[W] HttpHeader.h : 231 - Waiting for data...
[I] HttpHeader.h : 161 - HttpHeader::readLine -> HTTP/1.1 200 OK
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Connection: close
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Content-Length: 6212544
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Server: GitHub.com
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Content-Type: application/octet-stream
[I] HttpHeader.h : 161 - HttpHeader::readLine -> permissions-policy: interest-cohort=()
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Last-Modified: Wed, 10 Aug 2022 13:08:03 GMT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Access-Control-Allow-Origin: *
[I] HttpHeader.h : 161 - HttpHeader::readLine -> ETag: "62f3adb3-5ecbc0"
[I] HttpHeader.h : 161 - HttpHeader::readLine -> expires: Thu, 11 Aug 2022 04:31:38 GMT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Cache-Control: max-age=600
[I] HttpHeader.h : 161 - HttpHeader::readLine -> x-proxy-cache: MISS
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-GitHub-Request-Id: F466:7960:193DDF:1D1D87:62F483D2
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Accept-Ranges: bytes
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Age: 289
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Date: Thu, 11 Aug 2022 04:37:32 GMT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Via: 1.1 varnish
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Served-By: cache-qpg1237-QPG
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Cache: HIT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Cache-Hits: 0
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Timer: S1660192653.623433,VS0,VE1
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Vary: Accept-Encoding
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Fastly-Request-ID: 8a3a6f6f2dc70c1bc738f23b5eb5a670734af6f7
[I] HttpHeader.h : 161 - HttpHeader::readLine -> 
[I] URLStream.h : 93 - size: 6212544

starting I2S...
[I] I2SConfig.h : 95 - rx/tx mode: TX
[I] I2SConfig.h : 96 - port_no: 0
[I] I2SConfig.h : 97 - is_master: Master
[I] I2SConfig.h : 98 - sample rate: 8000
[I] I2SConfig.h : 99 - bits per sample: 16
[I] I2SConfig.h : 100 - number of channels: 2
[I] I2SConfig.h : 101 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 103 - auto_clear:1
[I] I2SConfig.h : 110 - buffer_count:10
[I] I2SConfig.h : 111 - buffer_size:512
[I] I2SConfig.h : 115 - pin_bck: 27
[I] I2SConfig.h : 116 - pin_ws: 26
[I] I2SConfig.h : 117 - pin_data: 25
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
.
Copy ended

pschatzmann, 2022-08-11:
Try to add a delay before the i2s is closed. 
It seems that it is closed while it is still playing...

aqildad-create, 2022-08-11:
ok i will add delay and will up date here  but why was audio not playing on i2s ? and why was copy ended in 2-3 second of program start ?

aqildad-create, 2022-08-12:
if added the delay , still result is same . no audio out and copy ended in few seconds of start . pls check this , and have you tested this example sir please ? 

void loop() {
  int len = copier.copy();
  if (len){
      Serial.print(".");
  } else {
      i2s.end();
      delay(1000);
      Serial.println("\nCopy ended");
      stop();
  }
}

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
.[I] URLStream.h : 78 - virtual bool audio_tools::URLStreamDefault::begin(const char*, const char*, MethodID, const char*, const char*): https://pschatzmann.github.io/arduino-audio-tools/resources/audio-8000.raw
[I] Url.h : 73 - Url::parse
[I] Url.h : 109 - url->https://pschatzmann.github.io/arduino-audio-tools/resources/audio-8000.raw
[I] Url.h : 110 - host->pschatzmann.github.io
[I] Url.h : 111 - protocol->https
[I] Url.h : 112 - path->/arduino-audio-tools/resources/audio-8000.raw
[I] Url.h : 113 - port->443
[I] URLStream.h : 248 - WiFiClientSecure
[I] URLStream.h : 248 - WiFiClientSecure
[I] HttpRequest.h : 193 - process connecting to host pschatzmann.github.io port 443
[I] HttpRequest.h : 181 - connected 1 timeout 60000
[I] HttpRequest.h : 204 - Free heap: 207248
[I] HttpHeader.h : 357 - HttpRequestHeader::setValues - path: /arduino-audio-tools/resources/audio-8000.raw
[I] HttpHeader.h : 254 - HttpHeader::write
[I] HttpHeader.h : 366 - HttpRequestHeader::write1stLine
[I] HttpHeader.h : 381 - -> GET /arduino-audio-tools/resources/audio-8000.raw HTTP/1.1
[I] HttpHeader.h : 191 -  -> Host: pschatzmann.github.io 
[I] HttpHeader.h : 191 -  -> Connection: close 
[I] HttpHeader.h : 191 -  -> Accept: */* 
[I] HttpHeader.h : 297 -  -> <CR LF> 
[I] HttpRequest.h : 227 - Request written ... waiting for reply
[I] HttpHeader.h : 224 - HttpHeader::read
[W] HttpHeader.h : 231 - Waiting for data...
[I] HttpHeader.h : 161 - HttpHeader::readLine -> HTTP/1.1 200 OK
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Connection: close
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Content-Length: 6212544
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Server: GitHub.com
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Content-Type: application/octet-stream
[I] HttpHeader.h : 161 - HttpHeader::readLine -> permissions-policy: interest-cohort=()
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Last-Modified: Wed, 10 Aug 2022 13:08:03 GMT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Access-Control-Allow-Origin: *
[I] HttpHeader.h : 161 - HttpHeader::readLine -> ETag: "62f3adb3-5ecbc0"
[I] HttpHeader.h : 161 - HttpHeader::readLine -> expires: Fri, 12 Aug 2022 04:28:49 GMT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Cache-Control: max-age=600
[I] HttpHeader.h : 161 - HttpHeader::readLine -> x-proxy-cache: MISS
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-GitHub-Request-Id: C408:2EEF:5D533C:9AE1EB:62F5D4A9
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Accept-Ranges: bytes
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Age: 0
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Date: Fri, 12 Aug 2022 04:44:28 GMT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Via: 1.1 varnish
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Served-By: cache-qpg1271-QPG
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Cache: HIT
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Cache-Hits: 0
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Timer: S1660279468.994014,VS0,VE234
[I] HttpHeader.h : 161 - HttpHeader::readLine -> Vary: Accept-Encoding
[I] HttpHeader.h : 161 - HttpHeader::readLine -> X-Fastly-Request-ID: 9bbf1bce4864b29b445088b637ae6bd74f3854f4
[I] HttpHeader.h : 161 - HttpHeader::readLine -> 
[I] URLStream.h : 93 - size: 6212544

starting I2S...
[I] I2SConfig.h : 95 - rx/tx mode: TX
[I] I2SConfig.h : 96 - port_no: 0
[I] I2SConfig.h : 97 - is_master: Master
[I] I2SConfig.h : 98 - sample rate: 8000
[I] I2SConfig.h : 99 - bits per sample: 16
[I] I2SConfig.h : 100 - number of channels: 2
[I] I2SConfig.h : 101 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 103 - auto_clear:1
[I] I2SConfig.h : 110 - buffer_count:10
[I] I2SConfig.h : 111 - buffer_size:512
[I] I2SConfig.h : 115 - pin_bck: 27
[I] I2SConfig.h : 116 - pin_ws: 26
[I] I2SConfig.h : 117 - pin_data: 25
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
.[I] AudioCopy.h : 121 - StreamCopy::copy 347 -> 347 -> 347 bytes - in 1 hops
.
Copy ended

pschatzmann, 2022-08-13:
Let me try to explain again: The issue is that i2s.end(); is called while the music is still playing so it will stop immediatley and has no chance to finish. 
Adding the delay after the i2s.end() will not help. It must be before!

closed 2022-08-16 by pschatzmann