pschatzmann/arduino-audio-tools #288: download mp3 from url to sd

aqildad-create, 2022-08-05:
your example which play mp3 from SD card is working but now further i Need example which download mp3 from URL to SD card . 

Do you have any relevant example or how can we achieve this ?

pschatzmann, 2022-08-05:
No I don't by I would expect that you can just copy from a URLStream to a File and don't forget to close the file at the end....

aqildad-create, 2022-08-05:
how to copy URLStream to a file ? Which example is this ?

pschatzmann, 2022-08-05:
There is no example: I suggest that you read a tutorial about Arduino and Files first. 
You can try to copy using the StreamCopy class or simply by calling readBytes(uint8_t buffer, size_t len) and write(uint8_t buffer, size_t len) in a loop...

pschatzmann, 2022-08-05:
I am neither providing MySQL nor basic ESP32 support.  Please use Google to find the answer yourself! 
There you should find e.g. https://esp32io.com/tutorials/esp32-mysql

closed 2022-08-16 by pschatzmann