pschatzmann/arduino-audio-tools #291: FLAC Support

faculapro, 2022-08-06:
Hello, First of all thanks for this great library.
I would like to know how to properly set up flac decode on a ESP32. I've tried the flac stream example but it gives me an error with opus lib (I've already installed it).
Also, it is possible to add flac decode on the audioplayer?
Thanks

Here is the error when compiling the example:
```
Arduino: 1.8.16 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None"

In file included from C:\Users\facun\Documents\Arduino\libraries\arduino-libflac-main\src/flac.h:13,

                 from C:\Users\facun\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioCodecs/CodecFLAC.h:13,

                 from C:\Users\facun\Documents\Arduino\libraries\arduino-audio-tools-main\examples\examples-stream\streams-url_flac-i2s\streams-url_flac-i2s.ino:12:

C:\Users\facun\Documents\Arduino\libraries\arduino-libopus-main\src/opus.h:1:6: error: too many decimal points in number

 opus-1.3.1/include/opus.h

      ^~~~~

C:\Users\facun\Documents\Arduino\libraries\arduino-libopus-main\src/opus.h:1:1: error: 'opus' does not name a type; did you mean 'open'?

 opus-1.3.1/include/opus.h

 ^~~~

 open

Multiple libraries were found for "WiFi.h"

 Used: C:\Users\facun\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\WiFi

 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi

exit status 1

Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

pschatzmann, 2022-08-07:
The library relies on symbolic links which work fine in linux and osx. In Windows this can be problematic so I suggest that you try: https://gist.github.com/huenisys/1efb64e57c37cfab7054c65702588fce

Alternatively you can replace all symlinks with proper includes that include the file that you find in the symlink. If you go this way I will be more then happy to accept a pull request...

pschatzmann, 2022-08-07:
I removed the symlinks and committed the change.
Please update the library and try again...

closed 2022-08-16 by pschatzmann