pschatzmann/arduino-audio-tools #292: Read A2dp volume problem

luke2023, 2022-08-08:
Hello,Thank you for your Library!
I am now having a problem of reading the volume of the a2dp player
Code:
`void loop(){
	player.copy();
	Serial.println(player.volume());
}
`
Out put:
ovf
ovf
ovf⋯⋯⋯⋯
I don't understand what's wrong .
Thank you
LinChe

pschatzmann, 2022-08-08:
For the time being, try to call e.g. player.setVolume(0.6); in the setup, this should avoid the problem.

luke2023, 2022-08-08:
@pschatzmann thank you so much for the work!I found  another problem is that if I set it to 0.05 in setup, I can't use my earphones to set it very loud ,max is around 0.2,Thank You!

luke2023, 2022-08-08:
Going to make a open source watch with Bluetooth player![image]

pschatzmann, 2022-08-08:
I suggest that you share your project here 
https://github.com/pschatzmann/arduino-audio-tools/discussions/categories/show-and-tell

luke2023, 2022-08-08:
@pschatzmann Thank you for your suggestions! Although I set it to 0.6 it still returns 'ovf' thanks!
LinChe

luke2023, 2022-08-08:
@pschatzmann Ok

pschatzmann, 2022-08-09:
the player.volume() issue has been corrected

luke2023, 2022-08-10:
@pschatzmann Thank you!I am looking forward to the update!

closed 2022-08-16 by pschatzmann

luke2023, 2022-08-19:
@pschatzmann good luck!

luke2023, 2022-08-24:
@pschatzmann Alsome!Thank you for the update!I'll try it next time! 
LinChe (KSAC)