pschatzmann/arduino-audio-tools #297: Compile error in AudioKit.h - missing 'setSampleRate'?

ryan-hyer, 2022-08-08:
In file included from C:\[...]\streams-url_mp3-audiokit.ino:21:
C:\[...]\arduino-audio-tools-0.8.7\src/AudioLibs/AudioKit.h: In member function 'virtual void audio_tools::AudioKitStream::setAudioInfo(audio_tools::AudioBaseInfo)':
C:\[...]\arduino-audio-tools-0.8.7\src/AudioLibs/AudioKit.h:260:11: error: 'class AudioKit' has no member named 'setSampleRate'

pschatzmann, 2022-08-09:
The example streams-url_mp3-audiokit.ino is compiling w/o errors.
Did you make any changes to the sketch ?

I confirm that AudioKit has no setSampleRate() method and this is correct.

ryan-hyer, 2022-08-09:
I only changed the ssid and password of the URLStream declaration. But, just in case I messed up somewhere else, I've re-downloaded and re-included the latest versions of all the associated libraries from scratch. And now it compiles fine. My bad, apparently.

I still get runtime errors in streams-url_mp3-audiokit.ino ("wifi:Association refused temporarily, comeback time 1024 mSec" repeating indefinitely), but it does compile successfully!

pschatzmann, 2022-08-09:
I guess there are some issues with your local Wifi, that have nothing to do with my code.
Just try to bring up the Wifi with your own code....

closed 2022-08-09 by ryan-hyer