pschatzmann/arduino-audio-tools #301: @file streams-url_raw-serial.ino stop serial when file end

aqildad-create, 2022-08-11:
@file streams-url_raw-serial.ino

This example continuously send data to serial plotter . How can we stop when the url file end or complete  ?

aqildad-create, 2022-08-11:
also can we stop in the mid and then resume later ??

pschatzmann, 2022-08-13:
Sure. Just make sure that the copy in the loop is not executed if you want pause.
If you are new in programming you can consider to use the AudioPlayer class.

aqildad-create, 2022-08-15:
Is serial data  same as data present in  raw file or its factor to some value in order to plot in serial-PLOTER . 

i want to compare this Serial data with original raw data present in file . . How to get back original data  ?

pschatzmann, 2022-08-15:
It is just convertering the binary values to a string so that you can read it. If you want to see the binary data instead you can use a HexDumpStream as output instead.