pschatzmann/arduino-audio-tools #313: Pause/play function in Streamcopy

gnpaone, 2022-08-26:
Is there any way to add pause/play functionality in streamcopy copier?

pschatzmann, 2022-08-26:
You can easily implement this functionality just with a bool flag (eg. bool active) in your sketch. In the loop you just use 
if (active) copier.copy();

gnpaone, 2022-08-26:
Oh I will try that. By the way I have one more issue too, I have this added to your player_sd_a2dp_audiokit example:
```bool player_active = true;
bool player_state = false;
bool a2dp_state = false;

void next(bool, int, void*) {
   if (player_active) {
      player.next();
   } else {
      a2dp_sink.next();
   }
}

void previous(bool, int, void*) {
   if (player_active) {
      player.previous();
   } else {
      a2dp_sink.previous();
   }
}

void changeState(bool, int, void*) {
  if (player_active) {
      player_state = !player_state;
      Serial.println(player_state);
      if (player_state){
          player.stop();
      } else {
          player.play();
      }
  } else {
      a2dp_state = !a2dp_state;
      Serial.println(a2dp_state);
      if (a2dp_state){
          a2dp_sink.pause();
      } else {
          a2dp_sink.play();
      }
  }
}```
and mode function as
```
.
.
.
player.setVolume(0.7);
player.begin();
Serial.println("Player started");
kit.addAction(PIN_KEY2, changeState);
kit.addAction(PIN_KEY4, next);
kit.addAction(PIN_KEY3, previous);
```
if I comment out `kit.addAction(PIN_KEY2, changeState);` the functions work fine and song gets played but player.stop() and player.play() doesn't play the song and gives timeout in serial monitor.

gnpaone, 2022-08-28:
Hi @pschatzmann, did you replicate my problem? did you find the solution? Please help! 🥲

gnpaone, 2022-08-28:
Also one more issue is I need Bluetooth a2dp to work as both source and sink at the same time, is that possible?

gnpaone, 2022-08-28:
And [streams-looper](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/sandbox/streams-looper) example is no longer working

pschatzmann, 2022-08-28:
> Oh I will try that. By the way I have one more issue too, I have this added to your player_sd_a2dp_audiokit example:
> 
> ```
> bool player_state = false;
> bool a2dp_state = false;
> 
> void next(bool, int, void*) {
>    if (player_active) {
>       player.next();
>    } else {
>       a2dp_sink.next();
>    }
> }
> 
> void previous(bool, int, void*) {
>    if (player_active) {
>       player.previous();
>    } else {
>       a2dp_sink.previous();
>    }
> }
> 
> void changeState(bool, int, void*) {
>   if (player_active) {
>       player_state = !player_state;
>       Serial.println(player_state);
>       if (player_state){
>           player.stop();
>       } else {
>           player.play();
>       }
>   } else {
>       a2dp_state = !a2dp_state;
>       Serial.println(a2dp_state);
>       if (a2dp_state){
>           a2dp_sink.pause();
>       } else {
>           a2dp_sink.play();
>       }
>   }
> }
> 
> void mode() {
> .
> .
> .
> player.setVolume(0.7);
> player.begin();
> Serial.println("Player started");
> kit.addAction(PIN_KEY2, changeState);
> kit.addAction(PIN_KEY4, next);
> kit.addAction(PIN_KEY3, previous);
> ```
> 
> if I comment out `kit.addAction(PIN_KEY2, changeState);` the functions work fine and song gets played but player.stop() and player.play() doesn't play the song and gives timeout in serial monitor.

> Hi @pschatzmann, did you replicate my problem? did you find the solution? Please help! 🥲
> 
> PS: I also tried it without Bluetooth a2dp (just normal playing code) the audio breaks periodically like stop...start...stop...start

I assume you use the AudioKit: different kits have different pin conflicts. Did you forget to deactivate the SD functionality with cfg.sd_active = false; ?

closed 2022-08-28 by pschatzmann

gnpaone, 2022-08-28:
> > if I comment out `kit.addAction(PIN_KEY2, changeState);` the functions work fine and song gets played but player.stop() and player.play() doesn't play the song and gives timeout in serial monitor.
> 
> I assume you use the AudioKit: different kits have different pin conflicts. Did you forget to deactivate the SD functionality with cfg.sd_active = false; ?
> 
> For Boothoth you are about the 10th person asking the same question!
> 
> I am closing this issue because the original question has been answered...

Ok, unfortunately sd card is not detected if sd_active is set to false, I'm using ai-thinker v2.2 a149 board. Please help. Understood for the bluetooth part,

pschatzmann, 2022-08-28:
If the SD and PIN_KEY2 are using internally the same GPIO there is nothing you can do and you need to use some other key for this.
Did you test the signals that you receive on PIN_KEY2 when you run some file operations...

gnpaone, 2022-08-28:
Yah, thanks for your reply :) I just now understood the problem, I changed the function of PIN_KEY2 and it's triggering that function everytime. I have one more doubt, I need to execute a function when pressing the button (like press and hold) and function to stop if released (it would be better to run another function if button released), how to implement it?

pschatzmann, 2022-08-28:
kit.audioActions() gives you an AudioAction object. There you have a add version with 2 callbacks
https://pschatzmann.github.io/arduino-audio-tools/html/classaudio__tools_1_1_audio_kit_stream.html