pschatzmann/arduino-audio-tools #314: There is no Note off when I release the key on the midi keyboard.

janczeresnia, 2022-08-27:
Hello you wrote an amazing toolkit thanks for your work.
I have a problem with midi support. I wrote using examples a simple bluetooth midi-controlled synthesizer with audio output on the built-in ESP32 DAC. Almost everything works fine. When I press the key on the midi keyboard, the synthesizer starts to play, but when I release the key, it does not stop playing. Function onNoteOff() is not executed. Only the onNoteOn() functions are performed when I press a key on the midi keyboard, onControlChange() when I turn the controllers and onPitchBend() when I turn the pitch is executed.

I am using the latest of Arduino ESP32 Version 2.0.4
MCU: ESP32-WROOM-DA
E-Mu Xboard25 midi keyboard connected to a computer via USB

Xboard25<-USB->Computer<-Bluetooth->ESP32-Synth<-Analog->Audio amplifier

Serial Log:
```
Note on : channel 0, note 60, velocity 127 <- when I press a key
Note on : channel 0, note 60, velocity 0 <- when I release a key
```
when I turn the controllers:
```
Control Change : channel 0, controller 1, value 0
Control Change : channel 0, controller 1, value 3
```
when I turn the pitch:
```
Pitch Bend : channel 0, value 0
Pitch Bend : channel 0, value 87
```
My sketch:
```
/**
 * ESP32 Bluetooth MIDI Syntch Basic with adsr on internal dac (pin 25 and 26)
 * https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/streams-synthbasic3-audiokit/streams-synthbasic3-audiokit.ino
 * https://www.pschatzmann.ch/home/2021/12/17/ai-thinker-audio-kit-building-a-simple-synthesizer-with-the-audiotools-library/
 */

#include "AudioTools.h"
#include <Midi.h>

typedef int16_t sound_t;                // sound will be represented as int16_t (with 2 bytes)
uint16_t sample_rate=44100;             // The stream will have 44100Hz sample rate
uint8_t channels = 2;                   // The stream will have 2 channels

SineWaveGenerator<sound_t> sine;
ADSRGain adsr(0.001, 0.001, 0.9, 0.002);

class SynthAction : public MidiAction {
  public:
    void onNoteOn(uint8_t channel, uint8_t note, uint8_t velocity) {
      Serial.printf("Note on: channel %d, note %d, velocity %d\n", channel, note, velocity);
      int freq = MidiCommon::noteToFrequency(note);
      sine.setFrequency(freq);
      adsr.keyOn();
      }
    }
    void onNoteOff(uint8_t channel, uint8_t note, uint8_t velocity) {
      Serial.printf("Note off: channel %d, note %d, velocity %d\n", channel, note, velocity);
      adsr.keyOff();
    }
    void onControlChange(uint8_t channel, uint8_t controller, uint8_t value) {
      Serial.printf("Control Change: channel %d, controller %d, value %d\n", channel, controller, value);
    }
    void onPitchBend(uint8_t channel, uint8_t value) {
      Serial.printf("Pitch Bend: channel %d, value %d\n", channel, value);
    }
} action;

MidiBleServer ble("ESP32-BTMidi-SynthBasic", &action);
AnalogAudioStream dac;
AudioEffects<SineWaveGenerator<int16_t>> effects(sine);
GeneratedSoundStream<sound_t> sound(effects);
StreamCopy copier(dac, sound);

void setup() {
  Serial.begin(115200);
  while (!Serial && (millis() < 1000)) delay(1000);
  Serial.println(F("\nESP32 Bluetooth MIDI SynthBasic\n"));
  //AudioLogger::instance().begin(Serial,AudioLogger::Info);
  AudioLogger::instance().begin(Serial,AudioLogger::Warning);
  //AudioLogger::instance().begin(Serial,AudioLogger::Debug);

  // setup midi
  Serial.println(F("Setup Bluetooth MIDI"));
  ble.begin(action);

  // setup effects
  Serial.println(F("Setup Effects"));
  effects.addEffect(adsr);

  // setup analog output
  Serial.println(F("Setup DAC Audio Output"));
  auto cfg = dac.defaultConfig(TX_MODE);
  cfg.sample_rate = sample_rate;
  cfg.channels = channels;
  dac.begin(cfg);

  // Setup sine wave
  Serial.println(F("Setup Sine Wave Generator"));
  sine.begin(channels, sample_rate, 0);
}

void loop() { copier.copy(); }
```

What is the reason for this?

pschatzmann, 2022-08-27:
I have tested this with PocketMidi. 
I suggest that you try if you can reproduce this with some other way (software) that generates midi messages.

janczeresnia, 2022-08-27:
Thank you for the quick reply. It looks like you're right. I installed a virtual midi keyboard and everything works fine :)
For reasons unknown to me, the Xboard25 keyboard does not send the Note Off midi message.
By the way, you can add my sketch to your library examples.
Regards and thank you.

closed 2022-08-27 by janczeresnia

pschatzmann, 2022-08-27:
Just some additional info: I just found the following on the internet: The use of MIDI ON with a velocity of zero is a perfectly normal and legitimate way of doing NOTE OFF

janczeresnia, 2022-08-27:
Yes of course I will do my best.
I have a few more questions for you.
For example, I do not know how to encode a sound from an analog microphone connected to the internal ADC with the Codec2 or Vorbis codec and send it via the RTSP protocol? To the computer. Is it even possible?
The tool you wrote gives a great opportunity.
I would like to test a dozen more functions.
Sorry for my English but it is not my native language.

janczeresnia, 2022-08-27:
I updated your arduino-midi library and everything works great on the Xboard25 keyboard great job :)
```
Note on: channel 0, note 60, velocity 98 <-- when I press a key
Note off: channel 0, note 60, velocity 0 <-- when I release a key
```

pschatzmann, 2022-08-27:
Good to know that the issue is resolved.

For RTSP I was playing around with https://github.com/pschatzmann/Micro-RTSP-Audio
I think it should be possible to integrate some encoding, but I have never tried...

janczeresnia, 2022-08-27:
I tested your example communication-generator-rtsp.ino and it works with ffmpeg without problems, I don't know how to add audio encoding to this example?

pschatzmann, 2022-08-27:
Have a look at the file RTSPFormat.h: You will need to implement your own subclass of RTSPFormat. 
You can take the class RTSPFormatPCM  as an example..

janczeresnia, 2022-08-27:
I'm too weak at programming to do that. I understand that in the current RTPS implementation it is possible to transmit PCM 16bit 16kHz audio.
I changed the example communication-generator-rtsp.ino so that the audio source was AnalogueAudioStream adc but it doesn't work.
FFMPEG not connect to ESP32.
I tested an analog microphone with base-adc-serial.ino example and there is audio from the microphone.

My sketch:
```
/**
 * ESP32 RTSP Analogue Microphone connect to internal DAC GPIO 34
 * https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-communication/rtsp/communication-generator-rtsp
 */

#include "AudioTools.h"
#include "AudioLibs/MicroRTSPAudio.h"
#include "AudioStreamer.h"
#include "RTSPServer.h"

int port = 554;
int channels = 1;
int sample_rate = 16000;
int bits_per_sample = 16;
const char* ssid = "ssid";
const char* password = "password";

AnalogAudioStream adc;                              // Stream from ADC Analogue Microphone
RTSPSourceAudioStream source(adc);                  // Stream sound via RTSP
AudioStreamer streamer = AudioStreamer(&source);
RTSPServer rtsp = RTSPServer(&streamer, port);

void setup() {
  Serial.begin(115200);
  while (!Serial && (millis() < 1000)) delay(1000);
  Serial.println(F("\nESP32 Analog Microphone to RTSP\n"));
  //AudioLogger::instance().begin(Serial, AudioLogger::Info);
  //AudioLogger::instance().begin(Serial, AudioLogger::Warning);
  AudioLogger::instance().begin(Serial, AudioLogger::Debug);

  // setup ADC
  auto cfg = adc.defaultConfig(RX_MODE);
  cfg.sample_rate = sample_rate;
  cfg.channels = channels;
  cfg.bits_per_sample = bits_per_sample;
  adc.begin(cfg);

  // Start Wifi & rtsp server
  rtsp.begin(ssid, password);
}

void loop() { delay(1000); }
```

Did I do it right?