pschatzmann/arduino-audio-tools #315: strange STK Voicer behavior voicer.noteOff() continues to play until it is run twice.

janczeresnia, 2022-08-28:
Hi I tested a synthesizer based on the STK library and noticed strange behavior. If I press the midi key on the keyboard and once the sytnezator plays forever until I press the midi key again.

My Sketch:
```
/**
 * ESP32 Bluetooth MIDI STK Voicer Syntch on internal DAC (pin 25 and 26)
 * https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/streams-synthstk-audiokit/streams-synthstk-audiokit.ino
 * https://www.pschatzmann.ch/home/2021/12/21/ai-thinker-audiokit-a-simply-synthesizer-with-stk/
 */

#include "AudioTools.h"
#include "StkAll.h"
#include <Midi.h>

int group = 0;

AnalogAudioStream dac;
Clarinet clarinet(440);
Voicer voicer;
ArdStreamOut output(&dac);

class SynthAction : public MidiAction {
  public:
    void onNoteOn(uint8_t channel, uint8_t note, uint8_t velocity) {
      Serial.printf("Note on: channel %d, note %d, velocity %d\n", channel, note, velocity);
      voicer.noteOn(note, (float)velocity, group);
    }
    void onNoteOff(uint8_t channel, uint8_t note, uint8_t velocity) {
      Serial.printf("Note off: channel %d, note %d, velocity %d\n", channel, note, velocity);
      voicer.noteOff(note, (float)velocity, group);
    }
    void onControlChange(uint8_t channel, uint8_t controller, uint8_t value) {
      Serial.printf("Control Change: channel %d, controller %d, value %d\n", channel, controller, value);
    }
    void onPitchBend(uint8_t channel, uint8_t value) {
      Serial.printf("Pitch Bend: channel %d, value %d\n", channel, value);
    }
} action;

MidiBleServer ble("ESP32-BTMidi-STKSynth", &action);

void setup() {
  Serial.begin(115200);
  while (!Serial && (millis() < 1000)) delay(1000);
  Serial.println(F("\nESP32 Bluetooth MIDI STK Voicer Synth\n"));
  //AudioLogger::instance().begin(Serial,AudioLogger::Info);
  AudioLogger::instance().begin(Serial,AudioLogger::Warning);
  //AudioLogger::instance().begin(Serial,AudioLogger::Debug);

  // setup midi
  Serial.println(F("Setup Bluetooth MIDI"));
  ble.begin(action);

  // Setup STK synthesizer
  Serial.println(F("Setup STK Voicer Instrument"));
  Stk::setSampleRate(44100.0);
  voicer.addInstrument(&clarinet, group);

  // setup analog output GPIO 25 and 26
  Serial.println(F("Setup DAC Audio Output"));
  auto cfg = dac.defaultConfig(TX_MODE);
  cfg.channels = 1;
  cfg.bits_per_sample = 16;
  cfg.sample_rate = Stk::sampleRate();
  dac.begin(cfg);
}

void loop() { output.tick(voicer.tick()); }
```

Serial Log:

```
Note on: channel 0, note 60, velocity 127 <-- press the key first time
Note off: channel 0, note 60, velocity 0  <-- synthesizer do not stops playing

Note on: channel 0, note 60, velocity 127 <-- press the key a second time
Note off: channel 0, note 60, velocity 0  <-- the synthesizer stops playing

Clarinet::stopBlowing: argument is less than or equal to zero!
```

Regards.

pschatzmann, 2022-08-28:
Not sure, but maybe the velocity 0 might disturb: Try to give a value...

janczeresnia, 2022-08-28:
I gave a constant velocity value and nothing has changed.
```
int v = 127;
class SynthAction : public MidiAction {
  public:
    void onNoteOn(uint8_t channel, uint8_t note, uint8_t velocity) {
      Serial.printf("Note on: channel %d, note %d, velocity %d\n", channel, note, velocity);
      voicer.noteOn(note, v, group);
    }
    void onNoteOff(uint8_t channel, uint8_t note, uint8_t velocity) {
      Serial.printf("Note off: channel %d, note %d, velocity %d\n", channel, note, velocity);
      voicer.noteOff(note, v, group);
    }
    void onControlChange(uint8_t channel, uint8_t controller, uint8_t value) {
      Serial.printf("Control Change: channel %d, controller %d, value %d\n", channel, controller, value);
    }
    void onPitchBend(uint8_t channel, uint8_t value) {
      Serial.printf("Pitch Bend: channel %d, value %d\n", channel, value);
    }
} action;
```
but there is no message in the log:
```
Clarinet::stopBlowing: argument is less than or equal to zero!
```

pschatzmann, 2022-08-28:
I am pretty sure that the following example was working for me:
https://github.com/pschatzmann/Arduino-STK/blob/master/examples/OutToBluetooth/OutToBluetooth.ino
Try it with v=20;

janczeresnia, 2022-08-28:
I tried nothing changed.

janczeresnia, 2022-08-28:
I changed the sketch to:
```
/**
 * ESP32 Bluetooth MIDI STK Voicer Syntch on internal DAC (pin 25 and 26)
 * https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/streams-synthstk-audiokit/streams-synthstk-audiokit.ino
 * https://www.pschatzmann.ch/home/2021/12/21/ai-thinker-audiokit-a-simply-synthesizer-with-stk/
 */

#include "AudioTools.h"
#include "StkAll.h"
#include <Midi.h>

int group = 0;

AnalogAudioStream dac;
Clarinet clarinet(440);
Voicer voicer;
ArdStreamOut output(&dac);

class SynthAction : public MidiAction {
  public:
    void onNoteOn(uint8_t channel, uint8_t note, uint8_t velocity) {
      Serial.printf("Note on: channel %d, note %d, velocity %d\n", channel, note, velocity);
      voicer.noteOn(note, (StkFloat)velocity, group);
    }
    void onNoteOff(uint8_t channel, uint8_t note, uint8_t velocity) {
      Serial.printf("Note off: channel %d, note %d, velocity %d\n", channel, note, velocity);
      voicer.noteOff(note, (StkFloat)velocity+20, group);
    }
    void onControlChange(uint8_t channel, uint8_t controller, uint8_t value) {
      Serial.printf("Control Change: channel %d, controller %d, value %d\n", channel, controller, value);
    }
    void onPitchBend(uint8_t channel, uint8_t value) {
      Serial.printf("Pitch Bend: channel %d, value %d\n", channel, value);
    }
} action;

MidiBleServer ble("ESP32-BTMidi-STKSynth", &action);

void setup() {
  Serial.begin(115200);
  while (!Serial && (millis() < 1000)) delay(1000);
  Serial.println(F("\nESP32 Bluetooth MIDI STK Voicer Synth\n"));
  //AudioLogger::instance().begin(Serial,AudioLogger::Info);
  AudioLogger::instance().begin(Serial,AudioLogger::Warning);
  //AudioLogger::instance().begin(Serial,AudioLogger::Debug);

  // setup midi
  Serial.println(F("Setup Bluetooth MIDI"));
  ble.begin(action);

  // Setup STK synthesizer
  Serial.println(F("Setup STK Voicer Instrument"));
  Stk::setSampleRate(44100.0);
  voicer.addInstrument(&clarinet, group);

  // setup analog output GPIO 25 and 26
  Serial.println(F("Setup DAC Audio Output"));
  auto cfg = dac.defaultConfig(TX_MODE);
  cfg.channels = 1;
  cfg.bits_per_sample = 16;
  cfg.sample_rate = Stk::sampleRate();
  dac.begin(cfg);
}

void loop() { output.tick(voicer.tick()); }
```
The strange behavior continues.

but there is no message in the log:
```
Clarinet::stopBlowing: argument is less than or equal to zero!
```

pschatzmann, 2022-08-28:
The only suggestion that I would have is to use the Clarinet directly - there you have more control over the parameters

janczeresnia, 2022-08-28:
Wow I changed the midi library implementation to <BLEMidi.h> and everything works great :)

There seems to be something wrong with your implementation of midi <Midi.h>.

New sketch:
```
/**
 * ESP32 Bluetooth BLE MIDI STK Voicer Synth on internal DAC (GPIO 25 and 26)
 * https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/streams-synthstk-audiokit/streams-synthstk-audiokit.ino
 * https://www.pschatzmann.ch/home/2021/12/21/ai-thinker-audiokit-a-simply-synthesizer-with-stk/
 * https://github.com/max22-/ESP32-BLE-MIDI/blob/master/examples/04-Dump-Messages/04-Dump-Messages.ino
 */

#include "AudioTools.h"
#include "StkAll.h"
#include <BLEMidi.h>

int group = 0;

AnalogAudioStream dac;
Clarinet clarinet(440);
Voicer voicer;
ArdStreamOut output(&dac);

// BLEMidi Callback
void onNoteOn(uint8_t channel, uint8_t note, uint8_t velocity, uint16_t timestamp) {
  Serial.printf("Note on : channel %d, note %d, velocity %d (timestamp %dms)\n", channel, note, velocity, timestamp);
  voicer.noteOn(note, (StkFloat)velocity, group);
}

void onNoteOff(uint8_t channel, uint8_t note, uint8_t velocity, uint16_t timestamp) {
  Serial.printf("Note off : channel %d, note %d, velocity %d (timestamp %dms)\n", channel, note, velocity, timestamp);
  voicer.noteOff(note, (StkFloat)velocity, group);
}

void onAfterTouchPoly(uint8_t channel, uint8_t note, uint8_t pressure, uint16_t timestamp) {
  Serial.printf("Polyphonic after touch : channel %d, note %d, pressure %d (timestamp %dms)\n", channel, note, pressure, timestamp);
}

void onControlChange(uint8_t channel, uint8_t controller, uint8_t value, uint16_t timestamp) {
  Serial.printf("Control change : channel %d, controller %d, value %d (timestamp %dms)\n", channel, controller, value, timestamp);
}

void onProgramChange(uint8_t channel, uint8_t program, uint16_t timestamp) {
  Serial.printf("Program change : channel %d, program %d (timestamp %dms)\n", channel, program, timestamp);
}

void onAfterTouch(uint8_t channel, uint8_t pressure, uint16_t timestamp) {
  Serial.printf("After touch : channel %d, pressure %d (timestamp %dms)\n", channel, pressure, timestamp);
}

void onPitchbend(uint8_t channel, uint16_t value, uint16_t timestamp) {
  Serial.printf("Pitch bend : channel %d, value %d (timestamp %dms)\n", channel, value, timestamp);
}

void setup() {
  Serial.begin(115200);
  while (!Serial && (millis() < 1000)) delay(1000);
  Serial.println(F("\nESP32 Bluetooth BLE MIDI STK Voicer Synth\n"));
  AudioLogger::instance().begin(Serial,AudioLogger::Info);
  //AudioLogger::instance().begin(Serial,AudioLogger::Warning);
  //AudioLogger::instance().begin(Serial,AudioLogger::Debug);

  // setup midi
  BLEMidiServer.begin("ESP32 STK Voicer Synth");
  BLEMidiServer.setOnConnectCallback([]() {
    Serial.println(F("ESP32 MIDI Connected"));
  });
  BLEMidiServer.setOnDisconnectCallback([]() {
    Serial.println(F("ESP32 MIDI Disconnected"));
  });
  BLEMidiServer.setNoteOnCallback(onNoteOn);
  BLEMidiServer.setNoteOffCallback(onNoteOff);
  BLEMidiServer.setAfterTouchPolyCallback(onAfterTouchPoly);
  BLEMidiServer.setControlChangeCallback(onControlChange);
  BLEMidiServer.setProgramChangeCallback(onProgramChange);
  BLEMidiServer.setAfterTouchCallback(onAfterTouch);
  BLEMidiServer.setPitchBendCallback(onPitchbend);
  //BLEMidiServer.enableDebugging();

  // setup STK synthesizer
  Serial.println(F("Setup STK Voicer Instrument"));
  Stk::setSampleRate(44100.0);
  voicer.addInstrument(&clarinet, group);

  // setup analog output GPIO 25 and 26
  Serial.println(F("Setup DAC Audio Output"));
  auto cfg = dac.defaultConfig(TX_MODE);
  cfg.channels = 1;
  cfg.bits_per_sample = 16;
  cfg.sample_rate = Stk::sampleRate();
  dac.begin(cfg);
}

void loop() { output.tick(voicer.tick()); }
```

pschatzmann, 2022-08-28:
Strange: what values are different, that you receive via BLEMidi?

janczeresnia, 2022-08-28:
Serial log:
```
Note on : channel 0, note 50, velocity 81 (timestamp 967ms) <-- I press the key (synth play)
Note on : channel 0, note 50, velocity 0 (timestamp 1330ms) <-- I release the key (synth stop play)

Clarinet::startBlowing: one or more arguments is less than or equal to zero!
```
callback funtcion onNoteOff() never executes.

pschatzmann, 2022-08-28:
I am confused. Isn't this the issue that you have closed yesterday, after the correction ?

janczeresnia, 2022-08-28:
I don't know, it might be like that. Strange thing.

pschatzmann, 2022-08-28:
You have updated the midi library...right ?

janczeresnia, 2022-08-28:
Yes i use <BLEMidi.h> from https://github.com/max22-/ESP32-BLE-MIDI

closed 2022-09-13 by janczeresnia