pschatzmann/arduino-audio-tools #316: Audio not getting recorded in streams-looper example

gnpaone, 2022-08-28:
In that example actions.add(PIN_KEY2, record_start, record_end); no longer working, the key pressing is not activated (confirmed by no output in serial monitor when pressed as it's present in example to serial print text when button pressed (record start) and released (record end), when I tried using kit.addAction(PIN_KEY2, record_start, record_end); I get compilation error with failed to convert void[](bool, int, void*) to void[-fpermissive]

gnpaone, 2022-08-30:
[image]

I'm getting this error. Please help!
P.S. PSRAM is enabled in Tools

pschatzmann, 2022-08-30:
These folder contains some work in progress, experiments and tests, so do not expect them to work. But it might still serve as a source of inspiration...

So I don't accept this as an issue.

closed 2022-08-30 by pschatzmann

gnpaone, 2022-08-30:
Logs when compiling verbose enabled in preferences:
```
In file included from C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioOutput.h:7,
                 from C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioFilter/Equilizer.h:3,
                 from C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:16,
                 from C:\Users\gnpao\Documents\Arduino\streams-looper\streams-looper.ino:12:
C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreams.h: In member function 'virtual void audio_tools::MemoryStream::resize(size_t)':
C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreams.h:277:9: error: label 'PS_RAM' defined but not used [-Werror=unused-label]
         PS_RAM:
         ^~~~~~
In file included from C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioI2S/I2SStream.h:9,
                 from C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:18,
                 from C:\Users\gnpao\Documents\Arduino\streams-looper\streams-looper.ino:12:
C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioI2S/I2SESP32.h: In member function 'bool audio_tools::I2SBase::begin(audio_tools::I2SConfig, int, int)':
C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioI2S/I2SESP32.h:156:7: warning: missing initializer for member 'i2s_driver_config_t::mclk_multiple' [-Wmissing-field-initializers]
       };
       ^
C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioI2S/I2SESP32.h:156:7: warning: missing initializer for member 'i2s_driver_config_t::bits_per_chan' [-Wmissing-field-initializers]
In file included from C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioAnalog/AnalogAudio.h:3,
                 from C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:20,
                 from C:\Users\gnpao\Documents\Arduino\streams-looper\streams-looper.ino:12:
C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioAnalog/AnalogAudioESP32.h: In member function 'bool audio_tools::AnalogAudioStream::begin(audio_tools::AnalogConfig)':
C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioAnalog/AnalogAudioESP32.h:215:9: warning: missing initializer for member 'i2s_driver_config_t::fixed_mclk' [-Wmissing-field-initializers]
         };
         ^
C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioAnalog/AnalogAudioESP32.h:215:9: warning: missing initializer for member 'i2s_driver_config_t::mclk_multiple' [-Wmissing-field-initializers]
C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioAnalog/AnalogAudioESP32.h:215:9: warning: missing initializer for member 'i2s_driver_config_t::bits_per_chan' [-Wmissing-field-initializers]
In file included from C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioKit.h:3,
                 from C:\Users\gnpao\Documents\Arduino\streams-looper\streams-looper.ino:13:
C:\Users\gnpao\Documents\Arduino\libraries\arduino-audiokit-main\src/AudioKitHAL.h: In member function 'i2s_config_t AudioKitConfig::i2sConfig()':
C:\Users\gnpao\Documents\Arduino\libraries\arduino-audiokit-main\src/AudioKitHAL.h:138:5: warning: missing initializer for member 'i2s_driver_config_t::fixed_mclk' [-Wmissing-field-initializers]
     };
     ^
C:\Users\gnpao\Documents\Arduino\libraries\arduino-audiokit-main\src/AudioKitHAL.h:138:5: warning: missing initializer for member 'i2s_driver_config_t::mclk_multiple' [-Wmissing-field-initializers]
C:\Users\gnpao\Documents\Arduino\libraries\arduino-audiokit-main\src/AudioKitHAL.h:138:5: warning: missing initializer for member 'i2s_driver_config_t::bits_per_chan' [-Wmissing-field-initializers]
In file included from C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:23,
                 from C:\Users\gnpao\Documents\Arduino\streams-looper\streams-looper.ino:12:
C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreamsConverter.h: In member function 'virtual void audio_tools::ChannelFormatConverterStream::setAudioInfo(audio_tools::AudioBaseInfo)':
C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreamsConverter.h:122:94: warning: this statement may fall through [-Wimplicit-fallthrough=]
                  static_cast<ChannelFormatConverterStreamT<int8_t>*>(converter)->setAudioInfo(cfg);
                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreamsConverter.h:123:15: note: here
               case 16:
               ^~~~
C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreamsConverter.h:124:95: warning: this statement may fall through [-Wimplicit-fallthrough=]
                  static_cast<ChannelFormatConverterStreamT<int16_t>*>(converter)->setAudioInfo(cfg);
                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreamsConverter.h:125:15: note: here
               case 24:
[... cut ...]
C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreamsConverter.h:126:95: warning: this statement may fall through [-Wimplicit-fallthrough=]
                  static_cast<ChannelFormatConverterStreamT<int24_t>*>(converter)->setAudioInfo(cfg);
                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreamsConverter.h:127:15: note: here
               case 32:
               ^~~~
cc1plus.exe: some warnings being treated as errors
```

gnpaone, 2022-08-30:
```0x40083e29: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c line 402
0x4008b7ad: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c line 128
0x40090bc1: __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c line 85
0x40090823: multi_heap_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c line 245
0x40084169: heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 340
0x40090bf1: free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/heap.c line 39
0x400eadb1: operator delete(void*) at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/del_op.cc line 49
0x400eb3b9: operator delete[](void*) at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/del_opv.cc line 35
0x400d2e61: audio_tools::DynamicMemoryStream::DataNode::~DataNode() at C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreams.h line 331
0x400d4465: audio_tools::DynamicMemoryStream::write(unsigned char const*, unsigned int) at C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioBasic/Collections/List.h line 248
0x400d44ec: audio_tools::StreamCopyT ::write(unsigned int, unsigned int&) at C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h line 247
0x400d461a: audio_tools::StreamCopyT ::copy() at C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h line 267
0x400d46b2: loop() at C:\Users\gnpao\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h line 337
0x400d8be1: loopTask(void*) at C:\Users\gnpao\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.4\cores\esp32\main.cpp line 50```

Stack trace result

gnpaone, 2022-08-30:
> Compiling with verbose is not supported!

Oh ok 😅, just tried 😉

pschatzmann, 2022-08-30:
Seems that some heap is overwritten. Maybe some memory allocations are failing...

gnpaone, 2022-08-30:
Ohh, should I change anything in code maybe? 🤔

pschatzmann, 2022-08-30:
My suggestion would be to build up and test the functionality in baby steps: first just try to write audio to PSRAM

gnpaone, 2022-08-30:
🥲 Yah, that thing I tried, it's failing at that point only, is it possible to not use inputmixer or maybe convert the incoming stream to bytes and play as raw as in some examples playing from memory

pschatzmann, 2022-08-30:
You would try to add the mixer only after the recording is working...

gnpaone, 2022-08-30:
Any simple example to try dynamicmemorystream?

pschatzmann, 2022-08-30:
I added this class to support the looper, so this is also not finally tested yet...

pschatzmann, 2022-08-30:
yes single bytes

gnpaone, 2022-08-30:
Oh ok

pschatzmann, 2022-08-30:
Maybe just another thought: it might be an option to build the functionality using files

gnpaone, 2022-08-31:
Unfortunately that's not working too (working with fixed raw or audio stream but not with mic). Please help

pschatzmann, 2022-08-31:
Why do you think that files are not working. You can just copy the output from the mic to a file. Just don't forget to close the file at the end or all the content is lost. That will give a PCM file that you can take as a copy source.

Or better use a WavEncoder/Decoder, so that you can easily double check the content on your desktop.

gnpaone, 2022-09-01:
I use this for recording:
AudioKitStream kit; // Access I2S as stream
SdFs SD;
InputMixer<int16_t> mixer;
File audioFile;                                                   // final output stream
EncodedAudioStream out_stream(&audioFile, new WAVEncoder()); 
StreamCopy copier_in(out_stream, mixer);

For recording:
void record_start(bool pinStatus, int pin, void* ref){
  Serial.println("record_start");
  //SD.begin(SdSpiConfig(PIN_AUDIO_KIT_SD_CARD_CS, DEDICATED_SPI, SD_SCK_MHZ(10) , &AUDIOKIT_SD_SPI));
  if (!SD.begin(SdSpiConfig(PIN_AUDIO_KIT_SD_CARD_CS, DEDICATED_SPI, SD_SCK_MHZ(10) , &AUDIOKIT_SD_SPI))) {
    Serial.println("Failed to mount");
    return;
  }
  //SPIFFS.begin(FORMAT_SPIFFS_IF_FAILED);
  audioFile = SD.open("002.wav", FILE_WRITE);
  is_record = true;
  //is_play = false;
  //copier_in.begin(out_stream, mixer);
  executed = true;
}

void record_end(bool pinStatus, int pin, void* ref){
  if (executed) {
    //copier_in.end();
    Serial.println("record_end");
    is_record = false;
    audioFile.close();
    /*audioFile = SD.open("002.wav");
    decoder.setNotifyAudioChange(kit);
    is_play = true;
    decoder.begin();
    copier_out.begin(decoder, audioFile);*/
    executed = false;
  }
}

In void setup:
record_start();
delay(5000);
record_end();

In void loop:
if (is_record) {
       copier_in.copy();
}

File is getting created but it's 0 bytes, recording is not being done

gnpaone, 2022-09-01:
PS if I use mixer.add(kit); intead of mixer.begin(cfg); I get AudioCopy.h : 253 - write to target has failed! in serial monitor

pschatzmann, 2022-09-01:
Your sequence is wrong. It should be
record_start();
copier_in.copy(); // during 5 seconds
record_end();

gnpaone, 2022-09-01:
Um actually I tried ur code without any change https://github.com/pschatzmann/arduino-audio-tools/wiki/Encoding-and-Decoding-of-Audio#encoding and even there I got write to target has failed error 
[image]

pschatzmann, 2022-09-01:
Here is an [file based example](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/streams-audiokit-sd-audiokit/streams-audiokit-sd-audiokit.ino) that works for me

gnpaone, 2022-09-01:
Awesome 😍, it's working perfectly fine

gnpaone, 2022-09-01:
Just that I made small correction [SPI RAM](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/streams-audiokit-ram-audiokit/streams-audiokit-ram-audiokit.ino) version, since the button is by default ActiveLow so it executes record_end too at program start, to correct it this program change: (other functions are normal like when clicked record_start runs and when released record_end runs it's working fine)

'''
uint16_t sample_rate = 16000;
uint8_t channels = 1;  // The stream will have 2 channels 
AudioKitStream kit;
MemoryManager memory(500); // Activate SPI RAM for objects > 500 bytes
DynamicMemoryStream recording(true); // Audio stored on heap 
StreamCopy copier; // copies data
bool executed = false;
 

void record_start(bool pinStatus, int pin, void* ref){
  Serial.println("Recording...");
  recording.begin();
  copier.begin(recording, kit);
  executed = true;
}

void record_end(bool pinStatus, int pin, void* ref){
  if (executed) {
    Serial.println("Playing...");
    copier.begin(kit, recording);  // start playback
    executed = false;
  }
}'''

pschatzmann, 2022-09-01:
If you check my latest update, I have added some additional post processing logic to deal with the clicking noise...

gnpaone, 2022-09-01:
Ohh ok, will check that too :)

gnpaone, 2022-09-02:
Thanks 😎, It's working great 😃

gnpaone, 2022-09-02:
One more doubt with copier.begin(kit, recording);  // start playback in ram example, how to stop output from looping infinitely and play only once?
Also is there any way to change a stream with for eg 16000 sample rate to 44100 without losing quality similar to how ffmpeg works (`ffmpeg -i input.wav -ar 44100 output.wav` where input.wav is of 16khz sample rate and output.wav is 44.1khz)

pschatzmann, 2022-09-02:
You could resample, but there is no point in doing this because you would not gain anything:
Details can be found in the Wiki

gnpaone, 2022-09-05:
I tried to resample
auto cfgr = resample.defaultConfig();
    cfgr.channels = 2;
    cfgr.sample_rate_from = 44100; 
    cfgr.sample_rate = 16000;
    resample.begin(cfgr);
but I am getting this error Incompatible buffer length 512 for down sampling. It must be a factor of 11
Actually for my case I use multiple audio source outputting at the same time having different sample rate so if one sample rate is set since it doesn't match other audio sample it sounds awful so I want to resample it

pschatzmann, 2022-09-05:
Did you try to define a copy buffer size with a factor of 11
StreamCopy copier(1100); // copies data

gnpaone, 2022-09-05:
For trial purpose I tried on audiokit-ram example where kit sample_rate is 44100 and resample is 16000

gnpaone, 2022-09-06:
Please help! 🥲

pschatzmann, 2022-09-06:
Just change it to a multiple of 11 as proposed!

gnpaone, 2022-09-06:
It's not working unfortunately, please try audiokit ram example with these
uint16_t sample_rate = 44100;
uint8_t channels = 2;  // The stream will have 2 channels 
AudioKitStream kit;
MemoryManager memory(500); // Activate SPI RAM for objects > 500 bytes
DynamicMemoryStream recording(true,); // Audio stored on heap 

instead of these 
uint16_t sample_rate = 16000;
uint8_t channels = 1;  // The stream will have 2 channels 
AudioKitStream kit;
MemoryManager memory(500); // Activate SPI RAM for objects > 500 bytes
DynamicMemoryStream recording(true); 

and any way for output sound in 44100 to sound same as 16000?

gnpaone, 2022-09-07:
> and any way for output sound in 44100 to sound same as 16000?

How to do that?

pschatzmann, 2022-09-07:
So why don't you record with 44100 in the first place ?
If you want to resample use some rates which are easily convertable by multiplying or dividing: e.g. 16000 and 48000 or 
 11025  and 44100  otherwise you will only get an approximation.

gnpaone, 2022-09-07:
Yah, I read the docs. I get Incompatible buffer length 512 for down sampling. It must be a factor of 29 for 11025 and 44100 and Incompatible buffer length 512 for down sampling. It must be a factor of 3 for 16000 and 48000

pschatzmann, 2022-09-07:
What is your definition for StreamCopy ?

gnpaone, 2022-09-08:
Actually, I need this audiokit-ram example in the place of sd card in bluetooth-sd-a2dp example, bluetooth plays good at 44100 and sounds awful at 16000 whereas recording plays good at 16000 and there's a kind of breaks in between in 44100. My test code:
```/**
 * @file streams-audiokit-sd-audiokit.ino
 * @author Phil Schatzmann
 * @brief We record the input from the microphone to SPI RAM and constantly repeat to play the file
 * The input is triggered by pressing key 1. Recording stops when key 1 is released!
 * @version 0.1
 * @date 2022-09-01
 * 
 * @copyright Copyright (c) 2022
 * 
 */
#include "AudioTools.h"
#include "AudioLibs/AudioKit.h"
#include "AudioLibs/MemoryManager.h"

uint16_t sample_rate = 44100;
uint8_t channels = 2;  // The stream will have 2 channels 
AudioKitStream kit;
//FilteredStream<int16_t, float> filtered(kit, 1);
MemoryManager memory(500); // Activate SPI RAM for objects > 500 bytes
DynamicMemoryStream recording(true); // Audio stored on heap 
ResampleStream<int16_t> resample(recording);
StreamCopy copier_in;
StreamCopy copier_out;
//float coef[] = { 0.021, 0.096, 0.146, 0.096, 0.021};
bool is_play = false;
bool is_record = false;
bool executed = false;
 

void record_start(bool pinStatus, int pin, void* ref){
  Serial.println("Recording...");
  recording.begin();
  is_play = false;
  is_record = true;
  copier_in.begin(recording, kit);
  executed = true;
}

void record_end(bool pinStatus, int pin, void* ref){
  if (executed) {
    Serial.println("Playing...");
    is_play = true;
    is_record = false;
    recording.postProcessSmoothTransition<int16_t>(2, 0.01, 6);
    auto cfgr = resample.defaultConfig();
    cfgr.channels = 2;
    cfgr.sample_rate_from = 44100; 
    cfgr.sample_rate = 11025;
    resample.begin(cfgr);
    copier_out.begin(kit, resample);
    executed = false;
  }
}

void setup(){
  Serial.begin(115200);
  while(!Serial); // wait for serial to be ready
  AudioLogger::instance().begin(Serial, AudioLogger::Warning);
  //filtered.setFilter(0, new FIR<float>(coef));
  //filtered.setFilter(1, new FIR<float>(coef));

  // setup input and output
  auto cfg = kit.defaultConfig(RXTX_MODE);
  cfg.sd_active = true;
  cfg.i2s_format = I2S_LSB_FORMAT;
  cfg.bits_per_sample = 16;
  cfg.sample_rate = sample_rate;
  cfg.channels = channels;
  cfg.input_device = AUDIO_HAL_ADC_INPUT_LINE2;
  kit.begin(cfg);
  kit.setVolume(1.0);

  // record when key 1 is pressed
  kit.audioActions().add(PIN_KEY1, record_start, record_end);
  Serial.println("Press Key 1 to record");

}

void loop(){

  if (is_record){
    copier_in.copy();
  }
  if (is_play && !copier_out.copy()){
    is_play = false;
  }

  // Process keys
  kit.processActions();

}

gnpaone, 2022-09-08:
Please help 🥲

pschatzmann, 2022-09-08:
You still did not change the size in StreamCopy to be a multiple of the indicated value.